MichaIng/DietPi #3930: How to flash Buster to emmc on NanoPC-T4?

keeno1982, 2020-11-24:
I found that only Stretch can be flash to EMMC, Buster is a standalone SD image, I have no idea to flash it to EMMC. Could you help to provide Buster like Stretch img with flash tools?

MichaIng, 2020-11-24:
The new image can be flashed to eMMC directly as is, but in case of NanoPC T4 the eMMC cannot be removed and has no port for an adapter cable or such, right?

I think for now this means you need to flash a simply image to SD card first and flash the final image from SD card booted system to internal eMMC 🤔.

Currently we have no one who knows about how to provide an image with those Android flash tools, I'm AFRAID.

keeno1982, 2020-11-25:
@MichaIng Yes, the Stretch pack was easy to use before. If possible, flash image to SD first then to EMMC also good as you said.
like official as blow
http://wiki.friendlyarm.com/wiki/index.php/NanoPC-T4#Flash_Image_to_eMMC

MichaIng, 2020-11-26:
Jep, I at least won't find time to look into the Android tools flash image soon, first of all Bullseye image building based on Debian kernel and bootloader from scratch is on the time table from my end on the image side. But if someone else manages to create one for NanoPi T4, probably including support for other (RK3399) boards, and either document it reproducible or via script, I'm happy to upload and update it 🙂.

StephanStS, 2022-03-19:
@keeno1982: Could our actual blog _[Moving a running DietPi system to a USB stick/disk or an onboard eMMC](https://dietpi.com/blog/?p=1236)_ help?