MichaIng/DietPi #3940: Error in OctoPrint software after update

Stinocon, 2020-12-01:
### Creating a bug report/issue

#### Required Information
- DietPi version

G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=33
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | 10.6
- Kernel version | Linux octopi 5.4.72-v8+ #1356 SMP PREEMPT Thu Oct 22 13:58:52 BST 2020 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Original RPi power supply
- SDcard used | Sandisk Ultra 32GB

#### Additional Information (if applicable)
- Software title | OctoPrint
- Was the software title installed freshly or updated/migrated? Updated/Migrated
- Can this issue be replicated on a fresh installation of DietPi? Yes
- Bug report ID | f8d69f3c-5ee8-4f97-92b2-0a5b40a7cf1d

#### Steps to reproduce
1. Install OctoPrint from DietPi store
2. Update OctoPrint form software web page

#### Expected behaviour
The software should update and a web interface should appear on port 5000

#### Actual behaviour
OctoPrint software stays blocked to the web page "Connecting to Octoprint Server..."

#### Extra details
If i try to start the service manually there are some lost components, OctoPrint probably expects to update things that were not imported in the installation from DietPi software store: 

```
2020-12-01 17:05:35,971 - octoprint.plugins.pi_support - WARNING - Fetching the current throttle state via "/usr/bin/vcgencmd get_throttled" doesn't work

2020-12-01 17:05:36,459 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 265, in fixed_wrap_cache
    content = func().getvalue()
  File "/usr/local/lib/python3.7/dist-packages/webassets/merge.py", line 250, in func
    data = StringIO(hunk.data())
  File "/usr/local/lib/python3.7/dist-packages/webassets/merge.py", line 78, in data
    f = open(self.filename, 'r', encoding='utf-8')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.7/dist-packages/octoprint/static/js/lib/jquery/jquery-ui.js'

2020-12-01 17:05:36,529 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 265, in fixed_wrap_cache
    content = func().getvalue()
  File "/usr/local/lib/python3.7/dist-packages/webassets/merge.py", line 250, in func
    data = StringIO(hunk.data())
  File "/usr/local/lib/python3.7/dist-packages/webassets/merge.py", line 78, in data
    f = open(self.filename, 'r', encoding='utf-8')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.7/dist-packages/octoprint/static/js/lib/hls.js'

2020-12-01 17:05:37,460 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 265, in fixed_wrap_cache
    content = func().getvalue()
  File "/usr/local/lib/python3.7/dist-packages/webassets/merge.py", line 250, in func
    data = StringIO(hunk.data())
  File "/usr/local/lib/python3.7/dist-packages/webassets/merge.py", line 78, in data
    f = open(self.filename, 'r', encoding='utf-8')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.7/dist-packages/octoprint/static/js/app/viewmodels/loginui.js'

2020-12-01 17:05:38,214 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
[... 56 lines cut ...]
    f = open(self.filename, 'r', encoding='utf-8')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.7/dist-packages/octoprint/static/js/lib/hls.js'

2020-12-01 17:05:40,633 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 265, in fixed_wrap_cache
    content = func().getvalue()
  File "/usr/local/lib/python3.7/dist-packages/webassets/merge.py", line 250, in func
    data = StringIO(hunk.data())
  File "/usr/local/lib/python3.7/dist-packages/webassets/merge.py", line 78, in data
    f = open(self.filename, 'r', encoding='utf-8')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.7/dist-packages/octoprint/static/js/app/viewmodels/loginui.js'

2020-12-01 17:05:40,733 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 265, in fixed_wrap_cache
    content = func().getvalue()
  File "/usr/local/lib/python3.7/dist-packages/webassets/merge.py", line 250, in func
    data = StringIO(hunk.data())
  File "/usr/local/lib/python3.7/dist-packages/webassets/merge.py", line 78, in data
    f = open(self.filename, 'r', encoding='utf-8')
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.7/dist-packages/octoprint/static/js/app/viewmodels/loginui.js'
2020-12-01 17:05:40,792 - octoprint.server.preemptive_cache - INFO - ... done in 4.97s

```

Joulinar, 2020-12-01:
Hi,

many thanks for your message. I tried to replicate the issue without success. I installed Octprint as well in my RPi4B (aarch64) and run the update offered by Octprint to version 1.5.0. 

[image]

the only thing needed was a reboot of the whole system as it seems self initiated service restart did not worked correctly.

Did you tried to reinstall Octoprint? `dietpi-software reinstall 153`

Stinocon, 2020-12-01:
Thank you for your prompt response!
I should also point out that only OctoPrint is running on this Raspberry, it was taken specifically for this purpose.

Having said that... 
I have now tried to give a `dietpi-software reinstall 153` but without any appreciable result.
The software reinstalls but the problems persist.

The problem arises either by starting octoprint as a service using `service octoprint start` or by placing in the /opt/octoprint/ directory and running the `octoprint serve` command.

The errors are the same as in the main post. 

I specify that I am running the commands as root.

MichaIng, 2020-12-01:
Let's start from the beginning. Can you run the following to check for the missing vcgencmd command:
```
vcgencmd get_throttled
which vcgencmd
```

Stinocon, 2020-12-02:
Sure, here is the output:

```
root@octopi:/home/dietpi# vcgencmd get_throttled
throttled=0x0
root@octopi:/home/dietpi# which vcgencmd
/usr/bin/vcgencmd
root@octopi:/home/dietpi#
```

MichaIng, 2020-12-02:
Ah I guess root permissions are required to call it:
```
2020-12-02 17:11:25 root@micha:/var/log# sudo -u www-data vcgencmd get_throttled
VCHI initialization failed
2020-12-02 17:13:17 root@micha:/var/log# vcgencmd get_throttled
throttled=0x0
```
And with the last update we switched to a non-root service user. Hmm, not sure whether permission to run vcgencmd can be gained via certain group or capability. Will check that, however, it is just a warning, not responsible for the failure.

Responsible for the failure is a bunch of missing files in `/usr/local/lib/python3.7/dist-packages/octoprint/`.
Can you please try:
```
cd /opt/octoprint
pip3 install -U . # in case with sudo, i.e. with root permissions
chown -R octoprint:octoprint /opt/octoprint
systemctl restart octoprint
```

If it still does not work, probably the left files from the old instance are responsible. Currently not sure why I chose to not remove the previous install dir but merge the new files into the old directory instead 🤔 (as configs have been moved to `/mnt/dietpi_userdata/octoprint`).

So another test would be, if the above fails the same way:
```
rm -R /opt/octoprint
git clone --depth 1 https://github.com/OctoPrint/OctoPrint.git /opt/octoprint
cd /opt/octoprint
pip3 install -U . # in case with sudo, i.e. with root permissions
chown -R octoprint:octoprint /opt/octoprint
systemctl restart octoprint
```

Stinocon, 2020-12-02:
Regarding the first test... Nothing to do, the errors remain the same.
I tried then also with the second test and... Same mistakes. 
I also tried rebooting the system between one test and another but the situation does not change.

At this point having a backup of OctoPrint... I tried a more drastic solution:

1) Enter the dietpi-software menu 
2) Remove the software
3) Restart system
4) Re-enter dietpi-software menu
5) Install OctoPrint
6) The installed version STARTS but is obviously not updated
7) I follow the installation wizard but I can't import the backup, so I finish it by entering random data and start the software update from OctoPrint, which works.
8) I restart the system, the UI starts correctly (EUREKA) and the backup imports correctly!

At this point... I can't say what happened/changed. The imported backup is 5 minutes before my last octoprint update (and just before it broke). 
The system doesn't have other particular packages except the dependencies required by OctoPrint... 

I really don't know

MichaIng, 2020-12-02:
> 6. The installed version STARTS but is obviously not updated

Hmm, will try the same now on RPi.

MichaIng, 2020-12-02:
Interesting, I'm pretty sure that Git was used by the internal updater, the last time I checked, however now pip is used only. This means the install/Git dir in `/opt/octoprint` is hence obsolete 😄.

The problem is that pip global install requires root permissions. The internal updater however does not recognise that itself is a globally installed instance and runs pip in user mode, hence installing a second instance to `/mnt/dietpi_userdata/octoprint/.local/bin/octoprint` (the binary + the python dependencies across other dirs in `.local`). As well it cannot restart itself, and the external restart loads the previous globally installed version.
________
Further check:
- The internal updater always invokes pip in "--user" mode, as long as it is not within a virtual environment.
- So running OctoPrint as non-root user without a virtual environment would require it to be installed in user mode in the first place, otherwise the internal updater will install an unused obsolete instance.

So now the question is, whether a virtual environment or user mode install is the better choice. Not sure what the practical difference are but on first view user mode install seems to be simpler since there is no need to "activate" a venv first but `sudo -u octoprint pip3 ...` is enough, as well for manual OctoPrint updates from console e.g.

MichaIng, 2020-12-02:
Ah found a better way, probably: https://docs.octoprint.org/en/master/configuration/config_yaml.html#server

Change setting: `localPipCommand: sudo pip3`, probably `sudo /usr/local/bin/pip3`

I'll add a sudoers config as well to allow OctoPrint restarting itself, calling pip the defined way, shutting down and restarting the server, to re-enable those commands from the web UI.

axieyangb, 2020-12-03:
Same error here!

MichaIng, 2020-12-04:
It did cost some headache and I ran into a nasty bug on Debian Bullseye, but have a working solution now that is still based on a system-level install of OctoPrint, i.e. being installed to `/usr/local/bin/octoprint`, but granting the octoprint user permission to use pip with sudo.

The alternative would be to make it a user-level install, i.e. installed with `octoprint` user into its own home directory `/mnt/dietpi_userdata/octoprint/.local/bin/octoprint` including all python dependencies. That might lead to some doubled modules but makes the environment independent from system-level installs, i.e. avoids version conflicts, similar to a venv.

For now I fixed/implemented it the way that it works as intended before, but I'm open for opinions whether a local install into the octoprint directory would be better in your opinion or not.

Stinocon, 2020-12-05:
Honestly I find the solution adopted now the best one, I have never been a fan of user level installed applications (where there is no real need of course).

Then in daily use I think that an installation method rather than another one to the end user who just wants to print in 3D changes very little...

MichaIng, 2020-12-05:
> I have never been a fan of user level installed applications

For single-user SBCs or pure server devices I totally agree. I've never heard of an SBC that is used by multiple users which all install their own server/daemon applications, so that e.g. multiple OctoPrint servers would be running on the same machine 😄.

But just to clarify, is would have been a per-user level install for the `octoprint` service user, so very similar to a per-service Python virtuelenv/venv, so not a separate instance for every user that logs in the system 😉.

> Then in daily use I think that an installation method rather than another one to the end user who just wants to print in 3D changes very little...

That is true, but with ~~2000 OctoPrint instances it is worth to think through pros and cons of the different options 🙂.

1activegeek, 2020-12-08:
Hey guys, just going to chime in for following and to offer another testing bed. I think I'm in the same boat of issue with the latest update to the Octoprint install. I'm receiving similar errors around permissions trying to update octoprint and install plugins. One such offending line excerpt:

```
octoprint.util.commandline.CommandlineError: (1, ["Python executable: '/usr/bin/python3'\n", '>>> Running: git diff --shortstat\n', "b'> git diff --shortstat'\n", '>>> Running: git fetch\n', "b'> git fetch'\n", '>>> Running: git pull\n', "b'> git pull'\n", "b'  Updating 38256815e..2697bd7a4'\n", "b'  Fast-forward'\n", "b'   docs/api/job.rst | 6 ++++--'\n", "b'   1 file changed, 4 insertions(+), 2 deletions(-)'\n", '>>> Running: python setup.py clean\n', "b'> /usr/bin/python3 setup.py clean'\n", "b'  running clean'\n", 'b"  removing \'build/bdist.linux-armv7l\' (and everything under it)"\n', 'b"  recursively removing *.pyc from \'src\'"\n', 'b"  removing \'t/octoprint/src/octoprint_setuptools/__pycache__/__init__.cpython-37.pyc\'"\n', "b'  removed t/octoprint/src/octoprint_setuptools/__pycache__ since it was empty'\n", 'b"  removing \'t/octoprint/src/octoprint_setuptools/__init__.pyc\'"\n', '>>> Running: python setup.py install\n', "b'> /usr/bin/python3 setup.py install'\n", "b'  running install'\n"], ['b"  \'build/lib\' does not exist -- can\'t clean it"\n', 'b"  \'build/scripts-3.7\' does not exist -- can\'t clean it"\n', 'b"  error: can\'t create or remove files in install directory"\n', "b'  '\n", "b'  The following error occurred while trying to add or remove files in the'\n", "b'  installation directory:'\n", "b'  '\n", 'b"      [Errno 13] Permission denied: \'/usr/local/lib/python3.7/dist-packages/test-easy-install-31032.write-test\'"\n', "b'  '\n", "b'  The installation directory you specified (via --install-dir, --prefix, or'\n", "b'  the distutils default setting) was:'\n", "b'  '\n", "b'      /usr/local/lib/python3.7/dist-packages/'\n", "b'  '\n", "b'  Perhaps your account does not have write access to this directory?  If the'\n", "b'  installation directory is a system-owned directory, you may need to sign in'\n", 'b\'  as the administrator or "root" account.  If you do not have administrative\'\n', "b'  access to this machine, you may wish to choose a different installation'\n", "b'  directory, preferably one that is listed in your PYTHONPATH environment'\n", "b'  variable.'\n", "b'  '\n", "b'  For information on other options, you may wish to consult the'\n", "b'  documentation at:'\n", "b'  '\n", "b'    https://setuptools.readthedocs.io/en/latest/easy_install.html'\n", "b'  '\n", "b'  Please make the appropriate changes for your system and try again.'\n", "b'  '\n", 'Traceback (most recent call last):\n', '  File "/usr/local/lib/python3.7/dist-packages/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 431, in <module>\n', '    main()\n', '  File "/usr/local/lib/python3.7/dist-packages/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 427, in main\n', '    install_source(python_executable, folder, user=args.user, sudo=args.sudo)\n', '  File "/usr/local/lib/python3.7/dist-packages/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 335, in install_source\n', '    returncode\n', 'RuntimeError: Could not update, "python setup.py install" failed with returncode 1\n'])
```

MichaIng, 2020-12-08:
I'm not sure how well the regular update works since the updater itself has vastly changed:
- As you can see in our output, it was using Git.
- Now it is using pip only to pull the OctoPrint module directly from PyPI repository.

I suggest to update to our dev branch, which will be released as beta later today:
```
dietpi-backup 1
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=dev' /boot/dietpi.txt
dietpi-update 1
```
This includes a reinstall of OctoPrint, preserving your current config and data, of course.

1activegeek, 2020-12-08:
> I'm not sure how well the regular update works since the updater itself has vastly changed:
>
> * As you can see in our output, it was using Git.
> * Now it is using pip only to pull the OctoPrint module [...]

Got it. I will look to try that later. Octoprint has been a decently finicky application to run on dietpi - but I think it's also a bit finicky in itself too.

Just to validate I understand, each of those 3 lines is a command to run verbatim on the CLI to the dietpi host?

MichaIng, 2020-12-09:
> Just to validate I understand, each of those 3 lines is a command to run verbatim on the CLI to the dietpi host?

Exactly. the first to create/updater your backup, which is always good when switching branch or installing new software, just in case 😉, the second to switch to the beta branch (I just changed it, since the beta is out now) and the third to run the update.

If you are anyhow unhappy, you can the easily revert via `dietpi-update -1` (or skip the `-1` argument and use the menu).

1activegeek, 2020-12-09:
Thanks for the confirm. Just wanted to be sure as I was thinking the second line might be an output from line 1. :) 

So in running through this, I believe I have in fact been able to get this running again. Specifically, my plugins have all come back, and thankfully I believe the data for them was still intact. In the Octoprint console there was a "cleanup" list of plugins, so I knew which plugins to re-install. After doing this, my Octoprint looks to be back in shape, pre-update. I think there are a few random settings that have been changed along somewhere, but I'll see after resetting them if there is any real issue (such as a permission missing maybe). I have one plugin that is being stubborn and won't install - but I'll have to dig into it later.

So for the time being - I would say it looks like this does indeed fix the issue. Unfortunately, it will require some manual work on re-installing the plugins, but their data seems intact. Will keep you posted when I try testing some more.

MichaIng, 2020-12-09:
Many thanks for the feedback. Yes the last OctoPrint update was a larger one. The changes in the internal updater was what we recognised (what broke the internal updater with our implementation basically), but very likely that something around the plugins and settings has changed as well. Yes, please let us know if there is anything not working, especially permissions, so that we can implement/fix it prior to v6.34 release. Also if there is some settings/plugin migration we can automate with the update.

1activegeek, 2020-12-09:
Ya it's a good catch and well timed. I think I had updated to the version that broke things first. Then the dietpi update helped the service at least start. Now its all working! 🙌🏻  

The plugin was a plugin specific issue it appears. In case anyone else is using the Bed Visualizer, there is a note on their plugin page that for Python3 implementations there is a library needed or it will fail silently. 

> Also if there is some settings/plugin migration we can automate with the update.

Can you point me to where the data/plugins may have been stored earlier? It might be possible they're still there and I can validate there might be a way to simply migrate them. As I recall, most of the plugins are simply a call to install necessary OS libs and then a few linked files usually. If I can test pulling those over you may be able to fully migrate? I do vaguely recall Octoprint indicating that plugins are not backed up, however, but their settings are. I'll leave that to the pro's to know! 😉

MichaIng, 2020-12-09:
> Can you point me to where the data/plugins may have been stored earlier?

AFAIK all plugins are own Python modules as well, right? So should have been installed to the same system Python path as now: `/usr/local/lib/python3.X/dist-packages/`
Ah but the problem is the Python 2 => Python 3 migration, so the plugins were installed for Python 2 but not Python 3 🤔. While this step was important to preserve support with upcoming Debian Bullseye (which does not natively ship EOL Python 2 anymore) and since Python 2 is EOL and won't get security updates anymore, when applying the switch, I didn't really have the OctoPrint plugins in mind. At least that would have been worth a warning in the changelog and/or update process.

1activegeek, 2020-12-09:
Ok that totally makes sense then. That would explain why the first update rendered it dead. Then my update of dietpi, brought me into the realm of living and the service starting. The plugins being python2 is a good point - so it likely wouldn't even work to move them as there may be minor discrepancies in the actual plugin. Perhaps a readout somewhere of current plugins installed (aka folders stored) just so user knows they'll need to re-install. Not sure there is a way to automate it on OP side. That would be best for them to implement so backups could also include a listing to auto re-install. 

Either way, looks good to me then. I'll let it run, see if I can just play around make sure no other odd errors. I'll keep myself subbed for when the fix gets put in prod release so I can switch back to mainline.

MichaIng, 2020-12-09:
> Perhaps a readout somewhere of current plugins installed (aka folders stored) just so user knows they'll need to re-install. Not sure there is a way to automate it on OP side. That would be best for them to implement so backups could also include a listing to auto re-install.

I was just thinking the same. Although probably not all plugins are available for Python 3 or names changed, also not all plugins for OctoPrint also have it in their name: https://plugins.octoprint.org/
Best that could be done:
- `pip2 freeze | grep -i octoprint-`
- Loop through that list and try to reinstall each via `pip3 install -U`.

Full loop:
```
for i in $(pip2 freeze | mawk -F\= '/[Oo]cto[Pp]rint-/{print $1}'); do pip3 install -U $i; done
```
But that might also reinstall abandoned plugins 🤔.

1activegeek, 2020-12-09:
Looks likely the best option that can allow for re-install automagically - though you make a valid point. One of my plugins for example is no longer available as it's not compatible with P3 I believe. But running a `pip3 freeze | grep -i octoprint-` listed all but 1 of the installed plugins I believe. Not sure offhand which is missing - not wanting to crossref 25 plugins in the console. :D 

Bigger issue is likely that there are 3 supported models for install (which 2 I think are really just pip anyway): https://docs.octoprint.org/en/master/plugins/distributing.html

Not sure if you have a sample but I can work to sanitize a copy if it's of use as well to look at possibly just dumping the plugins list from the current config file for the user to the console? At least again just knowing what you had is useful at the very least when you have a long list like me. It's a yaml located in `/mnt/dietpi_userdata/octoprint/.octoprint/config.yaml`.

MichaIng, 2020-12-09:
> pip3 freeze | grep -i octoprint-

Since you were coming from Python 2: `pip3 freeze | grep -i octoprint-`

> Bigger issue is likely that there are 3 supported models for install

Luckily the destination for any Python module that should be recognised by the system-wide installed OctoPrint instance, is installed into the same directory structure, and it can be detected via the same `pip` commands. A real difference would only be if  OctoPrint was installed into a venv/virtualenv or as a user-level install into a users home directory, as then the modules/plugins would need to be searched there as well.

MichaIng, 2020-12-16:
_@bombjack7000 I ping you here as you might be interested as well._

Hey guys. It looks like there are still OctoPrint internal bugs with the method I was relying on to do a system-wide/global OctoPrint install: https://github.com/OctoPrint/OctoPrint/issues/3700#issuecomment-744025398

There is another issue present since Debian Bullseye that breaks the method at a different level: https://github.com/MichaIng/DietPi/issues/3959
I proposed a patch to enhance OctoPrint's internal pip command estimation which does not suffer from that issue on Bullseye: https://github.com/OctoPrint/OctoPrint/pull/3882

But it is unlikely that both issues are addressed with an OctoPrint release until December 20 when I want to release DietPi v6.34 at latest.
_____________
The problem is the following:
- We want OctoPrint to run as non-root system service user "octoprint".
- It can hence only do global/system-wide Python module installs via "sudo", which is required when OctoPrint itself shall be installed as global/system-wide Python module.
- The OctoPrint internal plugin manager and updater can be configured with a custom pip command, e.g. to wrap it into sudo, since v1.5.0, but as of my tests it does not respect the related setting in all cases. Plugins are installed correctly, but the self-updater ignores the configured custom pip command and uses the regular one instead, without sudo. Since it has no permissions to use it for global module installs, it adds the `--user` which installs OctoPrint into its own home directory as a local instance. Our system service still loads the global OctoPrint instance, hence the update does not do anything.
- Furthermore, the pip command check fails on Debian Bullseye due to a new by default enabled security feature, if pip is called via sudo. Two different users access the same temporary file in `/tmp` which is not permitted then. This issue is what I aim to resolve with the PR at OctoPrint, by not using any temporary file but reading the commands output directly.

So, since I do not want to run OctoPrint as root user anymore, as a strict rule for all software titles where I touch the install code, there are only two possibilities that would work without the above bugs resolved:
- Making OctoPrint a "venv" instance. This means an own Python environment is stored to e.g. `/opt/octoprint`. Aside of the internal updater, updates can then only be done by the octoprint user itself after activating the venv. A bid similar to how it needs to be done for "Home Assistant", which uses a "pyenv" instead of a "venv", but that works pretty similar: https://dietpi.com/phpbb/viewtopic.php?p=10416#p10416
- Making OctoPrint a "--user" level install. This means that the Python environment is stored to the octoprint users home directory, which would currently be `/mnt/dietpi_userdata/octoprint/.local/`. Aside of the internal updater, updates can then only be done by the octoprint user, but instead of activating the "venv", one needs to add the "--user" option to the pip command call.

For me, both above solutions look pretty similar: A dedicated Python environment that can only be accessed by the octoprint user. The second option looks a bid simpler to me. In both cases it means that all OctoPrint plugins (which are Python modules as well) need to be reinstalled after the migration (`dietpi-update`). This is especially nasty for this which already needed to do this after the last update, when OctoPrint got migrated from Python 2 to Python 3 with the reinstall.

I would go with the "--user" level install which seems to be the easiest implementation, also when it's about maintaining it from console, but if someone has any suggestion or a different idea, I am happy to hear it.

1activegeek, 2020-12-17:
Thanks for the well document, thought out, and extensive research and involvement to solving this. I always appreciate the stuff you guys do here - it's why I try to always run DietPi on my small boards. 

I'm with you and totally understand it, as I've run into this with HASS on another board running DietPi. 😃 As I started to read that one block, I started to go oh ya, like HA - then you mentioned exactly that. I'd lean on your wisdom as I don't know any better solutions and I think the pro/con you laid out makes sense. In the end, I think if there is fork-lift effort involved (plugin re-installs) - its just the nature of the beast. As long as there is something like breaking change notice listed in the release notes, you've done what you can to advise and help alert people of the coming change. 

Thanks as always - happy to test when the new release is ready.

MichaIng, 2020-12-19:
Nice, an X-based touch screen interface for OctoPrint: https://github.com/Z-Bolt/OctoScreen
Worth to turn into a software request? It requires a custom implementation on DietPi since the official install instructions expect it to be the only GUI program on the system, which is not how we like it 😉.

1activegeek, 2020-12-19:
That is pretty neat - though I dont' have a touch screen on mine. It's actually hidden away in a full case as an addon to my printer. So mine just hides away, and I use the actual printer interface if I'm in there nearby it.

closed 2020-12-20 by MichaIng

1activegeek, 2020-12-21:
Does this mean you've rolled the changes into the next release? I saw an OP update, so I wanted to wait and test on the new revision of DP first, and use that to test upgrade to latest OP.

MichaIng, 2020-12-21:
> Does this mean you've rolled the changes into the next release?

It's part of the just released DietPi v6.34 🙂.

1activegeek, 2020-12-21:
Wow - I really could have just read the release notes. That was dumb. Thank you though, I'll try it out this week! Keep up the fantastic work!! And happy holidays!

1activegeek, 2020-12-24:
Just followup, tested latest update and it all worked. Had to re-install the plugins as expected, but hopefully the last time! 😉 

One tip for anyone who lands here after updating for the first time in a long time - don't freak out about forgetting what plugins are there. Simply login as usual and navigate to the Plugin Manager. Once there, click on the Cleanup tab at the top. This will show you a laundry list of the plugins that have data, but no associated plugin. I'd suggest a second tab or more separate window in your favorite browser connected to the UI and on that section, while on the other you open the Get More section and start searching all your plugins quickly to reinstall. Once done it'll need a restart of octoprint, and you should be all set. Mine went seamlessly ... both times now 😄

MichaIng, 2020-12-24:
> Simply login as usual and navigate to the Plugin Manager. Once there, click on the Cleanup tab at the top. This will show you a laundry list of the plugins that have data, but no associated plugin.

That is an extremely helpful hint, many thanks for sharing 👍. Merry Xmas or holidays at least 🎄.