MichaIng/DietPi #3942: Airprint (Providing Printers to iOS-Devices)

Gill-Bates, 2020-12-01:
Hey there,

I was looking for a way to print to a PDF from my iPhone. I made it with the Raspberry Pi and Dietpi without any problems:

sudo apt install avahi-daemon cups python3-cups
#cupsctl --remote-admin 
#cupsctl --share-printers

MichaIng, 2020-12-02:
Many thanks for this. While we dropped Google Cloud Print support for v6.34 (due to service shutdown at end of the year), an alternative is highly appreciated.

While it finally is for Apple AirPrint clients, I guess it can be similarly configured to work with other clients? I wish there would be a directory to place multiple files for multiple printer definitions so that we can install CUPS as a basis and add/remove multiple printers/backends easily 🤔.

Joulinar, 2020-12-02:
What about using plain `cups`?

MichaIng, 2020-12-02:
Since I never used it, does it work OOTB with Windows and Android stock network print implementation?

I'm thinking further, if this works fine, it should be possible to set up a VPN for remove printing (outside of local network), as a native replacement for Google Cloud Print without any special client requirements.

ravenclaw900, 2020-12-02:
[This article](https://medium.com/@anirudhgupta281998/setup-a-print-server-using-raspberry-pi-cups-part-3-aaced48e9b89) provides instructions on how to use CUPS OOTB with all platforms.

Joulinar, 2020-12-02:
Cups, is working quite well on my Windows box. What I did is following

1. `apt update && apt install cups`
2. change following `nano /etc/cups/cupsd.conf`
```
# Only listen for connections from the local machine.
Listen 0.0.0.0:631
Listen /run/cups/cups.sock

# Restrict access to the server...
<Location />
  Order allow,deny
  Allow @Local
</Location>

# Restrict access to the admin pages...
<Location /admin>
  Order allow,deny
  Allow @Local
</Location>

# Restrict access to configuration files...
<Location /admin/conf>
  AuthType Default
  Require user @SYSTEM
  Order allow,deny
  Allow @Local
</Location>
```
3. `systemctl restart cups.service`
4. access Cups via https://192.168.0.12:631/
5. my HP network printer was detected automatically
6. printer added as `HP_Color_LaserJet_MFP_M277dw`
7. Inside Windows added the printer manually as internet port
`http://192.168.0.12:631/printers/HP_Color_LaserJet_MFP_M277dw`
8. did a test print with success 👍 
9. printing job is visible inside cups

[image]

MichaIng, 2020-12-02:
Nice, I guess with help of Avahi daemon the printer can be found automatically via network printer search.

I think hence the problem with the AirPrint implementation is the manual `/etc/cups/printers.conf` replacement. The file says that it is auto-created/edited by CUPS service, likely based on config file and/or web UI choices. So I assume that overwriting this file will remove other added printers. So if this cannot be configured easily with the web UI, the config needs to be appended instead of overwritten, but in a careful way as I guess only one "DefaultPrinter" entry can exist at a time and of course it must not be doubled.

Joulinar, 2020-12-02:
@MichaIng 
I guess issue on the first post is to create a PDF printer on cups. 

I continued playing with it and had no issues printing a document using Adobe App from an iPhone SE. Printer was detected automatically correctly.

Android was a little bit more tricky and thinks are not working ootb as on iOS. It was needed to install an additional app to be able to detect the CUPS printer.

ravenclaw900, 2020-12-02:
According to the article I posted above, you might be able to do it natively by going to `Settings → Connected Devices → Connection Preferences → Printing → Default Print Service`.

Joulinar, 2020-12-02:
well default print service did not detect the CUPS printer

Waxhaw, 2021-02-03:
> well default print service did not detect the CUPS printer

You have to turn on the Avahi services to that printers are 'shared' and found via Bonjour.  I actually work for a printer mfr and could rant pages on this to put everyone to sleep.

Here is a GREAT tool to help you 'sniff out' services being broadcasted on your network:
http://www.tildesoft.com

If you're printer and Avahi services are running correctly and you've 'shared' the printer in CUPS via command line or the Web interface it will show up there.

That's a great tool to help you figure out if your Bonjour services and network at setup correctly.

Happy Printing!

Joulinar, 2021-04-05:
@MichaIng 
should we add CUPS as printing solution into DietPi? It's not that difficult as it's an `apt` package. Most tricky thing is to perform changes on `cupsd.conf` as we would need to add specific settings into specific places. Or we replace the file with our own one :)

MichaIng, 2021-04-06:
Jep, it's THE UNIX printing system, so should be added to cover stopped CloudPrint and other front-ends. `G_CONFIG_INJECT` allows to set and in case add settings below other lines, which works well in case of `[section]`-separated config files. But if it's too fragile/complicated we can use our own as well.

Joulinar, 2021-04-06:
basically it's about changing LISTEN address, which should be easy, and to grant local access by adding `Allow @Local` into the 3 sections I posted above https://github.com/MichaIng/DietPi/issues/3942#issuecomment-737398666

MichaIng, 2021-04-07:
Looks like Apache2 syntax, although there "Order" and "Allow" directives have been deprecated and replaced by "Require" 😄.
Are parent location directives not inherited by sub directories, i.e. would the 
```
Order allow,deny
Allow @Local
```
block not be sufficient in the root directory block only?

Joulinar, 2021-04-07:
nope you will still get `access denied` on the admin page and need to allow it explicitly.

MichaIng, 2021-04-07:
So sub directories inherit directives from parent directories, as long as there is no separate `<Location` block for the sub directory defined? Trying to understand how it works 😄.

Joulinar, 2021-04-07:
Yep I guess that's how it works

LexiconCode, 2021-04-08:
I will be keeping my eye on this. Number of days ago I started tinkering with a raspberry pi W getting my label printer network connected. A work in progress.

The OS may need need to pull down drivers from the server samba for the printer.

See configuration of [smb.conf](https://www.makeuseof.com/tag/make-wireless-printer-raspberry-pi/)

Joulinar, 2021-04-08:
thx for sharing the guide. It give one good hint how to allow local access to cups

@MichaIng 
`cupsctl --remote-admin` will adjust `/etc/cups/cupsd.conf` as follow 😄 

```
# Allow remote access
Port 631

<Location />
  # Allow remote administration...
  Order allow,deny
  Allow @LOCAL
</Location>

<Location /admin>
  # Allow remote administration...
  Order allow,deny
  Allow @LOCAL
</Location>

<Location /admin/conf>
  AuthType Default
  Require user @SYSTEM
  # Allow remote access to the configuration files...
  Order allow,deny
  Allow @LOCAL
</Location>
```

Means no need to tweak it from our end. At least as long as we don't want to have specific hardening setup.

at the end it is quite simple setup

1. `apt install cups`
2. `cupsctl --remote-admin`

Joulinar, 2021-04-08:
Ok will go with 187

MichaIng, 2021-04-11:
I turned this issue into a CUPS request. I know there is more to add Airprint, but CUPS is a basis from where we can add more.

closed 2021-04-24 by MichaIng

NiNo-213, 2021-04-24:
Hi dears,
I´ve installed Dietpi current verion - all fine so far - tried to install CUPS - but somehow it didn´t worked for me. Must say, I´m not that familiar with programming etc. but interested to test and try. If there is any solution , you need testperson, I could help. 
let me know.

rgds

Joulinar, 2021-04-24:
Cups is not yet available. It will be on next version 7.1. however you could try current beta which was made available same days ago. This way you could check out cups

NiNo-213, 2021-04-24:
ok noted - can give it a try for sure. btw whatr abt tis guidline?

Joulinar, 2021-04-24:
> whatr abt tis guidline?

What do you mean?

NiNo-213, 2021-04-24:
I mean - this issue topic is based on the missing cups incl. a guidline description ow Micha installed it, isn´t it?
anyway, can´t find the beta you mentioned on the website, atleat for 32bit (for 64bit there is one but it won´t work with pi zero w, or?)

MichaIng, 2021-04-24:
See the beta notes [here](#4294) including a link on how to apply. You do not need to flash a new image, it's like a regular update but to a different branch 😉.

Now sure what you mean with the guideline? Do you mean a documentation page for it? Joulinar added one (not yet online): https://github.com/MichaIng/DietPi-Docs/pull/445/files#diff-d3c037c2488c59a223aa4f44bffd4e31cd25844b9397c3778bb5fecd923ecef8
But I think we can extend it, at least a step-by-step instruction about how to add a local printer via web interface 🙂.

Joulinar, 2021-04-24:
a very small `Where Do I Begin?` section is available on cups docs http://www.cups.org/doc/overview.html

NiNo-213, 2021-04-24:
Great - yes, I meant the instruction obviously. :-) I´ll check and try this one - if interested/required- will update accordingly. 
thanks for such prompt support and help.

MichaIng, 2021-04-24:
Nice, probably it makes even more sense then to link this (additionally) below the web UI access info to have it more prominent. It's very short indeed but probably sufficient and we should try to avoid duplicating existing docs 😉.

Joulinar, 2021-04-24:
@MichaIng 
are you able to add this directly on dev branch or should I open new PR on docs?

MichaIng, 2021-04-24:
Done: https://github.com/MichaIng/DietPi-Docs/commit/f8203e5cac96c18b81dacd15dbe8eedd50b94c1b

NiNo-213, 2021-04-28:
Just an update - works like a charm - perfect!

MichaIng, 2021-04-28:
Many thanks for testing and feedback 🙂!

LexiconCode, 2021-05-18:
Any reason CUPS 2.2.10 version is being used versus the latest CUPS 2.3.3?

ravenclaw900, 2021-05-18:
Yes, CUPS is installed through APT, and the Debian repository (for Buster), only has 2.2.10.  https://packages.debian.org/buster/cups

LexiconCode, 2021-05-21:
This might be a nice enhancement depending on interest.
https://github.com/OpenPrinting/cups-filters

MichaIng, 2021-05-22:
This seems to be part of the Debian `cups` package already:
- [cups](https://packages.debian.org/buster/cups) depends on [cups-filters](https://packages.debian.org/buster/cups-filters)