MichaIng/DietPi #3943: Updating radarr via internal updater to v3

simonfisher, 2020-12-01:
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=33
G_DIETPI_VERSION_RC=3
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
buster

- Kernel version | `uname -a`
Linux DietPi 4.14.32+ #1 SMP PREEMPT Thu Apr 5 12:46:33 UTC 2018 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Radarr

- Was the software title installed freshly or updated/migrated?
Updated 

- Can this issue be replicated on a fresh installation of DietPi?
Unknown.

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
1. Radarr began showing version 3.0.0.4204 as a new version in the internal updater
2. Click update

#### Expected behaviour
- Update to the new version.

#### Actual behaviour
- Update process appears to work ok
- Radarr restarted itself
- refreshing the website showed some cute loading phrases, and then error: "Failed to load languages from API"
- Also, other messages, including "Failed to load tags from API" and "Failed to load quality profiles from API" were shown on subsequent refreshes. 

#### Extra details
-

Joulinar, 2020-12-02:
Hi,

many thanks for your message. Did you tried to reboot your system after the update of Radarr?

simonfisher, 2020-12-02:
No I didn't. However, I just tried accessing radarr again, and aftering leaving it for an hour or so it's now working. No restarts of the service etc. So, not sure what happened there...

It does say this message:
> Please upgrade to the .NET Core version of Radarr

More info here: https://wiki.servarr.com/Radarr_System#update-to-net-core-version

Is this something Dietpi can support?

Drew80, 2020-12-02:
In the same boat.  Updated Radarr with the internal updater and I'm getting the same error message @simonfisher mentioned above.

> Please upgrade to the .NET Core version of Radarr

I did a reboot and also tried to reinstall the software using the built-in dietpi command.

`dietpi-software reinstall 145`

no luck.

Running on a Pine64 2GB Kickstarter board with success for a few years now with no major issues.

```
cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=33
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

MichaIng, 2020-12-02:
The reinstall does not touch the install files, but instead suggests to use the internal updater, like we do in all cases where an internal updater is available. This assures that no possible migration steps are missing what we cannot know of. You should have seen a related INFO print during the reinstall, although it might not be prominent enough.

However, the reinstall would still download the Mono-based version´anyway, so I guess there wouldn't be a difference. I'll just doing a test install now to verify that it is generally still working, or whether something of our implementation is broken.

But for now it looks like that Radarr internal updater is not updating correctly from v0.2.0 to v3.0.0, probably some setting is invalidated or database structure changed or something and the related migration is missing.

.NET core migration is another point which I'll implement right away (not a big deal). It requires a reinstall then definitely to update the systemd unit. One important question is whether the "arm" build works on ARMv6 (RPi 1/Zero) as well or only on ARMv7, like it is in case of Jackett. Someone here with such an RPi model for testing?

Joulinar, 2020-12-02:
I have a RPiB that could be used for testing

athisun, 2020-12-02:
I can say that the update worked fine for me, everything is working great.

Note that before updating using the internal updater, I changed the update branch to “master” in Radarr under Settings > General > (Show Advanced) > Update branch, as suggested in the Radarr Wiki.

MichaIng, 2020-12-02:
Ah sorry, I missed the update. Okay not sure what happened during that delay then.

> Note that before updating using the internal updater, I changed the update branch to “master” in Radarr under Settings > General > (Show Advanced) > Update branch, as suggested in the Radarr Wiki.

That is a good point. Luckily that branch migration can be done scripted via `/mnt/dietpi_userdata/radarr/config.xml` => `<Branch>master</Branch>`.

Yes fresh install works well. Implementing .NET core install now.

@Joulinar 
Could you try the following on RPi B after installing Radarr via `dietpi-software`:
```
systemctl stop radarr
apt -y install libicu63
cd /tmp
wget https://github.com/Radarr/Radarr/releases/download/v3.0.0.4204/Radarr.master.3.0.0.4204.linux-core-arm.tar.gz
tar xf Radarr.master.3.0.0.4204.linux-core-arm.tar.gz
rm -R /opt/Radarr
mv Radarr /opt/radarr
G_CONFIG_INJECT 'ExecStart=' 'ExecStart=/opt/radarr/Radarr -nobrowser -data=/mnt/dietpi_userdata/radarr' /etc/systemd/system/radarr.service
sed -i 's|/opt/Radarr|/opt/radarr|g' /etc/systemd/system/radarr.service
G_EXEC chown -R radarr:dietpi /opt/radarr
systemctl daemon-reload
systemctl start radarr
```

MichaIng, 2020-12-02:
A branch/PR is up: #3945
Works fine on x86_64, less memory usage and much faster install due to missing mono wrapper, I guess.

Probably we could inform users that Mono is not required anymore and can be uninstalled as long as:
- Sonarr and Lidarr are not used.
- Jackett is not used on an ARMv6 device (RPi 1 or Zero).

Otherwise the above leaves only a very small number of users, I guess, as those programs are often used in combination.

Joulinar, 2020-12-02:
> @Joulinar
Could you try the following on RPi B after installing Radarr via dietpi-software:

Will do

athisun, 2020-12-02:
@MichaIng It looks like there are works underway for a .NET Core version of Sonarr (or perhaps there already is one). There are instructions on the Servarr Wiki for installing a .NET Core version of Sonarr [here](https://wiki.servarr.com/Sonarr_Installation#.NET_Core_Install), but some of the steps are specific to Radarr, so I’m not too sure what’s going on there.

MichaIng, 2020-12-02:
Lol, those docs are copies from the Radarr docs, see the screenshot. Lidarr does not ship .NET builds yet (I checked that already) and at least when I asked last time (regarding Sonarr v3), an answer from Sonarr dev was that there are no plans to move away from Mono.

The "official" install instructions support this: https://sonarr.tv/#downloads-v3-linux
So I guess servarr.com was a bid hasty to copy&paste content and while Radarr links this website, the others do not (from what I found) and there is not hint that it is maintained by the software developers directly, so I'd always follow the official website or GitHub repository instructions 😉.

athisun, 2020-12-02:
Aha, my mistake. I assumed with there was some sort of developer collaboration revolution going on with the Radarr UI update to match Sonarr v3 and new Wiki popping up. Fingers crossed for the future, I suppose!

MichaIng, 2020-12-02:
Jep would be awesome (from our outsiders view) if there was some sort of collaboration to bundle forces and align setup and usage experience, but for now it looks like each team goes own ways and the overlap in code and methods is decreasing.

Joulinar, 2020-12-02:
@MichaIng 
It doesn't seems to be working on my RPiB.

```
root@DietPi1:/var/log/radarr# /opt/radarr/Radarr -nobrowser -data=/mnt/dietpi_userdata/radarr
Segmentation fault
root@DietPi1:/var/log/radarr#
```

MichaIng, 2020-12-02:
Okay, sadly the same as with Jackett so that we need to stay with Mono for ARMv6. Thanks for testing 👍.

pechex, 2020-12-02:
I just wanted to say that I experienced the same thing:
After updating, I got all those API errors in each screen, but I saw that it was refreshing all my movies, I have ~1500. It can take a LONG time to refresh all. I'm running on RPI2
I left it running and returned today and everything is working fine, expect that I get that .NET warning. Is that solvable?

Thanks

MichaIng, 2020-12-02:
The .NET issue is solvable, but requires a reinstall with new systemd unit. I'm currently preparing and testing that (see attached pull request) and let you know once it's ready to test.

Since v3 was a major Radarr upgrade, I guess there is some more intense database migration required during which the API errors appears. Just a guess and might be the reason for the media re-scan as well.

Joulinar, 2020-12-02:
to bad that Radarr  guys did not inform user about this migration and the potential long run time causing "error" messages 🤔

MichaIng, 2020-12-02:
I agree, however since everything works fine after it has been done, it's okay I think. Also as long as not auto-update is enabled, once can always watch the process activity to see if there is still some task going on.

dethknite, 2020-12-03:
I had this issue on my Win10 PC (was working for a couple days prior on v3), and got it working by changing the branch from develop to master by hand in the config. There is a not about doing this on the github release notes for the version... perhaps it fixed it... perhaps not. But that was the only change I made besides setting logging to Trace.

<Branch>master</Branch>

Drew80, 2020-12-07:
@MichaIng 
> The .NET issue is solvable, but requires a reinstall with new systemd unit. I'm currently preparing and testing that (see attached pull request) and let you know once it's ready to test.
>
> Since v3 [...]

No luck with the reinstall on a Pine64 2GB model.  After an uninstall an then rest reinstall, I followed these instructions

```
systemctl stop radarr
apt -y install libicu63
cd /tmp
wget https://github.com/Radarr/Radarr/releases/download/v3.0.0.4204/Radarr.master.3.0.0.4204.linux-core-arm.tar.gz
tar xf Radarr.master.3.0.0.4204.linux-core-arm.tar.gz
rm -R /opt/Radarr
mv Radarr /opt/radarr
G_CONFIG_INJECT 'ExecStart=' 'ExecStart=/opt/radarr/Radarr -nobrowser -data=/mnt/dietpi_userdata/radarr' /etc/systemd/system/radarr.service
sed -i 's|/opt/Radarr|/opt/radarr|g' /etc/systemd/system/radarr.service
G_EXEC chown -R radarr:dietpi /opt/radarr
systemctl daemon-reload
systemctl start radarr
```

, and on the Pine64:

`apt -y install libicu63`

gives me an error message of 

`E: Unable to locate package libicu63`

I am able to test on a the Pine64 board if need be.

Thanks in advance

MichaIng, 2020-12-07:
If it's a Stretch system, you need `libicu57` instead, on Bullseye `libicu67`, see: https://github.com/MichaIng/DietPi/pull/3945/files#diff-98445566b948cc76bb45181ccd463ee93a7a441dbc8b25763d1b88f05f4ef8c5R5704-R5715

At best test the dev branch directoy:
```
dietpi-backup 1
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=dev' /boot/dietpi.txt
dietpi-update 1
```
The update includes a Radarr reinstall/migration.

Drew80, 2020-12-08:
> If it's a Stretch system, you need `libicu57` instead, on Bullseye `libicu67`, see: https://github.com/MichaIng/DietPi/pull/3945/files#diff-98445566b948cc76bb45181ccd463ee93a7a441dbc8b25763d1b88f05f4ef8c5R5704-R5715
> 
> At best test it with `dev` branch:
> 
> ```
> dietpi-backup 1
> G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=dev' /boot/dietpi.txt
> dietpi-update 1
> ```
> 
> The update includes a Radarr reinstall/migration.

It worked, the second time.  Not sure what happened but its working now.

Drew80, 2020-12-08:
@MichaIng 
There is another issue with the AAR to RAM not working
This is from the Radarr log file.

```
2020-12-7 20:47:28.6|Warn|RadarrErrorPipeline|Invalid request Validation failed: 
 -- Arguments: Arguments are no longer supported for custom scripts
```

I can see that there is what you call an "ugly workaround"

```
echo -e '#!/bin/dash\nexec /boot/dietpi/misc/dietpi-arr_to_RAM 2 radarr' > /var/lib/dietpi/dietpi-software/radarr_backup_database.sh
chmod +x /var/lib/dietpi/dietpi-software/radarr_backup_database.sh
```

This was posted [here](https://dietpi.com/phpbb/viewtopic.php?f=9&t=8326&p=29186&hilit=Arr+to+RAM&sid=ebf33f048af0a980df7a4a03cc34a675#p29186) in the Dietpi forums.  Is a non ugly workaround coming as well?

Thank you for getting us back up and running.

MichaIng, 2020-12-08:
Since Radarr (and Sonarr) intentionally removed the ability to add command arguments to custom scripts, there is currently no other way then creating a wrapper script, sadly. When I find time, I'll open a request to recover the previous feature as I don't see any reason why this has been removed. Would be great if someone else could do this, actually 😉.

closed 2020-12-20 by MichaIng