MichaIng/DietPi #3944: Ich wollte DietPi installieren und es gab einen Fehler infolge dessen ich hier die *-setup.log hochladen sollte ... nun bin ich einwenig überfordert wie und wo ... gern auch per mail: henrik.nahr@t-online.de - Danke!

fencing69, 2020-12-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

fencing69, 2020-12-02:
[dietpi-firstboot.log](https://github.com/MichaIng/DietPi/files/5628531/dietpi-firstboot.log)

fencing69, 2020-12-02:
[dietpi-ramlog.log](https://github.com/MichaIng/DietPi/files/5628556/dietpi-ramlog.log)
[fs_partition_resize.log](https://github.com/MichaIng/DietPi/files/5628557/fs_partition_resize.log)

fencing69, 2020-12-02:
More logs sind dort nicht ... ;)

Joulinar, 2020-12-02:
Hi,

many thanks for your message. I would like to ask you to use English language if possible as this is an international platform. This would allow others to follow the investigation and support if possible. As well I removed your personal email address from the title 😃

MichaIng, 2020-12-02:
FirstBoot swap file generation fails due to insufficient free space.
```
[FAILED] DietPi-Set_swapfile | Free space check: path=/var/ | available=0 MiB | required=1073 MiB
[FAILED] DietPi-Set_swapfile | Insufficient free space for desired swap files size
```

Strangely the partition expansion succeeded before:
```
Device         Boot  Start       End   Sectors  Size Id Type
/dev/mmcblk0p1        8192    532479    524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 124823551 124291072 59.3G 83 Linux

The partition table has been altered.
resize2fs 1.44.5 (15-Dec-2018)
```
However, the log ends at this point while I am missing further output from resize2fs about the new file system size.

So based only on the logs, please check/do the following and paste the output:
```
df -T
resize2fs "$G_ROOTFS_DEV"
```

If this succeeds and shows expected ~60 GiB space, the swap file can be re-created via:
```
/boot/dietpi/func/dietpi-set_swapfile 1
```

fencing69, 2020-12-03:
Hallo and thanks for your answer!
The current view at the partitions as on screenshot from Gparted 
[image]

MichaIng, 2020-12-03:
While the file system has been expended sufficiently, there is strangely still 5.27 GiB unused. Did you run `resize2fs "$G_ROOTFS_DEV"` again as suggested and tried to re-create the swap file?

closed 2021-02-14 by MichaIng