MichaIng/DietPi #3946: dietpi installation using pre script on armbian buster error

hpcre, 2020-12-03:
#### Details:
- Date           | Thu Dec  3 00:56:52 UTC 2020
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       |  (ID=51)
- Kernel version | Linux clpi1 5.8.16-sunxi #20.08.14 SMP Tue Oct 20 22:15:32 CEST 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | setupcon --save
- Exit code      | 1
- Software title | DietPi-PREP
#### Steps to reproduce:

1. Installing deitpi on mint intall of Armbian buster using PREP script
2. ...
#### Expected behaviour:

- Normal installation without errors

#### Actual behaviour:

- Error reported as mentioned below. cannot find symbols/eg in any known directory

#### Extra details:

- ...

#### Additional logs:
```
/usr/bin/ckbcomp: Can not find file "symbols/eg" in any known directory
```

MichaIng, 2020-12-03:
Many thanks for your report. Indeed `eg` is no valid keyboard layout. Did you change the keyboard layout on the fresh Armbian image?

Edit `/etc/default/keyboard` and change the line `XKBLAYOUT="eg"` to e.g. `XKBLAYOUT="gb"` (Great Britain, default on our images) or any other valid keyboard layout code. You can find a list of supported layouts in: `/usr/share/X11/xkb/symbols/ `

hpcre, 2020-12-03:
Thanks, I'll try that.
The thing is, i didn't change anything. Debian buster install image (Armbian official image for Banana PI Pro) automatically detected my location (Egypt) and configured everything with EG locale. I removed the locale and tried to configure things to US or GB. Seems both didn't completely remove eg references.

hpcre, 2020-12-03:
I just slipped this step in the installation by changing the command to 'ls' and completely ignoring keyboard configuration.

MichaIng, 2020-12-03:
I'm not familiar with the Armbian firstrun setup tools and how they aim to detect this, but while the locale `ar_EG` Arabic (Egypt) exists, the keyboard layout `eg` does not.

I suggest you use the native Debian keyboard setup tool: `dpkg-reconfigure keyboard-configuration`
I guess `ara` (Arabic) would be correct then.

Probably we should overwrite `/etc/default/keyboard` with the defaults of our images during PREP 🤔, on first boot it is anyway re-set to the `dietpi.txt` `AUTO_SETUP_KEYBOARD_LAYOUT` setting, which defaults to "gb".

MichaIng, 2020-12-03:
I applies the change: https://github.com/MichaIng/DietPi/commit/d28671af26e3fef6480f3245356484aaaa2c4e2b

> the installation ended with a crash and back to prompt

You mean during DietPi-PREP or during first boot of the actual DietPi system? Since `setupcon --save` is called there as well, but not error handled, which means you might see the error in console log but it will not have an effect on ongoing boot process. And AFAIK an invalid keyboard layout only has an effect on keyboard input itself, not any boot or background processes.

hpcre, 2020-12-03:
> 
> 
> I applies the change: [d28671a](https://github.com/MichaIng/DietPi/commit/d28671af26e3fef6480f3245356484aaaa2c4e2b)
> 
> > the installation ended with a crash and back to prompt
> 
> You mean during DietPi-PREP or during first boot of the actual DietPi system? Since `setupcon --save` is called there as well, but not error handled, which means you might see the error in console log but it will not have an effect on ongoing boot process. And AFAIK an invalid keyboard layout only has an effect on keyboard input itself, not any boot or background processes.

first boot setup/config steps reach the keyboard config steps. when the command `setupcon --save` fails, there should be a menu that gives options like retry, edit command, etc...
while i was just trying to read the error, the config tool just ended abruptly and fell back to  command prompt. on the second re-image and setup, i had a chance to edit the command to skip this step and cotinue with the rest of the setup process.

hpcre, 2020-12-03:
> Many thanks for your report. Indeed `eg` is no valid keyboard layout. Did you change the keyboard layout on the fresh Armbian image?
>
> Edit `/etc/default/keyboard` and change the line [...]

it turns out i dont have to change anything. i checked my default keyboard file, and it switched to gb.

```
# KEYBOARD CONFIGURATION FILE

# Consult the keyboard(5) manual page.

XKBMODEL="pc105"
XKBLAYOUT="gb"
XKBVARIANT=""
XKBOPTIONS=""

BACKSPACE="guess"

```

MichaIng, 2020-12-03:
> first boot setup/config steps reach the keyboard config steps. when the command setupcon --save fails, there should be a menu that gives options like retry, edit command, etc...

It depends on whether the command is wrapped into our `G_EXEC` error handler or not. In DietPi-PREP it is now, so you will have that prompt to retry, change command etc. During first boot of the actual DietPi image (after DietPi-PREP finished + reboot) on early boot phase, the command is called as well, but not error handled, as it is a completely non-interactive boot phase anyway, no chance to ask for interactive input, we don't even know if a screen and keyboard is attached or not and most importantly boot must continue, otherwise headless systems would not gain network for SSH access. In DietPi-Config it is as well not error-handled. It could be, actually, but it is anyway no really essential command, it only works around a bug in Debian: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=818065 (**sigh**, still not fixed)

MichaIng, 2021-03-25:
I mark the issue as closed. Feel free to reopen if required.

closed 2021-03-25 by MichaIng