MichaIng/DietPi #3948: Add ProtonVPN like NordVPN

black00019, 2020-12-03:
### Creating a software request

#### Formal software information
- Software title | ProtonVPN
- Short description | Add protonvpn client and vpn gateway
- Official URL (if available) | https://protonvpn.com/blog/linux-vpn-cli-beta/
- (Official) install documentation | https://protonvpn.com/blog/linux-vpn-cli-beta/

#### Are there similar/alternative software titles available with DietPi-Software?
NordVPN
- ...

#### What makes your requested software better than the above solutions, if available?

- the vpn should also be the gateway like this:
https://gist.github.com/superjamie/ac55b6d2c080582a3e64

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. ...

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- ...

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

ravenclaw900, 2021-03-01:
@MichaIng, would this add a new `dietpi-protonvpn` option or expand `dietpi-nordvpn` into a more general-purpose `dietpi-vpn`?

MichaIng, 2021-03-01:
The latter I would say, as long as ProtonVPN similarly allows to download `.ovpn` config files to connect via OpenVPN.

ravenclaw900, 2021-03-01:
While it does, I was actually thinking a whiptail frontend for the [CLI](https://protonvpn.com/blog/linux-vpn-cli-beta/) instead.

MichaIng, 2021-03-01:
Since the CLI is OpenVPN-based as well, does it make something much better than ours?

> Ah, I see what you mean, that would allow better integration with the already existing script.

Yes, I was thinking of a generic OpenVPN client GUI that allows to download and/or use any present `.ovpn` config and enables some features, like up/down scripts, a killswitch and the ability to still allow remote connections to defined applications (both ToDo).

Long-term, WireGuard could be added, but first this requires native dynamic IP support. Currently WireGuard natively requires a fixed IP for every client, breaking privacy with public providers. So providers currently ship each their very own WireGuard wrapper implementation to address this.

ravenclaw900, 2021-03-04:
Okay, so the `.ovpn` files are downloaded from a `blob:` URL, meaning they can't be downloaded directly.  We could do what the CLI does and parse the (massive) JSON server list at https://api.protonvpn.ch/vpn/logicals, but `mawk` complains about its size:
```sh
awk: program limit exceeded: maximum number of fields size=32767
```

MichaIng, 2021-03-04:
Here is their config file template: https://github.com/ProtonVPN/linux-cli/blob/master/protonvpn_cli/templates/openvpn_template.j2

So we need "only" a list of servers IPs and their location. Ports for UDP and TCP are default. Okay yes found that JSON as well now: https://github.com/ProtonVPN/linux-cli/search?q=SERVER_INFO_FILE
Probably we can used `sed` to add linebreaks after each server block and then mawk only needs to parse the domain (as sort of name) and IP for one server per line. With the template we can then generate a .ovpn on demand.

Last resort: [jq](https://packages.debian.org/jq)

blogabe, 2021-03-05:
Thank you for your work on this!

closed 2021-04-24 by MichaIng