MichaIng/DietPi #3950: mpd can not open alsa device after configuring eq - plus several bugs with eq module

wowcut, 2020-12-03:
current release. raspberry pi 2, no special audio hardware. this is a pretty default dietpi, just installed.
mpd works with default settings, no problems.
But activating the equalizer for alsa via the dietpi console ui breaks mpd alsa playback.

```
exception: Failed to open "DietPi Alsa" (alsa)
exception: nested: Failed to open ALSA device "default": Operation not permitted

# /etc/asound.conf

ctl.eq {                                                                                                                                                                                                                                       
        type equal                                                                                                                                                                                                                             
        controls "/root/.alsaequal.bin"                                                                                                                                                                                                        
}                                                                                                                                                                                                                                              
                                                                                                                                                                                                                                               
pcm.plugequal {                                                                                                                                                                                                                                
        type equal                                                                                                                                                                                                                             
        slave.pcm "plughw:0,0"                                                                                                                                                                                                                 
        controls "/root/.alsaequal.bin"                                                                                                                                                                                                        
}                                                                                                                                                                                                                                              
                                                                                                                                                                                                                                               
pcm.!default {                                                                                                                                                                                                                                 
        type plug                                                                                                                                                                                                                              
        slave.pcm plugequal                                                                                                                                                                                                                    
        slave.channels 2                                                                                                                                                                                                                       
}                                                                                                                                                                                                                                              
                                                                                                                                                                                                                                               
ctl.!default {                                                                                                                                                                                                                                 
        type hw                                                                                                                                                                                                                                
        card 0                                                                                                                                                                                                                                 
}   

# /etc/mpd.conf
audio_output {                                                                                                                                                                                                                                 
                                                                                                                                                                                                                                               
type "alsa"                                                                                                                                                                                                                                    
name "DietPi Alsa"                                                                                                                                                                                                                             
#device "hw:0,0"                                                                                                                                                                                                                               
format "44100:16:2"                                                                                                                                                                                                                            
mixer_type "software"                                                                                                                                                                                                                          
#mixer_control "PCM"                                                                                                                                                                                                                           
#mixer_index "0"                                                                                                                                                                                                                               
                                                                                                                                                                                                                                               
# Disable DSD DoP MPD 0.19                                                                                                                                                                                                                     
dop "no"                                                                                                                                                                                                                                       
                                                                                                                                                                                                                                               
}      

```

[... cut ...]
BTW also it is not clear if after selecting one item it is still needed to select OK or not - looks like everything is installed over and over again when selecting ok - why?

Also it is not clear, why the eq setting is below that "boom" thing - from what I understand it is a setting that will work also without these "boom" devices - at least I was able to configure alsa eq before without any additional device - maybe the eq setting should be a top level audio setting... 

This is btw not a hardware problem - I had the alsa eq configured before on that device with a manual raspbian config.

MichaIng, 2020-12-03:
Many thanks for your report.

Since I was as well playing around with our audio settings in another context, I recognised that this equalizer setup is not great since it requires root permissions to read/write the config file in root's home directory: `/root/.alsaequal.bin`

Note that a year or two ago all software in DietPi was executed as root user. Luckily to satisfy any security recommendations this is not the case anymore. But at least this equalizer plugin setup was never adopted, especially since I am only slowly digging inside ALSA and especially DietPi-JustBoom and any change there requires immediate review of the Allo web GUI to assure nothing got broken there.

The concatenated `-eq` indeed was a stupid bug, which is resolved for upcoming v6.34 already: https://github.com/MichaIng/DietPi/commit/c898bb8ba06d923689dda4bc0bd89ded7fb22eec#diff-c014199cb377c65ea8a787bda4bd115820794ad3d4ba173b2538d2ebba40b1ffR94-R95

> BTW also it is not clear if after selecting one item it is still needed to select OK or not

This is some glitch in whiptail in general. In the menus, selecting "Ok" while having a menu entry highlighted has the exact same effect as selecting that menu entry directly. In both cases, you should see `[Off]` <> `[On]` switching and yes, whenever enabling it.

> Also it is not clear, why the eq setting is below that "boom" thing

I though exactly the same. I added a new auto-conversion plug option (see commit above) and added this to the main sound option. There is AFAIK nothing that depends on any specific sound device, these are all ALSA settings/commands + if installed MPD, CAVA which of course both do as well not depend on any specific audio hardware. I have the plan do move everything audio related into a new `dietpi-audio` script, which is then opened as sub-script when going to dietpi-config > Audio Options but can be called as standalone script as well, like `dietpi-justboom` currently (which misses sound card selection). Currently everything is split across three scripts: `dietpi-config`, `dietpi-justboom` and `dietpi-set_hardware` (as kinda backend with CLI instead of GUI). A general aim I have is to move similar topics (video, audio, cpu-related, network) into own scripts which serve as both, whiptail-based GUI scripts but as well with a CLI for scripting configs which is currently only possible via `dietpi-set_hardware`.

MichaIng, 2020-12-08:
Permissions issue fixed with: #3953

closed 2020-12-20 by MichaIng