MichaIng/DietPi #3952: "/var/cache/apt/archives/partial is missing" when installing Aria 2.

razkal, 2020-12-06:
i am trying to install Aria 2 and i am getting the below error report.
i also tried [the fix of a similar issue here](https://github.com/MichaIng/DietPi/issues/3851) but i am getting the below error:
_root@DietPi:~# mkdir /var/cache/apt/archives/partial
mkdir: cannot create directory ‘/var/cache/apt/archives/partial’: No such file or directory
root@DietPi:~# cd /var/cache/apt/archives
-bash: cd: /var/cache/apt/archives: No such file or directory
root@DietPi:~# cd /var/cache/apt
root@DietPi:/var/cache/apt# ls
root@DietPi:/var/cache/apt# pwd
/var/cache/apt
root@DietPi:/var/cache/apt# mkdir /var/cache/apt/archives
mkdir: cannot create directory ‘/var/cache/apt/archives’: Structure needs cleaning_

### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
#### Details:
- Date           | Sun Dec  6 12:07:33 GMT 2020
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GN        U/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. installing Aria 2
#### Expected behaviour:

- Aria 2 installation through dietpi-software menu
#### Actual behaviour:

- [FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.
#### Extra details:

- ...
#### Additional logs:
```
E: Archives directory /var/cache/apt/archives/partial is missing. - Acquire (2: No such fil        e or directory)
```

Joulinar, 2020-12-06:
Hi,

many thanks for your message. pls can you try following

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

razkal, 2020-12-06:
Hi,

below the output:

root@DietPi:/var/cache/apt# /boot/dietpi/func/dietpi-set_software apt-cache clean
[ SUB1 ] DietPi-Set_software > apt-cache (clean)
[  OK  ] apt-cache clean | Completed
root@DietPi:/var/cache/apt# apt update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://apt.sonarr.tv master InRelease [13.3 kB]
Get:3 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Get:5 https://download.mono-project.com/repo/debian raspbianbuster InRelease [2238 B]
Get:6 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:7 https://apt.sonarr.tv master/main armhf Packages [67.9 kB]
Get:8 https://archive.raspberrypi.org/debian buster/main armhf Packages [346 kB]
Get:9 https://download.mono-project.com/repo/debian raspbianbuster/main armhf Packages [47.3 kB]
Get:10 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Get:11 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.7 kB]
Get:12 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages [12.9 MB]
Fetched 26.6 MB in 46s (582 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
6 packages can be upgraded. Run 'apt list --upgradable' to see them.
root@DietPi:/var/cache/apt# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  libcups2 libimobiledevice6 libjpeg62-turbo libsqlite3-0 libxml2 sqlite3
6 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 2422 kB of archives.
After this operation, 3072 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ftp.acc.umu.se/mirror/raspbian/raspbian buster/main armhf sqlite3 armhf 3.27.2-3+deb10u1 [839 kB]
Get:2 http://ftp.acc.umu.se/mirror/raspbian/raspbian buster/main armhf libsqlite3-0 armhf 3.27.2-3+deb10u1 [553 kB]
Get:3 http://ftp.acc.umu.se/mirror/raspbian/raspbian buster/main armhf libcups2 armhf 2.2.10-6+deb10u4 [287 kB]
Get:5 http://ftp.acc.umu.se/mirror/raspbian/raspbian buster/main armhf libjpeg62-turbo armhf 1:1.5.2-2+deb10u1 [112 kB]
Get:6 http://ftp.acc.umu.se/mirror/raspbian/raspbian buster/main armhf libxml2 armhf 2.9.4+dfsg1-7+deb10u1 [571 kB]
Get:4 http://mirror.as43289.net/raspbian/raspbian buster/main armhf libimobiledevice6 armhf 1.2.1~git20181030.92c5462-2+deb10u1 [59.4 kB]
Fetched 2422 kB in 4s (570 kB/s)
sh: 0: getcwd() failed: No such file or directory
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 109780 files and directories currently installed.)
Preparing to unpack .../sqlite3_3.27.2-3+deb10u1_armhf.deb ...
Unpacking sqlite3 (3.27.2-3+deb10u1) over (3.27.2-3) ...
Preparing to unpack .../libsqlite3-0_3.27.2-3+deb10u1_armhf.deb ...
Unpacking libsqlite3-0:armhf (3.27.2-3+deb10u1) over (3.27.2-3) ...
Setting up libsqlite3-0:armhf (3.27.2-3+deb10u1) ...
(Reading database ... 109780 files and directories currently installed.)
Preparing to unpack .../libcups2_2.2.10-6+deb10u4_armhf.deb ...
Unpacking libcups2:armhf (2.2.10-6+deb10u4) over (2.2.10-6+deb10u3) ...
Preparing to unpack .../libimobiledevice6_1.2.1~git20181030.92c5462-2+deb10u1_armhf.deb ...
Unpacking libimobiledevice6:armhf (1.2.1~git20181030.92c5462-2+deb10u1) over (1.2.1~git20181030.92c5462-2) ...
Preparing to unpack .../libjpeg62-turbo_1%3a1.5.2-2+deb10u1_armhf.deb ...
Unpacking libjpeg62-turbo:armhf (1:1.5.2-2+deb10u1) over (1:1.5.2-2+b1) ...
Preparing to unpack .../libxml2_2.9.4+dfsg1-7+deb10u1_armhf.deb ...
Unpacking libxml2:armhf (2.9.4+dfsg1-7+deb10u1) over (2.9.4+dfsg1-7+b3) ...
Setting up libjpeg62-turbo:armhf (1:1.5.2-2+deb10u1) ...
Setting up libcups2:armhf (2.2.10-6+deb10u4) ...
Setting up libimobiledevice6:armhf (1.2.1~git20181030.92c5462-2+deb10u1) ...
Setting up libxml2:armhf (2.9.4+dfsg1-7+deb10u1) ...
Setting up sqlite3 (3.27.2-3+deb10u1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
root@DietPi:/var/cache/apt#

MichaIng, 2020-12-06:
> mkdir: cannot create directory ‘/var/cache/apt/archives’: Structure needs cleaning_

You have a file system error that prevents this directory from being created. Please run a file system check, either on an external system, or on reboot:
```
> /forcefsck
reboot
# after reboot
journalctl -t systemd-fsck
```

razkal, 2020-12-06:
> 
> 
> > mkdir: cannot create directory ‘/var/cache/apt/archives’: Structure needs cleaning_
> 
> You have a file system error that prevents this directory from being created. Please run a file system check, either on an external system, or on reboot:
> 
> ```
> > /forcefsck
> reboot
> # After reboot, check result
> journalctl -t systemd-fsck
> ```
> 
> **EDIT: Ah okay, now everything went through as it should. Still, I would definitely do the above to assure there is nothing left as file system corruption can lead to various unexpected errors later on.**

Hi,

i am using the external partition  /mnt/9A845F9F845F7CA7.
should i run the filesystem check as _>/mnt/9A845F9F845F7CA7/forcefsck_ ?

MichaIng, 2020-12-06:
Is `/var/cache/apt` an external drive mount or part of your root file system? It doesn't matter if your root file system is on an external drive or SD card, `> /forcefsck` forces a root file system check. If you mounted an external file system into your root file system hierarchy at `/var/cache/apt` or `/var/cache` or `/var`, then you can simply unmount it and run `e2fsck -f` on that file system outside of reboot.

razkal, 2020-12-06:
after reboot:

root@DietPi:~# journalctl -t systemd-fsck
-- Logs begin at Thu 2019-02-14 10:11:58 GMT, end at Sun 2020-12-06 12:49:02 GMT. --
Dec 06 12:46:06 DietPi systemd-fsck[128]: Please pass 'fsck.mode=force' on
the kernel command line rather than creating /forcefsck on the root file system.
Dec 06 12:46:06 DietPi systemd-fsck[128]: e2fsck 1.44.5 (15-Dec-2018)
Dec 06 12:46:06 DietPi systemd-fsck[128]: Pass 1: Checking inodes, blocks, and sizes
Dec 06 12:46:29 DietPi systemd-fsck[128]: Pass 2: Checking directory structure
Dec 06 12:46:34 DietPi systemd-fsck[128]: Pass 3: Checking directory connectivity
Dec 06 12:46:34 DietPi systemd-fsck[128]: Pass 4: Checking reference counts
Dec 06 12:46:35 DietPi systemd-fsck[128]: Pass 5: Checking group summary information
Dec 06 12:46:37 DietPi systemd-fsck[128]: rootfs: 129688/1864000 files (0.3% non-contiguous), 10907
51/7613440 blocks
Dec 06 12:46:41 DietPi systemd-fsck[264]: Please pass 'fsck.mode=force' on
the kernel command line rather than creating /forcefsck on the root file system.
Dec 06 12:46:41 DietPi systemd-fsck[264]: fsck.fat 4.1 (2017-01-24)
Dec 06 12:46:41 DietPi systemd-fsck[264]: /dev/mmcblk0p1: 317 files, 114493/516190 clusters
root@DietPi:~#

MichaIng, 2020-12-06:
Okay that looks all fine. So I guess the APT cache clean suggested by Joulinar solved the file system issue as well, that is great.

According to the two warnings, just in case you wondered:

> Please pass 'fsck.mode=force' on the kernel command line rather than creating /forcefsck on the root file system.

Using a method that requires manipulating the kernel cmdline, which is not possible easily on all boards, and, when any issue happens, can break boot? And after reboot, the same must be done again to not do a root fsck on every reboot, same issue, same risk + an additional service or boot script must be placed to remove the `fsck.mode=force` entry. Compared to placing a single file that cannot break anything and is removed automatically by either initramfs fsck or systemd-fsck? I really hope systemd either un-deprecates this easy, safe and effective method of forcing a single root fsck check on next reboot only, or adds a similar one. Although why changing methods when there is one that has no downsides I can think of...

Joulinar, 2020-12-06:
@razkal 
pls can you restart your Aria2 installation

razkal, 2020-12-06:
nginx and aria 2 were installed!
thanks for your prompt support guys!

razkal, 2020-12-06:
in aria 2 web interface, i use therpc secret password  "dietpi" and i get the below error:

"Oh Snap! Authentication failed while connecting to Aria2 RPC server. Will retry in 10 secs. You might want to confirm your authentication details by going to Settings > Connection Settings"

is this normal?
[image]

ravenclaw900, 2020-12-06:
I think that you either need to leave out the password, or put in a username.

razkal, 2020-12-06:
i see there is a [similar issue ](https://github.com/aria2/aria2/issues/1009) open.

razkal, 2020-12-06:
in the aria settings:
# Connection
listen-port=6881-6999
ftp-pasv=true
check-certificate=false

listen port in web gui is 6800 though.
may i need to modify the above ports?

razkal, 2020-12-06:
solved!
instead of "dietpi", you need to use your ssh password as as secret token.

closed 2020-12-06 by razkal

MichaIng, 2020-12-06:
That part of the config is responsible for web access: `rpc-listen-port=6800`
That part for the password that is set by default on DietPi to the global software password chosen on first boot of the fresh system or via `dietpi.txt`. See `rpc-secret=<your global password>` in the config file.
Hmm, would be great if it could be stored hashed there 🤔.