MichaIng/DietPi #3954: Please create a ticket: https://github.com/MichaIng/DietPi/issues

Franky-star, 2020-12-06:
#### Details:
- Date           | Sun Dec  6 22:13:44 UTC 2020
- DietPi version | v6.29.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi B (armv6l) (ID=1)
- Kernel version | Linux DietPi 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq --allow-change-held-packages install mariadb-server
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
debconf: DbDriver "config": Error setting up format object 822: Unrecognized character ▒; marked by <-- HERE after nf::FormaT<-- HERE near column 24 at /usr/share/perl5/Debconf/Format/822.pm line 5, <DEBCONF_CONFIG> chunk 3.
Compilation failed in require at (eval 15) line 1, <DEBCONF_CONFIG> chunk 3.
BEGIN failed--compilation aborted at (eval 15) line 1, <DEBCONF_CONFIG> chunk 3.

(Reading database ... 25142 files and directories currently installed.)
Preparing to unpack .../mariadb-server-10.3_1%3a10.3.27-0+deb10u1_armhf.deb ...
debconf: DbDriver "config": Error setting up format object 822: Unrecognized character ▒; marked by <-- HERE after nf::FormaT<-- HERE near column 24 at /usr/share/perl5/Debconf/Format/822.pm line 5, <DEBCONF_CONFIG> chunk 3.
Compilation failed in require at (eval 14) line 1, <DEBCONF_CONFIG> chunk 3.
BEGIN failed--compilation aborted at (eval 14) line 1, <DEBCONF_CONFIG> chunk 3.

dpkg: error processing archive /var/cache/apt/archives/mariadb-server-10.3_1%3a10.3.27-0+deb10u1_armhf.deb (--unpack):
 new mariadb-server-10.3 package pre-installation script subprocess returned error exit status 1
debconf: DbDriver "config": Error setting up format object 822: Unrecognized character ▒; marked by <-- HERE after nf::FormaT<-- HERE near column 24 at /usr/share/perl5/Debconf/Format/822.pm line 5, <DEBCONF_CONFIG> chunk 3.
Compilation failed in require at (eval 14) line 1, <DEBCONF_CONFIG> chunk 3.
BEGIN failed--compilation aborted at (eval 14) line 1, <DEBCONF_CONFIG> chunk 3.

dpkg: error while cleaning up:
 new mariadb-server-10.3 package post-removal script subprocess returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/mariadb-server-10.3_1%3a10.3.27-0+deb10u1_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2020-12-06:
Many thanks for your report.

I think the debconf perl module is corrupted. Can you please paste the output of:
```
ls -l /usr/share/perl5/Debconf/Format/822.pm
dmesg -l emerg,alert,crit,err
```

Franky-star, 2020-12-06:
root@DietPi:~# ls -l /usr/share/perl5/Debconf/Format/822.pm
-rw-r--r-- 1 root root 2139 Feb 26  2019 /usr/share/perl5/Debconf/Format/822.pm
root@DietPi:~# dmesg -l emerg,alert,crit,err
[    0.932147] vc_vchi_sm_init: failed to open VCHI service (-1)
[    0.932161] [vc_sm_connected_init]: failed to initialize shared memory service
[   15.085278] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[   15.085293] [vc_sm_connected_init]: failed to initialize shared memory service
[14622590.325598] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
du: iget: bad extra_isize 16416 (inode size 256)
[14622590.370468] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
du: iget: bad extended attribute block 512
[14622600.659563] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14622600.667761] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14622648.068564] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14622649.294515] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14623854.415825] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
du: iget: bad extra_isize 16416 (inode size 256)
[14623854.923410] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
du: iget: bad extended attribute block 512
[14623859.399753] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14623859.410677] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14623869.613040] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14623869.656413] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14624140.840222] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
du: iget: bad extra_isize 16416 (inode size 256)
[14624141.361644] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
du: iget: bad extended attribute block 512
[14624143.060263] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14624143.084998] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14624148.512636] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14624148.547899] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14624341.648687] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
du: iget: bad extra_isize 16416 (inode size 256)
[14624341.672770] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
du: iget: bad extended attribute block 512
[14624343.320311] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14624343.345032] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14624348.715989] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14624348.751311] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14629958.380265] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
du: iget: bad extra_isize 16416 (inode size 256)
[14629958.881464] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
du: iget: bad extended attribute block 512
[14629967.321562] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14629967.330844] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14630122.440895] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14630122.893301] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
root@DietPi:~# —
                                                                                          
                                                                                          
[... 2 lines cut ...]
                                                                                          
                                                                                          
                                                                                          
                                                                                          
                                                                                          
                                                                                          
                                                                                          
[... cut ...]
                                                                                          
                                                                                          
                                                                                          
                                                                                          
                                                                                          
                                                                                          

Franky-star, 2020-12-06:
root@DietPi:~# ls -l /usr/share/perl5/Debconf/Format/822.pm
-rw-r--r-- 1 root root 2139 Feb 26  2019 /usr/share/perl5/Debconf/Format/822.pm

dmesg -l emerg,alert,crit,err
[    0.932147] vc_vchi_sm_init: failed to open VCHI service (-1)
[    0.932161] [vc_sm_connected_init]: failed to initialize shared memory service
[   15.085278] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[   15.085293] [vc_sm_connected_init]: failed to initialize shared memory service
[14622590.325598] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
du: iget: bad extra_isize 16416 (inode size 256)
[14622590.370468] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
du: iget: bad extended attribute block 512
[14622600.659563] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14622600.667761] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14622648.068564] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14622649.294515] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14623854.415825] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
du: iget: bad extra_isize 16416 (inode size 256)
[14623854.923410] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
du: iget: bad extended attribute block 512
[14623859.399753] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14623859.410677] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14623869.613040] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14623869.656413] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14624140.840222] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
du: iget: bad extra_isize 16416 (inode size 256)
[14624141.361644] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
du: iget: bad extended attribute block 512
[14624143.060263] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14624143.084998] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14624148.512636] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14624148.547899] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14624341.648687] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
du: iget: bad extra_isize 16416 (inode size 256)
[14624341.672770] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
du: iget: bad extended attribute block 512
[14624343.320311] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14624343.345032] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14624348.715989] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14624348.751311] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14629958.380265] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
du: iget: bad extra_isize 16416 (inode size 256)
[14629958.881464] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
du: iget: bad extended attribute block 512
[14629967.321562] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14629967.330844] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
[14630122.440895] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630095: comm
rsync: iget: bad extra_isize 16416 (inode size 256)
[14630122.893301] EXT4-fs error (device mmcblk0p2): ext4_lookup:1590: inode #630093: comm
rsync: iget: bad extended attribute block 512
root@DietPi:~# s -l /usr/share/perl5/Debconf/Format/822.pm
-bash: s: command not found
[... 4 lines cut ...]
                                                                                          
                                                                                          
                                                                                          
                                                                                          
                                                                                          
                                                                                          
                                                                                          
                                                                                          
[... cut ...]
                                                                                          
                                                                                          
                                                                                          
                                                                                          
                                                                                          
                                                                                          

MichaIng, 2020-12-06:
Okay, meta data of that file look all fine. It is plain text, so you can do `cat /usr/share/perl5/Debconf/Format/822.pm` to see that perl scripts content.

However, the `dmesg` output verifies that there is a file system error, luckily only on two inodes (files). Since the commands `du` and `rsync` are involved. Did you do or tried a `dietpi-backup` or `dietpi-sync` recently?

I would start with a root file system check on reboot, then checking the result:
```
> /forcefsck
reboot
# after reboot
journalctl -t systemd-fsck
```

Franky-star, 2020-12-07:
Dear Michalng, 
When I enterd > /forcefsck
reboot, unfortently ssh connection failed. I going to connect the PI with my PC monitor. If i am have some news about the status of my PI, i going to sen you some news

best reagards

MichaIng, 2020-12-07:
Sorry to hear, let's hope it is quickly solved. It's a pain with file system corruption, reminds of mould: When you see it the first time, it's already everywhere 😄.

Franky-star, 2020-12-13:
Dear Michalng, I had tried to fix the Issue on my pi, but it failed. I decided to reinstall the dietpi software, because it takes less time. Thanks for your help and your good work.

closed 2020-12-13 by Franky-star