MichaIng/DietPi #3957: DietPi-Software | Box86

ravenclaw900, 2020-12-08:
### Creating a software request

#### Formal software information
- Software title | 
Box86
- Short description | 
x86 userspace emulation
- GitHub URL (if available) | 
https://github.com/ptitSeb/box86
- (Official) install documentation | 

#### Are there similar/alternative software titles available with DietPi-Software?

- No

#### What makes your requested software better than the above solutions, if available?

- N/A

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. Build/Install

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- Probably

MichaIng, 2020-12-15:
Nice, @meveric has produced Box86 builds for Odroids available via his APT repository: https://dietpi.com/meveric/pool/main/b/
So on Odroids it is very easy to implement then, while for other SoCs we'd need to either build own packages or compile on demand on install.

meveric, 2020-12-15:
Box86 doesn't have much dependencies itself.
It does just calculations on the CPU.
It requires mainly x86 libraries to work with and armhf libraries that are wrapped (uses native armhf where it can, instead of emulated x86 libraries).

I package it currently mainly for Debian Buster, but it works on other OS as well (Debian Bullseye).
As the package itself has no "real" dependencies, it should work on all boards, as long as it supports armhf.

**Please note:** I currently use libgl-odroid (gl4es) as a hard dependency to make sure it's installed, and use it as the default for hardware acceleration. It can be overwritten with MESA=1 as an environment variable, but it will still require libgl-odroid to be installed (not sure if I want to change this or not).

MichaIng, 2020-12-16:
> As the package itself has no "real" dependencies, it should work on all boards, as long as it supports armhf.

That is awesome.

> Please note: I currently use libgl-odroid (gl4es) as a hard dependency to make sure it's installed, and use it as the default for hardware acceleration.

As long as other applications can still use regular GLES/EGL that should not be an issue.

ravenclaw900, 2020-12-20:
To compile, you have to have `cmake`, `make (build-essentials),` and `python3`.  Annoying, since `cmake` isn't included in `build-essentials`.

MichaIng, 2020-12-21:
Not a big issue to install additional APT packages as build dependencies, we do that in many other cases as well. Only "unclean" is that on uninstall we need to leave it, as we cannot know whether it is required or wanted for other things.

closed 2021-09-16 by MichaIng

RPTST, 2021-10-01:
#### Details:
- Date           | Fri Oct  1 09:56:29 EDT 2021
- Bug report     | 062c0eca-085c-4b46-bf2d-2eea87c4f90b
- DietPi version | v7.6.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | From scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux dietpi 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `systemctl restart systemd-binfmt`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Job for systemd-binfmt.service failed because the control process exited with error code.
See "systemctl status systemd-binfmt.service" and "journalctl -xe" for details.
```

MichaIng, 2021-10-01:
Many thanks for reporting. Please open a new issue about this, as usually discussion in closed issues tend to be overseen. Also please append the output of the following command to the new issue:
```sh
journalctl -u systemd-binfmt
```