MichaIng/DietPi #3958: dietpi-backup kills home assistant docker installation

AaronDavidSchneider, 2020-12-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.33.3 beta`
- Distro version | `buster`
- Kernel version | `Linux DietPi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)`
- Power supply used | `irrelevant`
- SDcard used | `irrelevant`

#### Additional Information (if applicable)
- Software title | `dietpi-backup` and `homeassistant`
- Was the software title installed freshly or updated/migrated? freshly
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Steps to reproduce

1. install home assistant using the supervised installer: https://github.com/home-assistant/supervised-installer 
2. start `dietpi-backup`

#### Expected behaviour

1. `dietpi-backup` should stop home assistant (it does that)
2. `dietpi-backup` should start home assistant (it does that) when the backup completed

#### Actual behaviour

1. `dietpi-backup` stops home assistant 
2. `dietpi-backup` does not start home assistant once completed

#### Extra details
Logs: <details>
```bash
Dez 08 01:25:01 DietPi CRON[19205]: pam_unix(cron:session): session opened for user root by (uid=0)
Dez 08 01:25:01 DietPi CRON[19206]: (root) CMD (test -x /usr/sbin/anacron || { cd / && run-parts --report /etc/cron.daily; })
Dez 08 01:25:05 DietPi sudo[19228]:     root : TTY=unknown ; PWD=/tmp/DietPi-Services ; USER=www-data ; COMMA
ND=/usr/bin/php /var/www/nextcloud/occ maintenance:mode --on
Dez 08 01:25:05 DietPi sudo[19228]: pam_unix(sudo:session): session opened for user www-data by (uid=0)
Dez 08 01:25:05 DietPi sudo[19228]: pam_unix(sudo:session): session closed for user www-data
Dez 08 01:25:05 DietPi systemd[1]: Stopping Regular background program processing daemon...
Dez 08 01:25:05 DietPi systemd[1]: cron.service: Main process exited, code=killed, status=15/TERM
Dez 08 01:25:05 DietPi systemd[1]: cron.service: Succeeded.
Dez 08 01:25:05 DietPi systemd[1]: Stopped Regular background program processing daemon.
Dez 08 01:25:05 DietPi systemd[1]: Stopping Hass.io supervisor...
Dez 08 01:25:05 DietPi d131db60880a[665]: [cont-finish.d] executing container finish scripts...
Dez 08 01:25:05 DietPi hassio-supervisor[1234]: [cont-finish.d] executing container finish scripts...
Dez 08 01:25:05 DietPi d131db60880a[665]: [cont-finish.d] done.
Dez 08 01:25:05 DietPi hassio-supervisor[1234]: [cont-finish.d] done.
Dez 08 01:25:05 DietPi d131db60880a[665]: [s6-finish] waiting for services.
Dez 08 01:25:05 DietPi hassio-supervisor[1234]: [s6-finish] waiting for services.
Dez 08 01:25:05 DietPi d131db60880a[665]: 20-12-08 00:25:05 INFO (MainThread) [supervisor.api] Stopping API o
n 172.30.32.2
Dez 08 01:25:05 DietPi hassio-supervisor[1234]: 20-12-08 00:25:05 INFO (MainThread) [supervisor.api] Stopping API on 172.30.32.2
Dez 08 01:25:05 DietPi d131db60880a[665]: 20-12-08 00:25:05 INFO (MainThread) [supervisor.misc.scheduler] Shu
tting down scheduled tasks
Dez 08 01:25:05 DietPi hassio-supervisor[1234]: 20-12-08 00:25:05 INFO (MainThread) [supervisor.misc.scheduler] Shutting down schedul
ed tasks
Dez 08 01:25:05 DietPi d131db60880a[665]: 20-12-08 00:25:05 INFO (MainThread) [supervisor.misc.hwmon] Stopped
 Supervisor hardware monitor
Dez 08 01:25:05 DietPi hassio-supervisor[1234]: 20-12-08 00:25:05 INFO (MainThread) [supervisor.misc.hwmon] Stopped Supervisor hardwa
re monitor
Dez 08 01:25:05 DietPi d131db60880a[665]: 20-12-08 00:25:05 INFO (MainThread) [supervisor.core] Supervisor is
 down - 0
Dez 08 01:25:05 DietPi hassio-supervisor[1234]: 20-12-08 00:25:05 INFO (MainThread) [supervisor.core] Supervisor is down - 0
Dez 08 01:25:05 DietPi d131db60880a[665]: 20-12-08 00:25:05 INFO (MainThread) [__main__] Closing Supervisor
m
Dez 08 01:25:05 DietPi hassio-supervisor[1234]: 20-12-08 00:25:05 INFO (MainThread) [__main__] Closing Supervisor
Dez 08 01:25:06 DietPi d131db60880a[665]: [s6-finish] sending all processes the TERM signal.
Dez 08 01:25:06 DietPi hassio-supervisor[1234]: [s6-finish] sending all processes the TERM signal.
Dez 08 01:25:09 DietPi d131db60880a[665]: [s6-finish] sending all processes the KILL signal and exiting.
[... 345 lines cut ...]
Dez 08 01:25:18 DietPi systemd[1]: Stopping Gitea (Git with a cup of tea)...
Dez 08 01:25:19 DietPi systemd[1]: gitea.service: Succeeded.
Dez 08 01:25:19 DietPi systemd[1]: Stopped Gitea (Git with a cup of tea).
Dez 08 01:25:19 DietPi systemd[1]: /etc/systemd/system/gitea.service:14: Unknown lvalue 'HOME' in section 'Se
rvice', ignoring
Dez 08 01:25:19 DietPi systemd[1]: /etc/systemd/system/gitea.service:14: Unknown lvalue 'HOME' in section 'Se
rvice', ignoring
Dez 08 01:25:19 DietPi systemd[1]: Stopping The Apache HTTP Server...
Dez 08 01:25:19 DietPi apachectl[19919]: AH00558: apache2: Could not reliably determine the server's fully qualified domain name, usi
ng 127.0.1.1. Set the 'ServerName' directive globally to suppress this message
Dez 08 01:25:19 DietPi systemd[1]: apache2.service: Succeeded.
Dez 08 01:25:19 DietPi systemd[1]: Stopped The Apache HTTP Server.
Dez 08 01:25:19 DietPi systemd[1]: Stopping Advanced key-value store...
Dez 08 01:25:20 DietPi systemd[1]: redis-server.service: Succeeded.
Dez 08 01:25:20 DietPi systemd[1]: Stopped Advanced key-value store.
Dez 08 01:25:20 DietPi systemd[1]: Stopping LSB: Starts ProFTPD daemon...
Dez 08 01:25:20 DietPi proftpd[19930]: Stopping ftp server: proftpd.
Dez 08 01:25:20 DietPi systemd[1]: proftpd.service: Succeeded.
Dez 08 01:25:20 DietPi systemd[1]: Stopped LSB: Starts ProFTPD daemon.
Dez 08 01:26:17 DietPi sshd[19974]: Received disconnect from 218.93.208.28 port 43080:11:  [preauth]
Dez 08 01:26:17 DietPi sshd[19974]: Disconnected from 218.93.208.28 port 43080 [preauth]
```
</details>

I am sorry if this is the wrong place for this issue, since it is probably not a bug but rather a help request...

Joulinar, 2020-12-08:
Hi,

many thanks for your message. Just for my understanding. You have HA running inside Docker? If yes, Docker is correctly restarted once backup competed?

AaronDavidSchneider, 2020-12-08:
yes that's correct! I suppose docker is correctly restarted. How can I check that?

Joulinar, 2020-12-08:
using `dietpi-services status` will show all DietPi managed services and their status

But I guess your HA container is not set to restart automatically once Docker is started. Btw how do you start the HA container once backup finished?

AaronDavidSchneider, 2020-12-08:
> But I guess your HA container is not set to restart automatically once Docker is started. Btw how do you start the HA container once backup finished?

That makes sense! I assumed that it should do so, since it automatically starts after reboot.
I have know linked it in the `After` section of the `docker.service` file and it works now.

closed 2020-12-08 by AaronDavidSchneider