MichaIng/DietPi #3961: certificate error while trying to install new optimized software

manilx, 2020-12-08:
#### Details:
- Date           | Tue Dec  8 18:04:57 WET 2020
- Bug report     | 3709cbe9-41b6-436e-b393-d86a27a12a97
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux piHole 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:
Selected Fail2Ban and started install.
#### Expected behaviour:
Install and run software.
- ...
#### Actual behaviour:
Fails with certificate warning

#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://downloads.plex.tv/repo/deb public InRelease
Ign:3 https://download.webmin.com/download/repository sarge InRelease
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Err:5 https://download.webmin.com/download/repository sarge Release
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification. [IP: 108.60.199.109 443]
Reading package lists...
E: The repository 'https://download.webmin.com/download/repository sarge Release' no longer has a Release file.

Joulinar, 2020-12-08:
Hi,

Many thanks for your message. Looks like an issue at webmin side as the https certificate is not valid. I guess nothing we can do from our side. However you could try changing from https to http on apt source file

manilx, 2020-12-08:
Hi

this was fast! Thx for the info.

Can you give me the commands tu run, please?

Thx in advance

Joulinar, 2020-12-08:
You can have a look into directory `/apt/sources.list.d`. This should contain a webmin file. Just open it with `nano` and change from `https` to `http`. Once done do `apt update && apt upgrade `

manilx, 2020-12-08:
Can't find apt in root folder.....
[image]

manilx, 2020-12-08:
found it in /etc ....

Joulinar, 2020-12-08:
Sorry yes a typo

manilx, 2020-12-08:
Worked!!  Thanks a lot.

ravenclaw900, 2020-12-08:
Strange, it is a valid certificate (issued by LetsEncrypt).

manilx, 2020-12-08:
This started today in the afternoon (GMT).... Has been fine in the morning and before (I have been testing a few installs and all went well). So it has to be on their end somehow.

Joulinar, 2020-12-08:
just for completeness, I can confirm the issue at webmin site on a short test

```
[ INFO ] DietPi-Software | APT install for: webmin, please wait...
E: Failed to fetch https://download.webmin.com/download/repository/pool/contrib/w/webmin/webmin_1.962_all.deb  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification. [IP: 108.60.199.109 443]
```

which is strange as the certificate is valid
```
root@DietPi3:~# openssl s_client -showcerts -connect https://download.webmin.com
1996255248:error:2008F002:BIO routines:BIO_lookup_ex:system lib:../crypto/bio/b_addr.c:724:Servname not supported for ai_socktype
connect:errno=2
root@DietPi3:~# openssl s_client -showcerts -connect download.webmin.com:443
CONNECTED(00000003)
depth=2 O = Digital Signature Trust Co., CN = DST Root CA X3
verify return:1
depth=1 C = US, O = Let's Encrypt, CN = Let's Encrypt Authority X3
verify return:1
depth=0 CN = download.webmin.com
verify return:1
---
Certificate chain
 0 s:CN = download.webmin.com
   i:C = US, O = Let's Encrypt, CN = Let's Encrypt Authority X3
-----BEGIN CERTIFICATE-----
...
-----END CERTIFICATE-----
 1 s:C = US, O = Let's Encrypt, CN = Let's Encrypt Authority X3
   i:O = Digital Signature Trust Co., CN = DST Root CA X3
-----BEGIN CERTIFICATE-----
...
-----END CERTIFICATE-----
---
Server certificate
subject=CN = download.webmin.com

issuer=C = US, O = Let's Encrypt, CN = Let's Encrypt Authority X3

---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: RSA
Server Temp Key: ECDH, P-256, 256 bits
---
SSL handshake has read 3285 bytes and written 437 bytes
Verification: OK
---
New, TLSv1.2, Cipher is ECDHE-RSA-AES256-GCM-SHA384
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES256-GCM-SHA384
    Session-ID: 8C0F0198D94D2145A208DF6C7923F7AF4CCBF3F034421C4267D14F5CABDD2C36
    Session-ID-ctx:
    Master-Key: 53C52CC8289545B74E55EBE3A7ABD66249AA35955B3F9BE08754110EBB42C1622A349455A452B57B060E49065013B883
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 7200 (seconds)
    TLS session ticket:
    0000 - ef 27 de 9d 7c 28 ae f3-dd 0d 5a 98 a6 5d e6 a1   .'..|(....Z..]..
    0010 - 76 0f 1c f6 55 ac 66 e9-7e ab bf a4 62 87 52 61   v...U.f.~...b.Ra
    0020 - 31 09 61 35 24 16 6a 6c-58 df 94 de 1e 16 6c cc   1.a5$.jlX.....l.
    0030 - 78 c3 ea 87 d8 bb 4b 02-29 4e 91 a3 3f 23 e0 e9   x.....K.)N..?#..
    0040 - 27 f1 c4 ae 4a 54 88 a8-71 68 d4 91 95 0d 53 18   '...JT..qh....S.
    0050 - 8b 06 51 e9 f8 c1 35 a9-0c 04 aa 9b 4a 01 55 7a   ..Q...5.....J.Uz
    0060 - e8 2b fe d0 41 72 e0 ce-05 aa 76 14 3f 01 aa 0e   .+..Ar....v.?...
    0070 - 61 49 87 7d 53 a1 53 48-1b b7 01 95 a4 a7 89 65   aI.}S.SH.......e
    0080 - 1b 2f 2e 21 1e d4 5d d7-0b 8d ed 15 43 11 ff 60   ./.!..].....C..`
    0090 - 2a 5d 76 a6 af 63 0b 1d-fa a3 dd ae cc c1 17 55   *]v..c.........U
    00a0 - 2d ac ac ca f2 8c a8 96-a5 63 0e 85 2f 52 69 cb   -........c../Ri.
    00b0 - 7e b7 5d 85 c3 d0 25 71-d6 96 ae c8 01 82 d2 59   ~.]...%q.......Y
    00c0 - f8 40 7a 57 3a 2f 96 d9-82 a7 70 b4 5c b9 86 23   .@zW:/....p.\..#

    Start Time: 1607465737
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
    Extended master secret: no
---
read:errno=0
root@DietPi3:~#
```

MichaIng, 2020-12-08:
In my browser it is as well shown as valid. APT on Bullseye fails as well, so it's not due to outdated trusted CA list or such 🤔.

Also it's Let's Encrypt, so well known certificate chain. I remember they aim to migrate to an own root certificate, but that should only apply for new ones. Let me check when this is done.

MichaIng, 2020-12-09:
Guys, it is not the HTTPS certificate, but the APT key.

@manilx 
Did you install webmin earlier, or did the error happen while you were installing webmin?
Can you please check if the APT gpg key exists?
```
ls -l /etc/apt/trusted.gpg.d/dietpi-webmin.gpg
```
In case do:
```
curl -sSLf 'http://www.webmin.com/jcameron-key.asc' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-webmin.gpg --yes
```
and switch back to https for that repo.

Joulinar, 2020-12-09:
I did a test an an empty system. So it was a fresh installation

MichaIng, 2020-12-09:
I was did a quick and dirty test and added only the repository without the key which lead to the error you faces. Then I added the key and everything went fine 🤔. I'll also try an install now:
```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Webmin: web interface system management

[  OK  ] DietPi-Software | Checking URL: http://www.webmin.com/jcameron-key.asc
[  OK  ] DietPi-Software | eval curl -sSLf 'http://www.webmin.com/jcameron-key.asc' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-webmin.gpg --yes
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Hit:4 https://deb.debian.org/debian buster-backports InRelease
Ign:5 https://download.webmin.com/download/repository sarge InRelease
Get:6 https://download.webmin.com/download/repository sarge Release [16.9 kB]
Get:7 https://download.webmin.com/download/repository sarge Release.gpg [173 B]
Get:8 https://download.webmin.com/download/repository sarge/contrib amd64 Packages [1379 B]
Fetched 18.4 kB in 5s (3562 B/s)
Reading package lists...
[  OK  ] DietPi-Software | APT update
```
all fine 🤔.

Joulinar, 2020-12-09:
yes correct. apt was fine pulling the package but installation of .deb file is failing.

MichaIng, 2020-12-09:
It is trusted the one second and not trusted the other second:
```
2020-12-09 01:23:13 root@VM-Buster:~# apt install webmin
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libauthen-pam-perl libgdbm-compat4 libgdbm6 libglib2.0-0 libicu63 libio-pty-perl libnet-ssleay-perl libperl5.28 libxml2 perl perl-modules-5.28 perl-openssl-defaults shared-mime-info
Suggested packages:
  gdbm-l10n perl-doc libterm-readline-gnu-perl | libterm-readline-perl-perl make libb-debug-perl liblocale-codes-perl
Recommended packages:
  libglib2.0-data xdg-user-dirs netbase
The following NEW packages will be installed:
  libauthen-pam-perl libgdbm-compat4 libgdbm6 libglib2.0-0 libicu63 libio-pty-perl libnet-ssleay-perl libperl5.28 libxml2 perl perl-modules-5.28 perl-openssl-defaults shared-mime-info webmin
0 upgraded, 14 newly installed, 0 to remove and 0 not upgraded.
Need to get 29.3 MB/47.8 MB of archives.
After this operation, 400 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://download.webmin.com/download/repository sarge/contrib amd64 webmin all 1.962 [29.3 MB]
88% [1 webmin 22.9 MB/29.3 MB 78%]                                                                                                                                                                                              2500 kB/s 2s^C
2020-12-09 01:25:05 root@VM-Buster:~# apt-get -qq --allow-change-held-packages install webmin
E: Failed to fetch https://download.webmin.com/download/repository/pool/contrib/w/webmin/webmin_1.962_all.deb  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification. [IP: 104.207.151.13 443]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
2020-12-09 01:25:20 root@VM-Buster:~# apt-get -qq --allow-change-held-packages install webmin
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package perl-modules-5.28.
(Reading database ... 14973 files and directories currently installed.)
Preparing to unpack .../00-perl-modules-5.28_5.28.1-6+deb10u1_all.deb ...
Unpacking perl-modules-5.28 (5.28.1-6+deb10u1) ...
Selecting previously unselected package libgdbm6:amd64.
Preparing to unpack .../01-libgdbm6_1.18.1-4_amd64.deb ...
Unpacking libgdbm6:amd64 (1.18.1-4) ...
```

Joulinar, 2020-12-09:
now working on my side

root@DietPi3:~# apt install webmin
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libauthen-pam-perl libgdbm-compat4 libgdbm6 libglib2.0-0 libicu63 libio-pty-perl libnet-ssleay-perl libperl5.28 libxml2 perl perl-modules-5.28
  perl-openssl-defaults shared-mime-info
Suggested packages:
  gdbm-l10n perl-doc libterm-readline-gnu-perl | libterm-readline-perl-perl make libb-debug-perl liblocale-codes-perl
Recommended packages:
  libglib2.0-data xdg-user-dirs netbase
The following NEW packages will be installed:
  libauthen-pam-perl libgdbm-compat4 libgdbm6 libglib2.0-0 libicu63 libio-pty-perl libnet-ssleay-perl libperl5.28 libxml2 perl perl-modules-5.28
  perl-openssl-defaults shared-mime-info webmin
0 upgraded, 14 newly installed, 0 to remove and 0 not upgraded.
Need to get 29.3 MB/46.4 MB of archives.
After this operation, 391 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://download.webmin.com/download/repository sarge/contrib armhf webmin all 1.962 [29.3 MB]
Fetched 29.3 MB in 11s (2778 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package perl-modules-5.28.
(Reading database ... 21146 files and directories currently installed.)
Preparing to unpack .../00-perl-modules-5.28_5.28.1-6_all.deb ...
Unpacking perl-modules-5.28 (5.28.1-6) ...
Selecting previously unselected package libgdbm6:armhf.
Preparing to unpack .../01-libgdbm6_1.18.1-4_armhf.deb ...
Unpacking libgdbm6:armhf (1.18.1-4) ...
Selecting previously unselected package libgdbm-compat4:armhf.
Preparing to unpack .../02-libgdbm-compat4_1.18.1-4_armhf.deb ...
Unpacking libgdbm-compat4:armhf (1.18.1-4) ...
Selecting previously unselected package libperl5.28:armhf.
Preparing to unpack .../03-libperl5.28_5.28.1-6_armhf.deb ...
Unpacking libperl5.28:armhf (5.28.1-6) ...
Selecting previously unselected package perl.
Preparing to unpack .../04-perl_5.28.1-6_armhf.deb ...
Unpacking perl (5.28.1-6) ...
Selecting previously unselected package perl-openssl-defaults:armhf.
Preparing to unpack .../05-perl-openssl-defaults_3_armhf.deb ...
Unpacking perl-openssl-defaults:armhf (3) ...
Selecting previously unselected package libnet-ssleay-perl.
Preparing to unpack .../06-libnet-ssleay-perl_1.85-2+b1_armhf.deb ...
Unpacking libnet-ssleay-perl (1.85-2+b1) ...
Selecting previously unselected package libauthen-pam-perl.
Preparing to unpack .../07-libauthen-pam-perl_0.16-3+b5_armhf.deb ...
Unpacking libauthen-pam-perl (0.16-3+b5) ...
Selecting previously unselected package libio-pty-perl.
Preparing to unpack .../08-libio-pty-perl_1%3a1.08-1.1+b4_armhf.deb ...
Unpacking libio-pty-perl (1:1.08-1.1+b4) ...
Selecting previously unselected package libglib2.0-0:armhf.
Preparing to unpack .../09-libglib2.0-0_2.58.3-2+deb10u2_armhf.deb ...
Unpacking libglib2.0-0:armhf (2.58.3-2+deb10u2) ...
Selecting previously unselected package libicu63:armhf.
Preparing to unpack .../10-libicu63_63.1-6+deb10u1_armhf.deb ...
Unpacking libicu63:armhf (63.1-6+deb10u1) ...
Selecting previously unselected package libxml2:armhf.
Preparing to unpack .../11-libxml2_2.9.4+dfsg1-7+deb10u1_armhf.deb ...
Unpacking libxml2:armhf (2.9.4+dfsg1-7+deb10u1) ...
Selecting previously unselected package shared-mime-info.
Preparing to unpack .../12-shared-mime-info_1.10-1_armhf.deb ...
Unpacking shared-mime-info (1.10-1) ...
Setting up perl-modules-5.28 (5.28.1-6) ...
Setting up libgdbm6:armhf (1.18.1-4) ...
Setting up libgdbm-compat4:armhf (1.18.1-4) ...
Setting up libperl5.28:armhf (5.28.1-6) ...
Setting up perl (5.28.1-6) ...
Selecting previously unselected package webmin.
(Reading database ... 23401 files and directories currently installed.)
Preparing to unpack .../archives/webmin_1.962_all.deb ...
Unpacking webmin (1.962) ...
Setting up libio-pty-perl (1:1.08-1.1+b4) ...
Setting up libglib2.0-0:armhf (2.58.3-2+deb10u2) ...
No schema files found: doing nothing.
Setting up perl-openssl-defaults:armhf (3) ...
Setting up libicu63:armhf (63.1-6+deb10u1) ...
Setting up libauthen-pam-perl (0.16-3+b5) ...
Setting up libxml2:armhf (2.9.4+dfsg1-7+deb10u1) ...
Setting up libnet-ssleay-perl (1.85-2+b1) ...
Setting up shared-mime-info (1.10-1) ...
Setting up webmin (1.962) ...
Webmin install complete. You can now login to https://DietPi3:10000/
as root with your root password, or as any user who can use sudo
to run commands as root.
Processing triggers for systemd (241-7~deb10u4+rpi1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
root@DietPi3:~#

MichaIng, 2020-12-09:
It is flaky:
```
2020-12-09 01:28:16 root@VM-Buster:~# apt update
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Hit:4 https://deb.debian.org/debian buster-backports InRelease
Ign:5 https://download.webmin.com/download/repository sarge InRelease
Err:6 https://download.webmin.com/download/repository sarge Release
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification. [IP: 104.207.151.13 443]
Reading package lists... 26%
```
So it is indeed the HTTPS certificate. What is going on there.

Joulinar, 2020-12-09:
could it be some load balancer on webmin side. one is working while the other one is failing?

MichaIng, 2020-12-09:
Seems to be something like that.
#### Failing access:
```
2020-12-09 01:33:29 root@VM-Buster:~# openssl s_client -showcerts -connect download.webmin.com:443
CONNECTED(00000003)
depth=2 O = Digital Signature Trust Co., CN = DST Root CA X3
verify return:1
depth=1 C = US, O = Let's Encrypt, CN = R3
verify return:1
depth=0 CN = download.webmin.com
verify return:1
---
Certificate chain
 0 s:CN = download.webmin.com
   i:C = US, O = Let's Encrypt, CN = R3
-----BEGIN CERTIFICATE-----
MIIFRDCCBCygAwIBAgISA9c9GNGk2TClBBcHf6xnkMEWMA0GCSqGSIb3DQEBCwUA
MDIxCzAJBgNVBAYTAlVTMRYwFAYDVQQKEw1MZXQncyBFbmNyeXB0MQswCQYDVQQD
EwJSMzAeFw0yMDEyMDgyMjEyNThaFw0yMTAzMDgyMjEyNThaMB4xHDAaBgNVBAMT
E2Rvd25sb2FkLndlYm1pbi5jb20wggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEK
AoIBAQC7hDxfl8uNnatzwTNCSnKggPjtDoX4nC6KTn9Oi6kvSD2KO0zsGMFO/ba8
UKZw8cdkUBxVpW2yCBwyxMrCWl6uB9Lc7mJMaMRa9HSIyeIHi5mTI0DZcLQJBZhE
VBXRIy/p2lCksYNPmmhRwqDcQ8fI21HLIo9Shmsr1yQiK4ixVPynXYvyzMBFmq2s
8N9KHfuC9/Puuh4GjkDIH2h1H72GGhl24W164C/XJ1ScvvILYnwR/ao/xUhXqSjV
rmSwB0rOHZcuO6sYIN0u2dwYpg58nPV+ySXEnWaWbMpbyWT/ViXkcAhaJWEJyjCy
7vV0xlYWRgAYbij8y5YeKaaMPQVZAgMBAAGjggJmMIICYjAOBgNVHQ8BAf8EBAMC
BaAwHQYDVR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMAwGA1UdEwEB/wQCMAAw
HQYDVR0OBBYEFMDFnE2JdXxgxpRa6sAH7hu820ajMB8GA1UdIwQYMBaAFBQusxe3
WFbLrlAJQOYfr52LFMLGMFUGCCsGAQUFBwEBBEkwRzAhBggrBgEFBQcwAYYVaHR0
cDovL3IzLm8ubGVuY3Iub3JnMCIGCCsGAQUFBzAChhZodHRwOi8vcjMuaS5sZW5j
ci5vcmcvMDcGA1UdEQQwMC6CE2Rvd25sb2FkLndlYm1pbi5jb22CF3d3dy5kb3du
bG9hZC53ZWJtaW4uY29tMEwGA1UdIARFMEMwCAYGZ4EMAQIBMDcGCysGAQQBgt8T
AQEBMCgwJgYIKwYBBQUHAgEWGmh0dHA6Ly9jcHMubGV0c2VuY3J5cHQub3JnMIIB
AwYKKwYBBAHWeQIEAgSB9ASB8QDvAHYAb1N2rDHwMRnYmQCkURX/dxUcEdkCwQAp
Bo2yCJo32RMAAAF2RKEfvQAABAMARzBFAiBMz3VD2Azka2AHdxdJZsbWOMQELJiJ
/+iwELf0HiAFjgIhALVmAMKZp+y2JDmAs303GjrvhqoUSYdc1KEaZoIV55PNAHUA
fT7y+I//iFVoJMLAyp5SiXkrxQ54CX8uapdomX4i8NcAAAF2RKEfywAABAMARjBE
AiBtb0dbs60pZhcuyWtGN2AfpVJhKeU+BSODKNb6IfGMwAIgQzWYPRuplQhnKrOa
vBqmOxn/4EBHe2kADDNR/KB9bG4wDQYJKoZIhvcNAQELBQADggEBAJJ77Gw8YTeg
C/6XSFX8Jz0Wb4wSbQMvpyYQuQKwR1R4mMiGI5D63wBClVjyAG5vgCQ6aDITa1r2
F/NE9hOMouvd1Fj9VbG5rbNkfgc6KNV95NW49T/SQoybRE9FivjRGiKXJ+Y9Hxzh
DTmxjeYqF3WwPDuJ3wi51wjkrqcaIv9IsFlB9FrOHf7pVUQWQWpUboak7rpOMOv9
aMBmRR8iMyHvr5sa2DzXZ5LTsmMlbMfmacdeJKuS5KEouHbwRtw6Ppe41WEr5CUR
DA/0GhuQCGPI7l0n2onpK984LmJ4RtzsCCuEksCXR8cryxQMmYECFQnkKDG5/sEV
SDYvHoWnrtU=
-----END CERTIFICATE-----
```
#### Successful access:
```
2020-12-09 01:34:21 root@VM-Buster:~# openssl s_client -showcerts -connect download.webmin.com:443
CONNECTED(00000003)
depth=0 CN = download.webmin.com
verify error:num=20:unable to get local issuer certificate
verify return:1
depth=0 CN = download.webmin.com
verify error:num=21:unable to verify the first certificate
verify return:1
---
Certificate chain
 0 s:CN = download.webmin.com
   i:C = US, O = Let's Encrypt, CN = Let's Encrypt Authority X3
-----BEGIN CERTIFICATE-----
MIIFdjCCBF6gAwIBAgISBC8YMUmfHBDCDKsrU/yr/lxrMA0GCSqGSIb3DQEBCwUA
MEoxCzAJBgNVBAYTAlVTMRYwFAYDVQQKEw1MZXQncyBFbmNyeXB0MSMwIQYDVQQD
ExpMZXQncyBFbmNyeXB0IEF1dGhvcml0eSBYMzAeFw0yMDEwMDkwNjA2MzJaFw0y
MTAxMDcwNjA2MzJaMB4xHDAaBgNVBAMTE2Rvd25sb2FkLndlYm1pbi5jb20wggEi
MA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDWYrQwBqXWoffLg1CSyqvF/NMt
D/lEYxRUIPLFBmb84BUO9Ze0ooXXLqhgEz++L04W70tN3aoLqhTu3C2XM+jf+pIO
tfLmKuEbcB7swPWVezT8HioTimvBQV7ofCs4nKts8cpwJTdfLCQIJ6gJDeabxIZI
uMqxGk8xwQx9vtDQNXAxd21xKFUd5as1L/mE943/ya9d1d1d9afj4HohSXurPDSV
+V4hNB0gNKMohn9usSAk3vIwTybNuSQ1ZNN3FcM4UAojTiecDhZ0KrDaQ/eY1WZe
r2ltCohKryEAdL29QiYLgHxrjC62n/hAWHHFr2TeZ8USshirAj5uzc1u/ACHAgMB
AAGjggKAMIICfDAOBgNVHQ8BAf8EBAMCBaAwHQYDVR0lBBYwFAYIKwYBBQUHAwEG
CCsGAQUFBwMCMAwGA1UdEwEB/wQCMAAwHQYDVR0OBBYEFBLJfnddTlnSI9sH6B+y
S9pWQezrMB8GA1UdIwQYMBaAFKhKamMEfd265tE5t6ZFZe/zqOyhMG8GCCsGAQUF
BwEBBGMwYTAuBggrBgEFBQcwAYYiaHR0cDovL29jc3AuaW50LXgzLmxldHNlbmNy
eXB0Lm9yZzAvBggrBgEFBQcwAoYjaHR0cDovL2NlcnQuaW50LXgzLmxldHNlbmNy
eXB0Lm9yZy8wNwYDVR0RBDAwLoITZG93bmxvYWQud2VibWluLmNvbYIXd3d3LmRv
d25sb2FkLndlYm1pbi5jb20wTAYDVR0gBEUwQzAIBgZngQwBAgEwNwYLKwYBBAGC
3xMBAQEwKDAmBggrBgEFBQcCARYaaHR0cDovL2Nwcy5sZXRzZW5jcnlwdC5vcmcw
ggEDBgorBgEEAdZ5AgQCBIH0BIHxAO8AdgBElGUusO7Or8RAB9io/ijA2uaCvtjL
MbU/0zOWtbaBqAAAAXUMLsGDAAAEAwBHMEUCIEpIXmihEPrSYGJiGdsH5nMLEy1j
nJG4kHgDJyO1Td7RAiEAm6riBCasr8HR8md8kKyn+wL507cWzNd3jvqUBp0nMh0A
dQD2XJQv0XcwIhRUGAgwlFaO400TGTO/3wwvIAvMTvFk4wAAAXUMLsF7AAAEAwBG
MEQCIFpZIgqHawdqfY/vOkC80Eq8HAL4GnqZPoi10hKGyeZIAiAI5nCeIdnvCy5F
3kU5DByOvJ3E+s195GCGEWpww72bfzANBgkqhkiG9w0BAQsFAAOCAQEAaPlpog93
ksoMtrclvUEHQDcILhxKZPxERwD5XIsc+zVSd86/eCvYwP7PcewcFpPJcyLiyL4F
xKpuscrL9q8Npo7eX3EQtFiinl34/m2yCEVavKSUdpfeAi2JrNeZSSHq/+9esnQu
7HMWB37uJx3E3d7lnd0ILNPhD7J0pzzX97QoigovuCdVnAT+7zlPgFY+boO73j5A
qlBKggT9NwH1sXcrBlJhNoH9EnwHp+FQZ5qQ/yDcW0Hdu9vpBzKQXPq1H5RQ8DYR
x9Hh6bhB7HlDrZaDcRBVGCo3XOejqiLXSMC5hq3pxv3Odr8VAuqqvq36++0VrAyP
Kl95H2ZoYqKj5Q==
-----END CERTIFICATE-----
```

Joulinar, 2020-12-09:
2 different certificates

```
Certificate chain
 0 s:CN = download.webmin.com
   i:C = US, O = Let's Encrypt, CN = R3
```

```
Certificate chain
 0 s:CN = download.webmin.com
   i:C = US, O = Let's Encrypt, CN = Let's Encrypt Authority X3
```

MichaIng, 2020-12-09:
Jep, it's two different CNs it seems, `R3` and `Let's Encrypt Authority X3` and if those match (in your case above both `Let's Encrypt Authority X3`, in my case both `R3`) it succeeds, if both have different CNs, it fails:
```
   i:C = US, O = Let's Encrypt, CN = Let's Encrypt Authority X3
+
 1 s:C = US, O = Let's Encrypt, CN = R3
```
=> failure

```
   i:C = US, O = Let's Encrypt, CN = R3
+
 1 s:C = US, O = Let's Encrypt, CN = R3
```
=> success

MichaIng, 2020-12-09:
So Let's Encrypt is using `Let's Encrypt Authority X3` sometimes and `R3` the other times? And how is a Let's Encrypt intermediate certificate mixed into a chain with a root certificate issued for Let's Encrypt but with a different CN? I'm a bid confused what is happening 😄.

Joulinar, 2020-12-09:
and to make it even more confusing. This is the 2nd tome today I see `R3` on a certificate that should have been issues by Let's Encrypt and seems not to be valid. Maybe just a coincident but who knows.

https://dietpi.com/phpbb/viewtopic.php?f=11&p=29336#p29336

MichaIng, 2020-12-09:
It is all not an issue if it is signed by the root certificate accordingly. In all my successful cases, LE CN was R3 but the root certificate was issued for CN=R3 as well, so all fine.

Little more about LE certificate updates, the intermediate ones and their new root certificate: https://scotthelme.co.uk/lets-encrypts-new-root-and-intermediate-certificates/

Not sure if the whole chain (that must be consistent) is shipped by the server, or only the initial certificate while clients pull intermediate and root certificates by themselves 😅, probably something mixes up there with the R3 ones. I could force a now cert of my server and see what I get 😄.

Interesting issue, somehow, but now it's time to sleep.

Joulinar, 2020-12-09:
> but now it's time to sleep

signed, bot not by Let's Encrypt 😉

manilx, 2020-12-09:
> curl -sSLf 'https://www.webmin.com/jcameron-key.asc' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-webmin.gpg --yes

manilx, 2020-12-09:
> Guys, it is not the HTTPS certificate, but the APT key.
> 
> @manilx
> Can you please check if the APT gpg key exists?
> 
> ```
> ls -l /etc/apt/trusted.gpg.d/dietpi-webmin.gpg
> ```
> 

It did.

> In case do:
> 
> ```
> curl -sSLf 'https://www.webmin.com/jcameron-key.asc' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-webmin.gpg --yes
> ```

done

> 
> and switch back to https for that repo.

Switched back. Did apt-get update and it loaded 4 packages from https://download.webmin.com/download/repository sarge Release

Did apt-get upgrade and it loaded 2 new ones: libssl1.1 openssl

Installed fail2ban as test and it worked fine this time.

MichaIng, 2020-12-09:
Okay you are lucky then, since I'm still getting randomly the false certificate in 50% of cases 😄. I reopen it for other users running into the same issue, until it is fixed.

manilx, 2020-12-09:
> The problem indeed is that not only webmin install fails but all APT update executions.

Exactly! This was my original issue, as I had webadmin installed and was installing something else.

MichaIng, 2020-12-10:
Issue fixed here, both servers now serve the new `R3` certificate.

closed 2020-12-10 by MichaIng