MichaIng/DietPi #3966: no more access to LMS/Squeezelite to stream music

nico3182, 2020-12-10:
hello
I can no more stream music from my Allo Usbridge with Dietpi installed and LMS/Squeezelite
The dietpi is no more seen by by LMS application on my laptop or squeezer on mobile phone.
I did no change since last streaming.
I tried several reboots, NIL help.
I removed "roon extension manager" and "roon bridge" because I don"t use them. I had errors and the dietpi is no more visisble. I tried to re-install them, errors:
npm ERR! Response timeout while trying to fetch https://registry.npmjs.org/mkdirp (over 300000ms)
npm ERR! A complete log of this run can be found in:
npm ERR!    /root/.npm/_logs/2020-12-10T11_22_58_060Z-debug.log
npm ERR! invalid options argument      

here are additional info:
- Date           | Thu Dec 10 12:36:35 CET 2020
- Bug report     | ed987da7-8247-4ac6-ac0e-f58a33c89e02
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi CM 3+ (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.73-v7+ #1266 SMP Tue Sep 17 20:52:18 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | ./setup.sh
- Exit code      | 1
- Software title | DietPi-Software

I don't understand why suddenly it does not work.

Thanks for your help
Best regards
Nicolas

Joulinar, 2020-12-10:
Hi,

many thanks for your message. Looks like your system is trying to connect to `https://registry.npmjs.org/mkdirp` and this is not working. 

`npm ERR! Response timeout while trying to fetch https://registry.npmjs.org/mkdirp (over 300000ms)`

There seems to be some error log available as well `/root/.npm/_logs/2020-12-10T11_22_58_060Z-debug.log`. Maybe you can have a look

Next to that you are still running Debian Stretch. Probably you could have a look to our new Buster image 😃 https://dietpi.com/downloads/images/DietPi_RPi-ARMv7-Buster_AlloGUI.7z

MichaIng, 2020-12-10:
To test access to this URL: `curl -IL https://registry.npmjs.org/mkdirp`

> I had errors and the dietpi is no more visisble.

Can you show the errors you had during Roon uninstall?

I'll also have a look into your bug report upload.

nico3182, 2020-12-10:
this is result of curl -IL https://registry.npmjs.org/mkdirp

HTTP/2 200
date: Thu, 10 Dec 2020 16:03:48 GMT
content-type: application/json
content-length: 58613
set-cookie: __cfduid=d30a6ac9ad0b98955d4c01de09baa2c531607616228; expires=Sat, 09-Jan-21 16:03:48 GMT; path=/; domain=.npmjs.org; HttpOnly; SameSite=Lax
cf-ray: 5ff832b28b1ff96d-BRU
accept-ranges: bytes
age: 5915
cache-control: public, max-age=300
etag: "0730d9c43c136c318d8e19c69cbd72aa"
last-modified: Wed, 22 Apr 2020 06:03:24 GMT
vary: accept-encoding, accept
cf-cache-status: HIT
cf-request-id: 06eefe039a0000f96dc1bb8000000001
expect-ct: max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
server: cloudflare

nico3182, 2020-12-10:
sorry I don't have anymore the bug report of roon uninstall

nico3182, 2020-12-10:
how to read/view/upload the error log "/root/.npm/_logs/2020-12-10T11_22_58_060Z-debug.log" ? (sorry I don't know anything about linux.....)

nico3182, 2020-12-10:
in fact I don't understand what happened as everything was running well up to now and I didn't change anything.
Could this be a material issue ?(allo usbridge because  my  network is ok)

ravenclaw900, 2020-12-10:
You could run `sudo more /root/.npm/_logs/2020-12-10T11_22_58_060Z-debug.log` to read the log.  To upload it, it would probably be best to run `sudo cat /root/.npm/_logs/2020-12-10T11_22_58_060Z-debug.log` and copy-and-paste into a GitHub comment.

nico3182, 2020-12-10:
0 verbose cli [
0 verbose cli   '/usr/local/bin/node',
0 verbose cli   '/usr/local/bin/npm',
0 verbose cli   'install',
0 verbose cli   '-g',
0 verbose cli   'https://github.com/TheAppgineer/roon-extension-manager.git'
0 verbose cli ]
1 info using npm@7.0.15
2 info using node@v15.4.0
3 timing config:load:defaults Completed in 7ms
4 timing config:load:file:/usr/local/lib/node_modules/npm/npmrc Completed in 3ms
5 timing config:load:builtin Completed in 3ms
6 timing config:load:cli Completed in 11ms
7 timing config:load:env Completed in 2ms
8 timing config:load:file:/tmp/DietPi-Software/.npmrc Completed in 0ms
9 timing config:load:project Completed in 3ms
10 timing config:load:file:/root/.npmrc Completed in 1ms
11 timing config:load:user Completed in 1ms
12 timing config:load:file:/usr/local/etc/npmrc Completed in 5ms
13 timing config:load:global Completed in 5ms
14 timing config:load:cafile Completed in 1ms
15 timing config:load:validate Completed in 1ms
16 timing config:load:setUserAgent Completed in 3ms
17 timing config:load:setEnvs Completed in 4ms
18 timing config:load Completed in 42ms
19 verbose npm-session 08644371a379a467
20 timing npm:load Completed in 76ms
21 timing arborist:ctor Completed in 3ms
22 timing idealTree:init Completed in 30ms
23 timing idealTree:userRequests Completed in 2916ms
24 silly idealTree buildDeps
25 silly fetch manifest roon-extension-manager@https://github.com/TheAppgineer/roon-extension-manager.git
26 silly placeDep ROOT roon-extension-manager@0.11.8 OK for:  want: https://github.com/TheAppgineer/roon-extension-manager.git
27 silly fetch manifest node-roon-api@github:roonlabs/node-roon-api
28 silly fetch manifest node-roon-api-settings@github:roonlabs/node-roon-api-settings
29 silly fetch manifest node-roon-api-status@github:roonlabs/node-roon-api-status
30 silly fetch manifest node-api-extension-installer@github:TheAppgineer/node-api-extension-installer#v0.11.10
31 silly fetch manifest node-api-time-input@github:TheAppgineer/node-api-time-input
32 timing idealTree:#root Completed in 5564ms
33 silly placeDep node_modules/roon-extension-manager node-api-extension-installer@0.11.10 OK for: roon-extension-manager@0.11.8 want: github:TheAppgineer/node-api-extension-installer#v0.11.10
34 silly placeDep node_modules/roon-extension-manager node-api-time-input@0.1.0 OK for: roon-extension-manager@0.11.8 want: github:TheAppgineer/node-api-time-input
35 silly placeDep node_modules/roon-extension-manager node-roon-api@1.2.2 OK for: roon-extension-manager@0.11.8 want: github:roonlabs/node-roon-api
36 silly placeDep node_modules/roon-extension-manager node-roon-api-settings@1.0.0 OK for: roon-extension-manager@0.11.8 want: github:roonlabs/node-roon-api-settings
37 silly placeDep node_modules/roon-extension-manager node-roon-api-status@1.0.0 OK for: roon-extension-manager@0.11.8 want: github:roonlabs/node-roon-api-status
38 silly fetch manifest mkdirp@^0.5.0
39 silly fetch manifest tar@^4.4.10
40 silly fetch manifest node-api-extension-runner@github:TheAppgineer/node-api-extension-runner#v0.3.0
41 silly fetch manifest node-api-extension-installer-docker@github:TheAppgineer/node-api-extension-installer-docker#v0.3.2
42 silly fetch manifest ip@^1.1.3
43 silly fetch manifest node-uuid@^1.4.7
44 silly fetch manifest ws@>=3.3.1
45 verbose stack TypeError: invalid options argument
45 verbose stack     at optsArg (/usr/local/lib/node_modules/npm/node_modules/mkdirp/lib/opts-arg.js:13:11)
45 verbose stack     at mkdirp (/usr/local/lib/node_modules/npm/node_modules/mkdirp/index.js:11:10)
45 verbose stack     at tryCatcher (/usr/local/lib/node_modules/npm/node_modules/bluebird/js/release/util.js:16:23)
45 verbose stack     at ret (eval at makeNodePromisifiedEval (/usr/local/lib/node_modules/npm/node_modules/bluebird/js/release/promisify.js:184:12), <anonymous>:13:39)
45 verbose stack     at Object.mkdirfix (/usr/local/lib/node_modules/npm/node_modules/npm-registry-fetch/node_modules/cacache/lib/util/fix-owner.js:36:10)
45 verbose stack     at makeTmp (/usr/local/lib/node_modules/npm/node_modules/npm-registry-fetch/node_modules/cacache/lib/content/write.js:121:19)
45 verbose stack     at write (/usr/local/lib/node_modules/npm/node_modules/npm-registry-fetch/node_modules/cacache/lib/content/write.js:35:19)
45 verbose stack     at putData (/usr/local/lib/node_modules/npm/node_modules/npm-registry-fetch/node_modules/cacache/put.js:11:10)
45 verbose stack     at Object.x.put (/usr/local/lib/node_modules/npm/node_modules/npm-registry-fetch/node_modules/cacache/locales/en.js:28:37)
45 verbose stack     at WriteStream._flush (/usr/local/lib/node_modules/npm/node_modules/npm-registry-fetch/node_modules/make-fetch-happen/cache.js:156:21)
45 verbose stack     at WriteStream._write (/usr/local/lib/node_modules/npm/node_modules/flush-write-stream/index.js:36:35)
45 verbose stack     at doWrite (/usr/local/lib/node_modules/npm/node_modules/flush-write-stream/node_modules/readable-stream/lib/_stream_writable.js:428:64)
45 verbose stack     at writeOrBuffer (/usr/local/lib/node_modules/npm/node_modules/flush-write-stream/node_modules/readable-stream/lib/_stream_writable.js:417:5)
45 verbose stack     at WriteStream.Writable.write (/usr/local/lib/node_modules/npm/node_modules/flush-write-stream/node_modules/readable-stream/lib/_stream_writable.js:334:11)
45 verbose stack     at WriteStream.end (/usr/local/lib/node_modules/npm/node_modules/flush-write-stream/index.js:45:41)
45 verbose stack     at WriteStream.end (/usr/local/lib/node_modules/npm/node_modules/flush-write-stream/index.js:42:47)
46 verbose cwd /tmp/DietPi-Software
47 verbose Linux 4.19.73-v7+
48 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "-g" "https://github.com/TheAppgineer/roon-extension-manager.git"
49 verbose node v15.4.0
50 verbose npm  v7.0.15
51 error invalid options argument
52 verbose exit 1

nico3182, 2020-12-10:
maybe I should re-install the whole system with https://dietpi.com/downloads/images/DietPi_RPi-ARMv7-Buster_AlloGUI.7z , no ?
I have seen this topic, I could do the same ?
https://dietpi.com/phpbb/viewtopic.php?f=11&t=8368 

thanks

MichaIng, 2020-12-10:
> 45 verbose stack TypeError: invalid options argument

Some option is invalid, probably a Roon version update invalidated left-over config file entries? If you don't use Roon anymore, I'd definitely uninstall it. If your run into issues elsewhere, let's investigate those directly as they should not be related to Roon 😉.

But I'll do a test install on Stretch, just to be sure, probably it is a general Stretch incompatibility.

nico3182, 2020-12-10:
ok thanks but do you see why I can't stream music with Squeezelite anymore?

MichaIng, 2020-12-10:
Not yet, but did you go through the other steps I added above below the horizontal rule? https://github.com/MichaIng/DietPi/issues/3966#issuecomment-742587439

Otherwise, indeed a switch to the new Buster-based RPi image would be an idea, as the Stretch systems don't get package or kernel updates anymore. The Allo web UI image would be an idea (if you use the web UI actively) or the regular 32-bit image where software needs to be selected for install manually: https://dietpi.com/#download

nico3182, 2020-12-15:
Hello again,

I have installed

nico3182, 2020-12-15:
DietPi v6.33.3 : 16:40 - Tue 12/15/20
 ─────────────────────────────────────────────────────
 - Device model : RPi CM 3+ (armv7l)
 - CPU temp : 48'C : 118'F (Optimal temperature)
 - LAN IP : 192.168.0.41 (eth0)
 - MOTD : Take care of yourself and the people around you :)
 ─────────────────────────────────────────────────────

nico3182, 2020-12-15:
I have installed the buster image

All seems fine except I have no sound.
I have a USB DAC (Soncoz SG1) connected to the Allo Usbridge with USB
If I select usb-dac in the audio options, soundcard, nothing happens,
Soundcard           : [none]

Can you help please?
Thanks
Nicolas

MichaIng, 2020-12-15:
Please paste the output of the following commands:
```
aplay -l
cat /etc/asound.conf
lsmod
```

Did this DAC work before, i.e. is is supposed to work OOTB on Raspberry Pi or does it required a kernel driver to function?

nico3182, 2020-12-15:
root@DietPi:~# aplay -l
aplay: device_list:272: no soundcards found...

root@DietPi:~# cat /etc/asound.conf
cat: /etc/asound.conf: No such file or directory

root@DietPi:~# lsmod
Module                  Size  Used by
binfmt_misc            20480  1
bcm2835_isp            32768  0
ax88179_178a           32768  0
bcm2835_mmal_vchiq     28672  1 bcm2835_isp
videobuf2_dma_contig    20480  1 bcm2835_isp
videobuf2_memops       16384  1 videobuf2_dma_contig
videobuf2_v4l2         28672  1 bcm2835_isp
videobuf2_common       57344  2 bcm2835_isp,videobuf2_v4l2
videodev              237568  3 bcm2835_isp,videobuf2_common,videobuf2_v4l2
raspberrypi_hwmon      16384  0
mc                     40960  4 bcm2835_isp,videobuf2_common,videodev,videobuf2_v4l2
vc_sm_cma              32768  2 bcm2835_isp,bcm2835_mmal_vchiq
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
ip_tables              28672  0
x_tables               32768  1 ip_tables
ipv6                  458752  36

nico3182, 2020-12-15:
yes it was working before perfectly, but I can't say if it requires a kernel driver

nico3182, 2020-12-15:
in fact the DAC does not need anything particular, no? it only receives sounds from the  Allo Usbridge via the USB port?

nico3182, 2020-12-15:
I have seen that I had selected the wrong entry on my USB-DAC (USB C instead of USB B), I have corrected and now  Dietpi detects my DAC, but I can't hear any sound. I don' know if it comes from the dietpi ?
I did the same commands as before:

root@DietPi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 1: SGD1 [SGD1], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

root@DietPi:~# cat /etc/asound.conf
pcm.!default {
        type hw
        card 1
        device 0
}

ctl.!default {
        type hw
        card 1
}

root@DietPi:~# lsmod
Module                  Size  Used by
snd_usb_audio         217088  2
snd_hwdep              16384  1 snd_usb_audio
snd_usbmidi_lib        32768  1 snd_usb_audio
bcm2835_isp            32768  0
snd_rawmidi            32768  1 snd_usbmidi_lib
snd_seq_device         16384  1 snd_rawmidi
bcm2835_mmal_vchiq     28672  1 bcm2835_isp
videobuf2_dma_contig    20480  1 bcm2835_isp
snd_pcm                94208  2 snd_usb_audio
videobuf2_memops       16384  1 videobuf2_dma_contig
videobuf2_v4l2         28672  1 bcm2835_isp
videobuf2_common       57344  2 bcm2835_isp,videobuf2_v4l2
snd_timer              32768  1 snd_pcm
snd                    73728  10 snd_hwdep,snd_usb_audio,snd_timer,snd_rawmidi,snd_usbmidi_lib,snd_seq_device,snd_pcm
ax88179_178a           32768  0
videodev              237568  3 bcm2835_isp,videobuf2_common,videobuf2_v4l2
mc                     40960  5 bcm2835_isp,snd_usb_audio,videobuf2_common,videodev,videobuf2_v4l2
vc_sm_cma              32768  2 bcm2835_isp,bcm2835_mmal_vchiq
raspberrypi_hwmon      16384  0
uio_pdrv_genirq        16384  0
fixed                  16384  0
uio                    20480  1 uio_pdrv_genirq
ip_tables              28672  0
x_tables               32768  1 ip_tables
ipv6                  458752  38

Any idea why I can't hear no sound ?
Thanks

nico3182, 2020-12-15:
I have seen that sound can be mute by alsa mixer. Here is alsa mixer window:
lqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq AlsaMixer v1.1.8 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqk
x Card: SGD1                                           F1:  Help               x
x Chip: USB Mixer                                      F2:  System information x
x View: F3: Playback  F4: Capture  F5: All             F6:  Select sound card  x
x Item:                                                Esc: Exit               x
x                                                                              x
x                                                                              x
x                                                                              x
x                This sound device does not have any controls.                 x
x                                                                              x
x                                                                              x
x                                                                              x
x                                                                              x
x                                                                              x
x                                                                              x
x                                                                              x
x                                                                              x
x                                                                              x
x                                                                              x
x                                                                              x
x                                                                              x
x                                                                              x
x                                                                              x
mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj

MichaIng, 2020-12-16:
Can you please try to select the `i-sabre-q2m` sound card from the `dietpi-config` menu? If I'm not mistaken it uses the same chip: `ES9028Q2M`

Since sound device is detected and `asound.conf` correctly configured for it, the missing `alsamixer` controls look like a missing driver. The above choice will enable the device tree overlay that is shipped together with the Raspberry Pi kernel to implement this driver.

nico3182, 2020-12-16:
Hello,

thanks for you answer.
In fact I tried to listen to QOBUZ using BubbleUPnP and it works well, I have sound using my DAC (I used i-sabre-q2m and now usb-dac), so I guess the problem is coming from LMS/Squeezelite/Squeezer because with squeezer I have  no sound at all. 
Also, the application Squeezer on my mobile phone has a lot of difficulties to connect to Dietpi (Allo Usbridge), and when it connects, I have no sound. Before it was working very well, so I don't know what happened.
I uninstalled LMS and Squezelite and re-installed them, same issue.
Is there something I missed? a software to install?
Any idea ?
Thanks.

MichaIng, 2021-02-15:
Sorry for losing track on this issue. Did you find a solution meanwhile?

MichaIng, 2021-03-25:
I mark the issue as closed. Feel free to reopen if required.

closed 2021-03-25 by MichaIng