MichaIng/DietPi #3970: dietpi-software cannot install apache2

rijotech, 2020-12-12:
No changes have been detected. Unable to start installation.

DietPi_RPi-ARMv8-Buster

#### Required Information
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=33
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

ravenclaw900, 2020-12-12:
Usually that means that no software has been chosen to install.  Are you sure you selected it?

Joulinar, 2020-12-12:
Anyway we have some changes for Ampache on v6.34 beta as we changed composer version. Best to switch to actual beta branch or to wait for release of 6.34.

MichaIng, 2020-12-13:
> Anyway we have some changes for Ampache

While that is true, it's about "Apache2", not "Ampache" 😉.

I guess Apache2 was selected from the preference menu but not within the "Optimised Software" menu. The preference does not actually select a webserver to be installed, but only selects a preference to be installed when another selected software requires a webserver. So e.g. you select "Apache2" there, then you select "Nextcloud" from "Optimised Software", and when hitting "Install", Apache2 is automatically installed as webserver, as Nextcloud requires one.

Closest to install a webserver only is to select a webserver stack, e.g. LAMP (Apache2 + MariaDB database + PHP). Or calling `dietpi-software install 83`, which will install Apache2 + PHP only. Standalone webserver installs are hidden from the menu, a decision from the previous DietPi lead developer, but actually I aim to change that by times, probably we have another vote for that here.

Joulinar, 2020-12-13:
> While that is true, it's about "Apache2", not "Ampache" 😉.

Whoops my bad. Need to read with more care. Sorry

rijotech, 2020-12-17:
This is a 64bit beta os maybe that's why is not working properly.

MichaIng, 2020-12-17:
Nope, in this case the 64-bit images has no issues, please read my post above, it is all expected.

rijotech, 2020-12-17:
> Nope, in this case the 64-bit images has no issues, please read my post above, it is all expected.

ok

closed 2020-12-17 by rijotech

MichaIng, 2020-12-17:
Did it work now one of the ways I mentioned above?