MichaIng/DietPi #3971: NoIp2 config script error

RaptorSDS, 2020-12-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=33
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | 
buster
- Kernel version |
Linux DietPi 5.4.79+ #1373 Mon Nov 23 13:18:15 GMT 2020 armv6l GNU/Linux

- SBC model | 
RPi Zero W
- Power supply used | (EG: 5V 1A RAVpower)
5V 1A 
- SDcard used | 
Intenso 16GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
NoIP2 config script
- Was the software title installed freshly or updated/migrated?
fresh install
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
[  OK  ] DietPi-Config | systemctl stop noip2
/boot/dietpi/dietpi-config: Zeile 4449: noip2: Kommando nicht gefunden.
Press any key to continue...

2. ...

#### Expected behaviour

- start of NoIp config

#### Actual behaviour

- ...[  OK  ] DietPi-Config | systemctl stop noip2
/boot/dietpi/dietpi-config: Zeile 4449: noip2: Kommando nicht gefunden.
Press any key to continue...

#### Extra details

- ...

Joulinar, 2020-12-13:
Hi,

Many thanks for your message. Looks like NoIP is not installed correctly. You can try to reinstall as follow

`dietpi-software reinstall 67`

RaptorSDS, 2020-12-13:
same error occure
also missing noip2.conf  in /etc/

Joulinar, 2020-12-13:
Does the reinstall finished without issue?

MichaIng, 2020-12-13:
The configuration file btw is located at `/usr/local/etc/no-ip2.conf`.

This should help to prompt on errors in the first place during install: https://github.com/MichaIng/DietPi/commit/0a9f5d981a681f5e8d8bfa6121a11ed50a1672bf
At least in theory all should work, just tested on VM and verified that our archives contain the expected file.

Another thing is that the PATH variable might be malformed. Very rare but could explain why a binary in /usr/local/bin is not found. With v6.34 this will be rules out as well since we force a consistent default PATH in our scripts then: https://github.com/MichaIng/DietPi/commit/a1f7bb928d67d42414040440ffc63b42bbb4306e#diff-c793d9b8219c44005903fe22fd5cfb1096b7789e1351ee0f9ed7c5c148c52c1bR90-R92

RaptorSDS, 2020-12-13:
yes install without error 

i change my mind and use dyndns on my router

closed 2020-12-13 by RaptorSDS