MichaIng/DietPi #3973: Nextcloud no :4443 port and no nc-config

RaptorSDS, 2020-12-13:
DietPi version |
    G_DIETPI_VERSION_CORE=6
    G_DIETPI_VERSION_SUB=33
    G_DIETPI_VERSION_RC=3
    G_GITBRANCH='master'
    G_GITOWNER='MichaIng'

    Distro version |
    buster

    Kernel version |
    Linux DietPi 5.4.79

    SBC model |
    RPi 3b+

    Power supply used | (EG: 5V 1A RAVpower)
    5V 1A

    SDcard used |
    sandisk 16GB

#### Additional Information (if applicable)
- Software title | 
 Nextcloud)
 Was the software title installed freshly or updated/migrated?
fresh install
pihole and nextcloud on same dietpi
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- want to at a usb stick to nextcloud

#### Actual behaviour

- no nc-config found on bash 
-no :4443 port avaible 404 error

#### Extra details

- ...

Joulinar, 2020-12-13:
Hi,

many thanks for your message. Pls can you give some more information where do you get this error message. Maybe you can describe the way how to get to it

RaptorSDS, 2020-12-13:
i get the error on bash cli ( over Putty ssh) for nc-config /ncc-config / nextcloud-config ->command not found
also i can not open the webcli of nextcloud localhost:4443 ( firefox and edge both do not found any site)

i want to config a usb stick for nextcloud for that i have to use Webcli or local cli for the nc-tools like in this manuel
[https://docs.nextcloudpi.com/en/how-to-configure-nextcloudpi/](url)
[https://docs.nextcloudpi.com/en/how-to-configure-an-external-usb-drive-with-nextcloudpi/](url)

Joulinar, 2020-12-13:
well this is NextCloudPi. This seems to be a fork of the original NextCloud. on DietPi we offer the original version. I will link out online docs for your reference https://dietpi.com/docs/software/cloud/#nextcloud

RaptorSDS, 2020-12-13:
ok thanks than this is not a bug , we can close here .

closed 2020-12-13 by RaptorSDS

MichaIng, 2020-12-14:
NextCloudPi is an appliance for the original Nextcloud, which installs the webserver stack around it and includes some CLI (nc-config) and web UI to configure Nextcloud. We had thoughts about implementing NextCloudPi into DietPi or base NextCloudPi on DIetPi, but the two projects simply aim too different goals: NextCloudPi is a single software appliance, so everything is setup and optimised to run Nextcloud and nothing else. DietPi aims to be a multi-purpose platform, so we need un-intrusive software installs: #1778

You can use `ncc` from console to configure Nextcloud via it's native CLI, but we have no own special Nextcloud CLI like NCP has.