MichaIng/DietPi #3975: LAN IP address is not recognized when using br0 interface

lwqcz, 2020-12-15:
#### Required Information
- DietPi version
`G_DIETPI_VERSION_CORE=6 G_DIETPI_VERSION_SUB=33 G_DIETPI_VERSION_RC=3 G_GITBRANCH='master' G_GITOWNER='MichaIng'`
- Distro version | `10.7 - buster`
- Kernel version | `4.4.143-55-rockchip-g6b7accbc999b #1 SMP PREEMPT Wed Jul 29 02:54:35 UTC 2020 aarch64 GNU/Linux`
- SBC model | `ROCK Pi S (aarch64)`
- Power supply used | `LM2596 3V stepdown DCDC from 12V/3A PSU to 5V header rail pins`
- SDcard used | `8GB Kingston class 4`

#### Additional Information (if applicable)
- Software title | `dietpi-config`
- Was the software title installed freshly or updated/migrated? `Fresh install + apt update&upgrade`
- Can this issue be replicated on a fresh installation of DietPi? `Yes`

#### Steps to reproduce
1. Remove IP address from `eth0` + `wlan0`
2. Add an IP address to `br0` (there is no way to specify which interface is "master" AFAIK)
3. MOTD will show `no-ip-address default notice`
```
 ─────────────────────────────────────────────────────
 DietPi v6.33.3 : 10:39 - Tue 12/15/20
 ─────────────────────────────────────────────────────
 - Device model : ROCK Pi S (aarch64)
 - CPU temp : 40'C : 104'F (Optimal temperature)
 - LAN IP : Use dietpi-config to setup a connection (eth0)
 - MOTD : Take care of yourself and the people around you :)
 ─────────────────────────────────────────────────────
```

#### Expected behavior
- as a `LAN IP` I would expect to show the only (non-localhost) IP address specified in the system
- or maybe in network configuration, there would be fine to have an option to specify which interface is a "master" LAN?
#### Actual behaviour
- as shown above - `LAN IP` is wrongly shown as unconfigured

#### Extra details
- I've done a complete procedure to switch WiFi hotspot from the state when `wlan0` is behind NAT on different IP subnet to
the state when `wlan0` is a part of a transparent bridge between `eth0` and `wlan0` - it would be lovely to have a `bridge` option in hotspot's configuration in `dietpi-config` - should I make another feature request or whole MR? It is quite simple and therefore I'm not sure.

Joulinar, 2020-12-15:
Hi,

many thanks for your message. I guess this is as expected as DietPi did not count with a `br0` interface. Usually expectation is `eth0` as well as `wlan0`. But @MichaIng can confirm for sure

lwqcz, 2020-12-15:
@Joulinar thx for your answer, I would rather make a MR, but I have no spare time no - sorry.

MichaIng, 2020-12-15:
Jep exactly, currently `dietpi-config` is unflexible when it's about interface names and types. I'm already working on a complete network rewrite, but that involves a large number of scripts and needs careful testing and whenever I hope for a few spare time days, something urgent is being thrown in 😄.

lwqcz, 2020-12-15:
I see, would be then better a bash script, callable with some params to change NATed hotspot to bridge? As I've tested today - when one uses `dietpi-config` to change something regarding wifi, `dietpi-config` will apply NATed hotspot settings once again and therefore this could be a way to switch back from a bridged hotspot. A neat idea isn't it? 🤔

MichaIng, 2021-02-15:
I mark this as closed. Feel free to reopen if required. Network options rework is in process to cover such cases.

closed 2021-02-15 by MichaIng