MichaIng/DietPi #3976: Proble upgrade HA 2020.12.0

maury77, 2020-12-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=6 G_DIETPI_VERSION_SUB=33 G_DIETPI_VERSION_RC=3
- Distro version | 10.6
- Kernel version | Linux sdomotica 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
- SBC model |RPi 3 Model B+ (armv7l) 
- Power supply used | EG: 5V 3A 

during the upgrade I have this problem

  Building wheel for multidict (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /home/homeassistant/.local/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /tmp/tmppeoxpwcz
       cwd: /tmp/pip-install-3giu4eds/multidict_52c95df19b0c47828be3d9aec1adb2ab
  Complete output (47 lines):
  *********************
  * Accelerated build *
  *********************
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.7
  creating build/lib.linux-armv7l-3.7/multidict
  copying multidict/_multidict_py.py -> build/lib.linux-armv7l-3.7/multidict
  copying multidict/_multidict_base.py -> build/lib.linux-armv7l-3.7/multidict
  copying multidict/_compat.py -> build/lib.linux-armv7l-3.7/multidict
  copying multidict/_abc.py -> build/lib.linux-armv7l-3.7/multidict
  copying multidict/__init__.py -> build/lib.linux-armv7l-3.7/multidict
  running egg_info
  writing multidict.egg-info/PKG-INFO
  writing dependency_links to multidict.egg-info/dependency_links.txt
  writing top-level names to multidict.egg-info/top_level.txt
  reading manifest file 'multidict.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no previously-included files matching '*.pyc' found anywhere in distribution
  warning: no previously-included files found matching 'multidict/_multidict.html'
  warning: no previously-included files found matching 'multidict/*.so'
  warning: no previously-included files found matching 'multidict/*.pyd'
  warning: no previously-included files found matching 'multidict/*.pyd'
  no previously-included directories found matching 'docs/_build'
  writing manifest file 'multidict.egg-info/SOURCES.txt'
  copying multidict/__init__.pyi -> build/lib.linux-armv7l-3.7/multidict
  copying multidict/_multidict.c -> build/lib.linux-armv7l-3.7/multidict
  copying multidict/py.typed -> build/lib.linux-armv7l-3.7/multidict
  creating build/lib.linux-armv7l-3.7/multidict/_multilib
  copying multidict/_multilib/defs.h -> build/lib.linux-armv7l-3.7/multidict/_multilib
  copying multidict/_multilib/dict.h -> build/lib.linux-armv7l-3.7/multidict/_multilib
  copying multidict/_multilib/istr.h -> build/lib.linux-armv7l-3.7/multidict/_multilib
  copying multidict/_multilib/iter.h -> build/lib.linux-armv7l-3.7/multidict/_multilib
  copying multidict/_multilib/pair_list.h -> build/lib.linux-armv7l-3.7/multidict/_multilib
  copying multidict/_multilib/views.h -> build/lib.linux-armv7l-3.7/multidict/_multilib
  running build_ext
  building 'multidict._multidict' extension
  creating build/temp.linux-armv7l-3.7
  creating build/temp.linux-armv7l-3.7/multidict
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.7m -c multidict/_multidict.c -o build/temp.linux-armv7l-3.7/multidict/_multidict.o -O2 -std=c99 -Wall -Wsign-compare -Wconversion -fno-strict-aliasing -pedantic
  multidict/_multidict.c:1:10: fatal error: Python.h: File o directory non esistente
   #include "Python.h"
            ^~~~~~~~~~
  compilation terminated.
  error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for multidict
Failed to build multidict
ERROR: Could not build wheels for multidict which use PEP 517 and cannot be installed directly

MichaIng, 2020-12-15:
Stange, the Python headers are of course present in the pyenv environment. I'll try to replicate.

maury77, 2020-12-15:
sorry I made a mistake. I did not run the script correctly

I lunch this
**sudo -u homeassistant dash -c '. /home/homeassistant/pyenv-activate.sh; pip3 install -U homeassistant==2020.12.0'**

The installation was successful

sorry...

closed 2020-12-15 by maury77

MichaIng, 2020-12-15:
Worked fine here, tested on RPi2:
```
2e3f425
  Stored in directory: /home/homeassistant/.cache/pip/wheels/0c/61/d6/4db4f4c28254856e82305fdb1f752ed7f8482e54c384d8cb0e
  Building wheel for multidict (PEP 517): started
  Building wheel for multidict (PEP 517): finished with status 'done'
  Created wheel for multidict: filename=multidict-5.1.0-cp38-cp38-linux_armv7l.whl size=180955 sha256=1cda7fa781c3dc430e4c5dbdd06e5ff760d500fb0cea21e3c7e8218d7d                                                                             3ee7bf
  Stored in directory: /home/homeassistant/.cache/pip/wheels/72/97/bc/5bffee542d18b454f73bd1a593cd104de3c46a50e4f2946728
Successfully built aiohttp bcrypt ciso8601 cryptography python-slugify pyyaml ruamel.yaml yarl MarkupSafe multidict
Installing collected packages: sniffio, rfc3986, pycparser, multidict, idna, h11, yarl, voluptuous, urllib3, typing-extensions, text-unidecode, six, pytz, Marku                                                                             pSafe, httpcore, chardet, cffi, certifi, attrs, async-timeout, voluptuous-serialize, ruamel.yaml, requests, pyyaml, python-slugify, PyJWT, pip, jinja2, httpx, c                                                                             ryptography, ciso8601, bcrypt, astral, aiohttp, homeassistant
```

maury77, 2020-12-15:
yes I don't run pyenv-activate.sh

MichaIng, 2020-12-15:
Okay, great it got sorted. It's a bid annoying to remember, but with this large amount of specific Python module requirements that Home Assistant has, we cannot really get around a pyenv, venv or virtualenv, which all require such an activation.