MichaIng/DietPi #3977: Reboot loop on RPi4 after setting to boot off USB drive

zedrdave, 2020-12-15:
### Creating a bug report/issue

#### Required Information
[Can no longer boot, so can't run commands, but I the version of DietPi/kernel/etc should be the latest stable as of today]

- DietPi version: DietPi_RPi-ARMv6-Buster [Thu Sep 10 14:55:33 CEST 2020]
- SBC model: Rpi4 Rev B
- Power supply used: 5V 3A 
- SDcard used: SanDisk ultra

#### Steps to reproduce
1. Set-up external USB drive as boot volume through `dietpi-config` (format + create mount partition + move root directory)
2. Reboot (reboots from external drive and everything is fine)
3. [?] Additional setup (misc software installs)
4. Reboot

#### Expected behaviour
RPi should boot normally

#### Actual behaviour
RPi displays following message for about 20s:
```
[sda] No Caching node page found
[sda] Assuming drive cache: write through
```
then spit out a few more lines (see picture) and reboots in a never-ending loop.
[image]

#### Extra details
If unplugging the USB drive, I only get a `_` prompt (which I'm assume is normal, since it can't find the drive it's supposed to boot from).

Is there at least a way to make it boot back off the SD card by editing the config in the boot partition? (can't seem to find the instructions for that)

Joulinar, 2020-12-15:
Hi,

many thanks for your message. You still have the SD card connected to your Raspberry Pi 4? Did you tried connecting the USB device to both USB2 as well as USB3 port?

zedrdave, 2020-12-15:
Yes. I did have the SD card in the whole time.

I did not try switching USB port (but the drive worked fine initially, so I doubt there'd be any USB connectivity issue).

Since I needed to regain access to the system, and couldn't seem to switch the boot back to SD, I re-flashed the card, and restarted the set-up. Upon inspection, the USB drive seemed to have some issues (fixed with fsck): presumably introduced sometime between the first and second reboot? Could there be an issue with the disk not getting unmounted properly before the Pi reboot (I have not idea what dietpi's disk tool does to fstab behind the scene)…

MichaIng, 2020-12-16:
Drives are explicitly unmounted by Debian/systemd on shutdown, so it should hang or you see a failure then.
```
[sda] No Caching node page found
[sda] Assuming drive cache: write through
```
Those pretty common for USB-attached drives.

> I have not idea what dietpi's disk tool does to fstab behind the scene

`cat /etc/fstab` => nothing special, especially for root and boot partitions, aside of reduced file timestamp writes it's default mount options.

Please have a look into `dmesg -l err,crit,alert,emerg` if there are any file system or voltage errors or similar. Btw, is it powered by a dedicated PSU or via USB only?
`vcgencmd get_throttled` is another RPi specific command to check if there was an under-voltage detected now or before.

zedrdave, 2020-12-16:
Since I could no longer boot (and my other local machines run on OSX, making reading ext4 partitions a bit difficult), I had to flash the SD card again and lost all logs. I am afraid I won't be able to give more details on this, and maybe this issue should be closed…

The disk was powered via USB, which might cause issues with early access at boot?
I don't think it is underpowered: it was running fine before and after I reset everything (I did not think to test it with external power when the reboot loop happened).

As mentioned above, after the reset, I repaired the drive and now using it for user data (I did not move the root FS this time) and everything works fine. If I have some time in the future, I will try and repeat all the steps (after making a backup of the SD card, and using a different HD) to see if I can reproduce the problem…

Thanks for your hard work and your help!

MichaIng, 2020-12-16:
> The disk was powered via USB, which might cause issues with early access at boot?

This can cause issues at any time. SBC USB power is not really designed to reliably power 2.5" drives. It works in many cases (especially modern SSDs) but I personally would not count on it.

> I don't think it is underpowered: it was running fine before and after I reset everything (I did not think to test it with external power when the reboot loop happened).

Power consumption can very, also with kernel upgrades, which include USB driver updates as well. If you have a docking station or something with external power supply it is at least worth to give it a try, when running into similar issues. The above two commands btw are to check CPU under-voltage, but that might be tied to USB voltage as well. In combination with other kernel errors (I/O or fs errors) it could give a hint at least.

> Thanks for your hard work and your help!

Many thanks, I'm happy that it works for now and keep the suggestions in mind when you boot from USB drive again.
__________
Ah now I remember the UAS issue. Some external drives do not support UAS, which is by default enabled since RPi 4. So another attempt would be to either disable UAS system-wide of for the single drive only:
- https://github.com/MichaIng/DietPi/issues/3858
- https://github.com/MichaIng/DietPi/issues/3754

closed 2021-02-15 by MichaIng