MichaIng/DietPi #3978: Squeezelite fails to install

glogiotatidis, 2020-12-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=33
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | squeezelight
- Was the software title installed freshly or updated/migrated? Fresh install
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
Select squeezelite from dietpi-software

#### Expected behaviour
Squeezelite installs successfully
- ...

#### Actual behaviour
Install fails. I got a subshell and run the install manually which returns 
```sh
root@homeassistant:/tmp/DietPi-Software# dpkg -i 36.deb 
Selecting previously unselected package squeezelite.
(Reading database ... 52582 files and directories currently installed.)
Preparing to unpack 36.deb ...
Unpacking squeezelite (1.9.8-1287-dietpi1) ...
dpkg: dependency problems prevent configuration of squeezelite:
 squeezelite depends on libmad0 (>= 0.15.1b-8+deb9u1); however:
  Package libmad0 is not installed.
 squeezelite depends on libmpg123-0 (>= 1.23.8-1+b1); however:
  Version of libmpg123-0:armhf on system is 1.23.8-1.
 squeezelite depends on libfaad2 (>= 2.8.0~cvs20161113-1+deb9u2); however:
  Package libfaad2 is not installed.
 squeezelite depends on libavformat57 (>= 7:3.2.15-0+deb9u1); however:
  Version of libavformat57:armhf on system is 7:3.2.14-1~deb9u1+rpt1.

dpkg: error processing package squeezelite (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 squeezelite
```

#### Extra details

- ...

Joulinar, 2020-12-16:
Hi,

Many thanks for your message. I guess you are running RPi with Debian Stretch. issue is that Raspberry OS Stretch did not provide the correct version of `libmpg123-0`, needed to be able to install SqueezeLite

Probably similar to the following https://dietpi.com/phpbb/viewtopic.php?f=11&t=8368

glogiotatidis, 2020-12-16:
Indeed I am. And thanks a perfect opportunity to say how well dietpi works that I forgot I'm a major Debian release behind!

I'll upgrade to Buster and see if things get fixed. Will close after.

Joulinar, 2020-12-16:
@MichaIng 

Probably we should block SqueezeLite on Raspberry OS Stretch. On Debian this is not an issue but on Raspberry OS.

glogiotatidis, 2020-12-16:
Indeed it works just fine on Buster. Thanks!

MichaIng, 2020-12-16:
Since I cannot reasonably know which exact package version is the minimum requirement, I simply take the currently installed ones (most recent ones at time of build). Since ARMv7 builds are done on Debian, in theory there could be version mismatches. I was checking the version available on Raspbian:
- http://raspbian.raspberrypi.org/raspbian/pool/main/f/ffmpeg/libavformat57_3.2.15-0+deb9u1_armhf.deb
- http://raspbian.raspberrypi.org/raspbian/pool/main/libm/libmad/libmad0_0.15.1b-8+deb9u1_armhf.deb
- http://raspbian.raspberrypi.org/raspbian/pool/main/m/mpg123/libmpg123-0_1.23.8-1_armhf.deb
    - This one seems to be the issue since the `+b1` is missing. I'll remove it from Stretch dependencies. No upstream version chance, only a re-build on Debian side, so this won't cause any issues.
- http://raspbian.raspberrypi.org/raspbian/pool/main/f/faad2/libfaad2_2.8.0~cvs20161113-1+deb9u2_armhf.deb

MichaIng, 2020-12-16:
Should work now with the new build: https://dietpi.com/downloads/binaries/stretch/squeezelite_armv7l.deb

Joulinar, 2020-12-16:
ok will need to check for a spare SD card tomorrow to perform a test an Stretch 😄

MichaIng, 2020-12-16:
Just tested here, but a second test would not hurt.

Joulinar, 2020-12-17:
ok installation is working fine on a RPi3B+ running Stretch

Joulinar, 2020-12-17:
can we close @MichaIng

MichaIng, 2020-12-17:
Yes, should be fine.

closed 2020-12-17 by MichaIng