MichaIng/DietPi #3985: Rock64 | 3.5mm audio not working again

lupa18, 2020-12-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.33.33
- Distro version | buster
- Kernel version | Linux rock64 5.9.14-rockchip64
- SBC model | ROCK64 (aarch64)

#### Additional Information (if applicable)
- Software title | drivers
- Can this issue be replicated on a fresh installation of DietPi? YES
- Maybe related to #2522

#### Steps to reproduce

Try to play some sound

#### Expected behaviour

Hear sounds

#### Actual behaviour

Can't hear anything

#### Extra details

When trying to use aplay with a sound:
```
aplay ./share/sounds/alsa/Side_Right.wav
Playing WAVE './share/sounds/alsa/Side_Right.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1345: Channels count non available
```

MichaIng, 2020-12-20:
Many thanks for your report.

This is a known issue when playing sound sources which are not compatible with the sound card/driver. The format needs to be converted. By coincidence, we added a feature with DietPi v6.34 (will be released later today) to enable the ALSA auto-conversion plugin. Please try it out:
```
cat << '_EOF_' > /etc/asound.conf
pcm.!default {
	type plug
	slave.pcm {
		type hw
		card 0
		device 0
	}
}

ctl.!default {
	type hw
	card 0
}
_EOF_
```
_Please adjust card and device index, if 0,0 is not correct, as probably HDMI sound is on card 0._

lupa18, 2020-12-20:
Hi @MichaIng  thanks for your answer! 

I don't have HDMI connected and I use the audio native jack output.

Yesterday I got MPD working manually adding sound card device (uncommenting a line at mpd.conf) and then doing the same at diepi-config. 

But I think the problem is "Auto Detection" feature, is not working at dietpi-config -> "Audio Options" -> "Soundcard". It returns an empty option.

MichaIng, 2020-12-20:
Auto-detection is based on ALSA sound card interfaces:
```
cat  /proc/asound/cards
```
respectively:
```
aplay -l
```
Everything that is listed there should be listed in `dietpi-config` as well. Was this not the case?

lupa18, 2020-12-20:
Ok ! I was confused. I think auto detection will chooose a card for me. 

What I did: Choose "auto detection" and it gives me an empty option (screenshot). Then click Ok and nothing working. 

[image]

It seems that after the empty option I must get inside the option again and choose the correct one.

closed 2020-12-20 by lupa18

MichaIng, 2020-12-20:
Ah, that line is meant to be a separator only, below are the detected sound cards + "usb-dac", which will enable the first found USB DAC.

Indeed it should not be possible to pass an empty value, hence this will cause an error now in the underlying script: https://github.com/MichaIng/DietPi/commit/7e1f6b6b0f2818f8feffcdd549073ba376a0b110

And our whiptail menu wrapper will now stay in the menu when such a separator line was selected: https://github.com/MichaIng/DietPi/commit/62903e25fb699a8471f504d4221faab4965a3619
This allows some other code simplifications as well as we currently often explicitly check for a non-empty output. So that one was overdue anyway 😃.