MichaIng/DietPi #3986: Nextcloud web update hits PHP-FPM memory limit

WatskeBart, 2020-12-20:
### Creating a bug report/issue

#### Required Information
- DietPi version = 6.33
- SBC model = RPi3

#### Additional Information (if applicable)
- Software title - Nextcloud
- Fresh installed
- Also reproduced with fresh install

#### Steps to reproduce

1. Use Nextcloud web updater

#### Expected behaviour

- Updating to latest stable release

#### Actual behaviour

- Error on Step 5 (verifing)

#### Extra details

- SOLUTION: Increase php-fpm memory limit to 256MB

MichaIng, 2020-12-20:
Many thanks for your report. It is also shown as warning in admin panel when below 512 MiB (I repeatedly forgot to address it) and will be an error with Nextcloud 21: https://github.com/nextcloud/server/commit/1995097cb8623b2447a2805a8e8e045c24203311

Just the right time before DietPi v6.34 to add it to Nextcloud PHP config override 👍.

MichaIng, 2020-12-20:
Done: https://github.com/MichaIng/DietPi/commit/5b9546519418da1f1aa877f7580504d4b719338a
As well applied during DietPi update: https://github.com/MichaIng/DietPi/commit/42c7854d2c3bf0d731073656b8c9991460f0e1a6

closed 2020-12-20 by MichaIng