MichaIng/DietPi #3990: SMB errors after upgrading to 6.34.3

seniorgod, 2020-12-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
- 
- G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
-
 Distro version | 9.13
- Kernel version | Linux DietPi 3.10.38 #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017 armv7l GNU/Linux
- SBC model | Sparky SBC (armv7l)

#### Additional Information (if applicable)
mpd, spotifyd, shairport_sync, upmpdcli, ifi-tidal-release

#### Steps to reproduce
After upgrading to 6.34.3. yesterday morning, hearing music with mpd is quite problematic. Sometimes it runs without error. Sometimes mpd freeze and i have to reboot sparky. I found the messages below in journalctl. I remember, that there were new ethernet drivers during update. So i guess, it is getting more unstable due to the new drivers or due to the smb version

    Dez 22 10:25:21 DietPi kernel: INFO: task lighttpd:1336 blocked for more than 120 seconds.
    Dez 22 10:25:21 DietPi kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
    Dez 22 10:25:21 DietPi kernel: ethernet link state:linked
    Dez 22 10:25:21 DietPi kernel: lighttpd        D
    Dez 22 10:25:21 DietPi kernel:  c07bfe48     0  1336      1 0x00000000
    Dez 22 10:25:21 DietPi kernel: [<c07bfe48>] (__schedule+0x390/0x7cc) from [<c07c065c>]    (schedule_preempt_disabled+0x14/0x20)
    Dez 22 10:25:21 DietPi kernel: [<c07c065c>] (schedule_preempt_disabled+0x14/0x20) from [<c07bef44>] (__mutex_lock_slowpath+0x174/0x368)
     Dez 22 10:25:21 DietPi kernel: [<c07bef44>] (__mutex_lock_slowpath+0x174/0x368) from [<c07bf144>]  (mutex_lock+0xc/0x24)
    Dez 22 10:25:21 DietPi kernel: [<c07bf144>] (mutex_lock+0xc/0x24) from [<c0247eb4>] (small_smb2_init+0x16c/0x390)
    Dez 22 10:25:21 DietPi kernel: [<c0247eb4>] (small_smb2_init+0x16c/0x390) from [<c0248c38>] (SMB2_open+0x40/0x544)
    Dez 22 10:25:21 DietPi kernel: [<c0248c38>] (SMB2_open+0x40/0x544) from [<c024a39c>] (smb2_open_op_close.constprop.0+0x70/0x1e4)
    Dez 22 10:25:21 DietPi kernel: [<c024a39c>] (smb2_open_op_close.constprop.0+0x70/0x1e4) from [<c024a5dc>] (smb2_query_path_info+0x70/0xa8)
    Dez 22 10:25:21 DietPi kernel: [<c024a5dc>] (smb2_query_path_info+0x70/0xa8) from [<c023c0b8>] (cifs_get_inode_info+0x238/0x45c)
    Dez 22 10:25:21 DietPi kernel: [<c023c0b8>] (cifs_get_inode_info+0x238/0x45c) from [<c023d558>] (cifs_revalidate_dentry_attr+0xb8/0xd8)
    Dez 22 10:25:21 DietPi kernel: [<c023d558>] (cifs_revalidate_dentry_attr+0xb8/0xd8) from [<c023d5b0>] (cifs_revalidate_dentry+0xc/0x2c)
    Dez 22 10:25:21 DietPi kernel: [<c023d5b0>] (cifs_revalidate_dentry+0xc/0x2c) from [<c0234424>] (cifs_d_revalidate+0x20/0xc8)
    Dez 22 10:25:21 DietPi kernel: [<c0234424>] (cifs_d_revalidate+0x20/0xc8) from [<c011b390>] (lookup_fast+0x248/0x280)
    Dez 22 10:25:21 DietPi kernel: [<c011b390>] (lookup_fast+0x248/0x280) from [<c011becc>] (link_path_walk+0x190/0x8e4)
    Dez 22 10:25:21 DietPi kernel: [<c011becc>] (link_path_walk+0x190/0x8e4) from [<c011d240>] (path_lookupat+0x54/0x7b8)
    Dez 22 10:25:21 DietPi kernel: [<c011d240>] (path_lookupat+0x54/0x7b8) from [<c011d9c4>] (filename_lookup+0x20/0xbc)
    Dez 22 10:25:21 DietPi kernel: [<c011d9c4>] (filename_lookup+0x20/0xbc) from [<c011ffb4>] (user_path_at_empty+0x50/0x7c)
    Dez 22 10:25:21 DietPi kernel: [<c011ffb4>] (user_path_at_empty+0x50/0x7c) from [<c011fff4>] (user_path_at+0x14/0x1c)
    Dez 22 10:25:21 DietPi kernel: [<c011fff4>] (user_path_at+0x14/0x1c) from [<c0116674>] (vfs_fstatat+0x48/0x94)
    Dez 22 10:25:21 DietPi kernel: [<c0116674>] (vfs_fstatat+0x48/0x94) from [<c01168e4>] (SyS_stat64+0x14/0x30)
    Dez 22 10:25:21 DietPi kernel: [<c01168e4>] (SyS_stat64+0x14/0x30) from [<c000f340>] (ret_fast_syscall+0x0/0x30)
    Dez 22 10:25:21 DietPi kernel: INFO: task update:2043 blocked for more than 120 seconds.
    Dez 22 10:25:21 DietPi kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
    Dez 22 10:25:21 DietPi kernel: update          D c07bfe48     0  2043      1 0x00000800
    Dez 22 10:25:21 DietPi kernel: [<c07bfe48>] (__schedule+0x390/0x7cc) from [<c07c065c>] (schedule_preempt_disabled+0x14/0x20)
    Dez 22 10:25:21 DietPi kernel: [<c07c065c>] (schedule_preempt_disabled+0x14/0x20) from [<c07bef44>] (__mutex_lock_slowpath+0x174/0x368)
    Dez 22 10:25:21 DietPi kernel: [<c07bef44>] (__mutex_lock_slowpath+0x174/0x368) from [<c07bf144>] (mutex_lock+0xc/0x24)
    Dez 22 10:25:21 DietPi kernel: [<c07bf144>] (mutex_lock+0xc/0x24) from [<c0247eb4>] (small_smb2_init+0x16c/0x390)
    Dez 22 10:25:21 DietPi kernel: [<c0247eb4>] (small_smb2_init+0x16c/0x390) from [<c0248c38>] (SMB2_open+0x40/0x544)
    Dez 22 10:25:21 DietPi kernel: [<c0248c38>] (SMB2_open+0x40/0x544) from [<c024a39c>] (smb2_open_op_close.constprop.0+0x70/0x1e4)
    Dez 22 10:25:21 DietPi kernel: [<c024a39c>] (smb2_open_op_close.constprop.0+0x70/0x1e4) from [<c024a5dc>] (smb2_query_path_info+0x70/0xa8)
    Dez 22 10:25:21 DietPi kernel: [<c024a5dc>] (smb2_query_path_info+0x70/0xa8) from [<c023c0b8>] (cifs_get_inode_info+0x238/0x45c)
    Dez 22 10:25:21 DietPi kernel: [<c023c0b8>] (cifs_get_inode_info+0x238/0x45c) from [<c023d558>] (cifs_revalidate_dentry_attr+0xb8/0xd8)
    Dez 22 10:25:21 DietPi kernel: [<c023d558>] (cifs_revalidate_dentry_attr+0xb8/0xd8) from [<c023d5b0>] (cifs_revalidate_dentry+0xc/0x2c)
    Dez 22 10:25:21 DietPi kernel: [<c023d5b0>] (cifs_revalidate_dentry+0xc/0x2c) from [<c0234424>] (cifs_d_revalidate+0x20/0xc8)
    Dez 22 10:25:21 DietPi kernel: [<c0234424>] (cifs_d_revalidate+0x20/0xc8) from [<c011b390>] (lookup_fast+0x248/0x280)
    Dez 22 10:25:21 DietPi kernel: [<c011b390>] (lookup_fast+0x248/0x280) from [<c011becc>] (link_path_walk+0x190/0x8e4)
    Dez 22 10:25:21 DietPi kernel: [<c011becc>] (link_path_walk+0x190/0x8e4) from [<c011f3c0>] (path_openat+0x84/0x4a0)
    Dez 22 10:25:22 DietPi kernel: [<c011f3c0>] (path_openat+0x84/0x4a0) from [<c0120028>] (do_filp_open+0x2c/0x80)
    Dez 22 10:25:22 DietPi kernel: [<c0120028>] (do_filp_open+0x2c/0x80) from [<c01118b4>] (do_sys_open+0xe4/0x170)
    Dez 22 10:25:22 DietPi kernel: [<c01118b4>] (do_sys_open+0xe4/0x170) from [<c000f4a8>] (__sys_trace_return+0x0/0x18)

seniorgod, 2020-12-22:
Hi, i'm going to close the issue. It seems my smb server produces the problems.....

closed 2020-12-22 by seniorgod

MichaIng, 2020-12-22:
Indeed the Ethernet driver was updated: https://github.com/sparky-sbc/sparky-test/tree/master/sparky-eth
If it really does cause issues, we could quite easily test older ones.