MichaIng/DietPi #3991: Medusa - after reinstalling can't open start page

aermak, 2020-12-22:
Medusa - after reinstalling can't open start page

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=6
                          G_DIETPI_VERSION_SUB=34
                          G_DIETPI_VERSION_RC=3
                          G_GITBRANCH='master'
                          G_GITOWNER='MichaIng'
- Distro version | 9.13
- Kernel version | Linux DietPi 4.19.66-v7+ #1253
- SBC model | RPi 3 Model B (armv7l)

#### Additional Information (if applicable)
- Software title | Medusa
- Was the software title installed freshly or updated/migrated? updated then reinstalled then fresh install
- Can this issue be replicated on a fresh installation of DietPi? no spare Pi
 3b7c015e-f04c-4d56-a63c-3877834ba365
- Bug report ID | echo $G_HW_UUID

#### Steps to reproduce
1. dietpi-software - removed Medusa since service won't start
2. Reboot
3. dietpi-software install Medusa 116
4. Reboot
2. Go to http://192.168.0.100:8081/ get nothing

#### Expected behaviour
- get a login page like before

#### Actual behaviour
- Unable to connect - can’t establish a connection to the server at 192.168.0.100:8081

MichaIng, 2020-12-22:
Many thanks for your report. I'm pasting the service error message you posted at the other issue:
```
root@DietPi:~# systemctl status medusa.service
● medusa.service - Medusa Daemon
   Loaded: loaded (/etc/systemd/system/medusa.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-12-22 15:59:36 CET; 2min 28s ago
  Process: 1662 ExecStart=/usr/bin/python /mnt/dietpi_userdata/medusa/start.py -q --daemon --nolaunch --datadir=/mnt/dietpi_userdata/medusa (code=exited, status=1/FAILURE)

Dec 22 15:59:36 DietPi python[1662]:     from feedparser.api import parse
Dec 22 15:59:36 DietPi python[1662]:   File "/mnt/dietpi_userdata/medusa/ext/feedparser/__init__.py", line 28, in <module>
Dec 22 15:59:36 DietPi python[1662]:     from .api import parse
Dec 22 15:59:36 DietPi python[1662]:   File "/mnt/dietpi_userdata/medusa/ext/feedparser/api.py", line 30, in <module>
Dec 22 15:59:36 DietPi python[1662]:     import urllib.parse
Dec 22 15:59:36 DietPi python[1662]: ImportError: No module named parse
Dec 22 15:59:36 DietPi systemd[1]: medusa.service: Control process exited, code=exited status=1
Dec 22 15:59:36 DietPi systemd[1]: Failed to start Medusa Daemon.
Dec 22 15:59:36 DietPi systemd[1]: medusa.service: Unit entered failed state.
Dec 22 15:59:36 DietPi systemd[1]: medusa.service: Failed with result 'exit-code'.
```

MichaIng, 2020-12-22:
I'm afraid Medusa dropped Python 2 support and does not support the old Python 3.5 (min. 3.6) that is shipped with Debian/Raspbian Buster:
- https://github.com/pymedusa/Medusa#dependencies
- https://github.com/pymedusa/Medusa/releases

You can build Python 3.9 manually and adjust the service file to launch Medusa with the `python3` binary, or downgrade to Medusa v0.4.6. But as Stretch will suffer more and more limitations of this kind, I would recommend to upgrade to Raspbian Buster or better flash our newest Raspberry Pi image, if that does not cause too much work during holidays 🙂.

As probably quickest workaround I'll check how to hardcode Medusa v0.4.6 into `dietpi-software`.

aermak, 2020-12-22:
@MichaIng  is there a way to upgrade without flashing the SD card?

Joulinar, 2020-12-22:
Hi,

we are not supporting it officially but you could have a look to following forum post. https://dietpi.com/phpbb/viewtopic.php?p=18988#p18988

However pls do a backup before

aermak, 2020-12-22:
Found some spare microSD cards flashing Buster now and will spend my xmas week setting it up like it was, worst case i go back to the original microSD and live without Medusa till i figure out what to do. Anyways thanks for the help.

Joulinar, 2020-12-22:
it shouldn't be a problem to clone you current SD card as you have a spare one. Once done just follow the link I posted above for instructions how to update from Stretch to Buster.

aermak, 2020-12-22:
I'm back in business with everything restored. Using Buster and Medusa updated and running. Want to keep this open for the sad souls using Strech @MichaIng  ? Otherwise just close it.

MichaIng, 2020-12-22:
Great to hear.

I'll try to pin an older version for Stretch, probably the internal updater is intelligent enough to know that it must not update, when running under Python 2. Else we need to disable Medusa on Stretch. I'll keep the issue open until this is done.

pedrom34, 2021-01-27:
Hey,

I'm sorry, I have the exact same issue.
So, if I read everything correctly, I have to use buster instead of stretch in order for medusa to work again?

And that requires a fresh install. Am I right?

Thanks !

Joulinar, 2021-01-27:
You can have a look to this post https://github.com/MichaIng/DietPi/issues/3991#issuecomment-749697021

pedrom34, 2021-01-27:
OK, I'll do a backup and try.

Joulinar, 2021-01-27:
If possible try to clone your SD card

pedrom34, 2021-01-28:
Well. Medusa works on buster, but I have other issues now 😢 
It's not really related to dietpi, but to other programs I installed on my own, so I'll do a fresh install.
Thanks for the tip anyway 😉

MichaIng, 2021-01-31:
I just verified, if fails with Python 3.5 and Python 2 on Debian Stretch, so we need to disable it:
- Done with: https://github.com/MichaIng/DietPi/commit/6e76c3c0a13b2fae47997eda99c89b5c43ef30f6
- Changelog: https://github.com/MichaIng/DietPi/commit/556756d0a4690a5ddff1d4a9933df49a4600e2a0

closed 2021-02-14 by MichaIng