MichaIng/DietPi #3992: Cloudshell not show Pihole Stats

black00019, 2020-12-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster
- Kernel version | Linux DietPi 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
- SBC model | RPi 3 Model B+ (armv7l)
- Power supply used | (5V 1A RAVpower)
- SDcard used | (Kingston)

#### Additional Information (if applicable)
- Software title | Dietpi-CloudShell & PiHole
- Was the software title installed freshly or updated/migrated? Updated
- Can this issue be replicated on a fresh installation of DietPi?
Maybe
- Bug report ID | 63230cff-167a-4eba-8610-2a9d2687a1e2

#### Steps to reproduce

1. Launch Dietpi-Cloudshell
2. Select Pihole stats
3. Restart Dietpi-cloudshell

#### Expected behaviour
View a Pihole stats on my display with cloudshell

#### Actual behaviour
The pihole stats not show and automatic deselected by cloudshell settings

#### Extra details

ravenclaw900, 2020-12-22:
Hello,
As of Pi-hole v5 a database file is used to log queries.  This scene was built with earlier versions in mind, which used text files instead.

black00019, 2020-12-22:
> Hello,
> As of Pi-hole v5 a database file is used to log queries. This scene was built with earlier versions in mind, which used text files instead. Since DietPi doesn't find the text file, it automatically deselects the scene.

text file is
/var/log/pihole.log
or
/etc/pihole/gravity.list ??

ravenclaw900, 2020-12-22:
The file is `/etc/pihole/gravity.list`.

MichaIng, 2020-12-22:
We'll use the FTL interface directly instead of scraping logs files or the database. That's what the "Faster Than Light" interface is made for 😉.

Here the API: https://docs.pi-hole.net/ftldns/telnet-api/
That can be used via socket file directly as well, e.g. see how Chronometer does it: https://github.com/pi-hole/pi-hole/blob/master/advanced/Scripts/chronometer.sh#L14-L40
Or PADD: https://github.com/pi-hole/PADD/blob/master/padd.sh#L119-L148

MichaIng, 2021-02-10:
Pi-hole stats have been fixed: https://github.com/MichaIng/DietPi/commit/8e766d310b0b8e9e6ed42c7b12031eee4dad9cf8

Additionally, in case of failure, the first output line, normally showing time, CloudShell version or uptime, is replaced with an error message, based on step at which the Pi-hole FTL connection failed. So e.g. if Pi-hole has not started yet, if it has stopped again (gracefully), or if it crashed or had an internal error so that connection to the FTL socket failed. In case of failure, the previously obtained data are still shown, below the error message.

I mainly added this for debugging reasons our side, but I think it's useful for users as well.

closed 2021-02-28 by MichaIng