MichaIng/DietPi #3994: Diet

ravenclaw900, 2020-12-23:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```text
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=35
G_DIETPI_VERSION_RC=-1
G_GITBRANCH='box86'
G_GITOWNER='ravenclaw900'
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
`buster`
- Kernel version | `uname -a`
`Linux rpiDev 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 4 Model B (armv7l)`
- Power supply used | (EG: 5V 1A RAVpower)
`CanaKit`
- SDcard used | (EG: SanDisk ultra)
`Samsung EVO+ 32GB`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
LXDE
- Was the software title installed freshly or updated/migrated?
Freshly installed
- Can this issue be replicated on a fresh installation of DietPi?
Yes, with changes to `DEV_GITOWNER`

#### Steps to reproduce

1. Try to install LXDE with a different `DEV_GITOWNER`

#### Expected behaviour

- Everything installs successfully, with the install using `$G_GITOWNER`

#### Actual behaviour

- Many `G_THREAD` errors.

C0D3-M4513R, 2021-02-05:
This should be closed @MichaIng . This was fixed by #4015 , but github didn't close them.

MichaIng, 2021-02-05:
As mentioned in the other issue, the commits need to be merged into the default/master branch to have related issues closed. But in this case I agree, it's trivial enough and does not require any discussion e.g. during beta phase 👍.

PR #4015 (2021-02-03): Apply desktop Customisations to all users
**Status**: Testing
- [x] LXDE
- [x] MATE
- [x] LXQt
- [x] Xfce
- [x] GNUstep

Also please link the issues below.
**Reference**: fixes #4014 fixes #3994
closed 2021-02-05 by MichaIng