MichaIng/DietPi #3996: DietPi has encountered an error

LeatherMolly, 2020-12-23:
#### Details:
- Date           | Wed Dec 23 05:35:47 CET 2020
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux DietPizza 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | ping -c 1 -W 5 one.one.one.one
- Exit code      | 2
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
ping: one.one.one.one: Temporary failure in name resolution
```

Joulinar, 2020-12-23:
Hi,

Looks like DNS issues on your system. Can you check in `dietpi-config` what DNS server is used?

MichaIng, 2020-12-23:
From your post on the other forum it seems that you mixed two configurations:
- DietPi-Software installs Unbound (when Pi-hole is or was installed) to listen on port `5353`.
- But you created additionally `/etc/unbound/unbound.conf.d/pi-hole.conf` to listen to and tested DNS on port `5335`.

Please decide which config you want to use and either:
- `rm /etc/unbound/unbound.conf.d/dietpi*.conf` => Unbound will listen on port 5335 as the blog/guide suggests it.
- or: `rm /etc/unbound/unbound.conf.d/pi-hole.conf` => Unbound will listen on port 5353, as DietPi configures it.

Those port settings seem to sometimes override each other, sometimes not, so having multiple port directives in place seems to not work reliably. Since port 5353 is [mDNS](https://wikipedia.org/wiki/Multicast_DNS), port 5335 is actually better for Unbound as a default. So I'd go with the first option and use the `pi-hole.conf` instead.

Another topic: It seems that you configured DietPi to use it's own Pi-hole for DNS resolution. Please do not do that, as on Pi-hole or Unbound updates, or if anything else goes wrong, the system can lock out itself from network access. The server itself should use a regular upstream DNS provider, so you are able to stop every service without loosing network capabilities on the server itself.

closed 2021-02-14 by MichaIng