MichaIng/DietPi #3997: Pi-Hole not forwarding to Unbound

trinsic, 2020-12-23:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster

- Kernel version | Linux towncrier 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux

- SBC model |RPi 3 Model B (armv7l)

- Power supply used | RPi supplied adapter

- SDcard used | Samsung EVO 32GB

#### Additional Information (if applicable)
- Software title | Unbound and Pi-Hole
- Was the software title installed freshly or updated/migrated? Unbound installed to an existing Pi-hole system

#### Steps to reproduce
1. Installed Unbound on existing Pi-Hole system

#### Expected behaviour
1. Forward DNS to local Unbound server

#### Actual behaviour
1. Pi-Hole still forwarding DNS entries to old upstream server

Today I noticed when I installed Unbound that Pi-Hole refused to forward anything to Unbound on port 5353 under Pi-Hole's Query Log. 

I restarted all systems, flushed caches, restarted the Pi-Hole DNS resolver, as well as Pi-Hole itself and tried to find out what was wrong. Pi-Hole's DNS tab showed 127.0.0.1#5353 but was not sending anything to it, I confirmed through dig that Unbound was working on port 5353. However from the Query Log it was still sending to my old upstream selection even without it being visible in the DNS tab.

Solution: I had to manually go into the Pi-Hole DNS tab and hit the Save button. After that I tried DNS leak testing and it now showed my public IP and the Query Log shows that the upstream is sending to Unbound as well.

Joulinar, 2020-12-23:
Hi,

This is already know issue. https://github.com/MichaIng/DietPi/issues/2409#issuecomment-749258433

trinsic, 2020-12-23:
Ahh you had the same issue. Good to know and easy to solve.

closed 2020-12-23 by trinsic