MichaIng/DietPi #3998: Dietpi-Nordvpn | command missing after software install

aermak, 2020-12-23:
- DietPi version | 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | 10.7
- Kernel version | Linux DietPi 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
- SBC model | RPi 3 Model B (armv7l)
- Power supply used | 
- SDcard used | 

- Software title | Dietpi-NordVPN
- Was the software title installed freshly or updated/migrated? fresh install
- Can this issue be replicated on a fresh installation of DietPi? yes
- Bug report ID | b09b501d-d9a2-4a5a-992f-8e1042eab75f

#### Steps to reproduce
1. Fresh image installed on new SDcard
2. Install dietpi-nordvpn
3. reboot
4. type dietpi-nordvpn and get | -bash: dietpi-nordvpn: command not found

#### Expected behaviour
dietpi-Nordvpn menu popping up

#### Actual behaviour
nothing, openvpn is installed and running but the nordvpn setup/gui command doesn't exist

Joulinar, 2020-12-23:
Hi,

Many thanks for your message. NordVPN could be open via `dietpi-launcher` > `DietPi-NordVPN` or directly `/boot/dietpi/misc/dietpi-nordvpn`

aermak, 2020-12-23:
@Joulinar thank you, maybe update the after install message, closing this.

MichaIng, 2020-12-23:
You need to first configure it, before the alias is available. It is only loaded if the service file has been created. But it does not make much sense to have the condition in this case. When it has not been "installed" yet, the script will recognise it and even offer you to install it right away. Hence I changed it: https://github.com/MichaIng/DietPi/commit/12052633fb99256ee695438f5c988cef8a62bba6

aermak, 2020-12-23:
- Command: curl -sSfLO https://downloads.nordcdn.com/configs/archives/servers/ovpn.zip                              │
                                 │  - Exit code: 7                                                                                                      │
                                 │  - DietPi version: v6.34.3 (MichaIng/master) | HW_MODEL: 3 | HW_ARCH: 2 | DISTRO: 5                                  │
                                 │  - Image creator: DietPi Core Team                                                                                   │
                                 │  - Pre-image: Raspbian Lite                                                                                          │
                                 │  - Error log:                                                                                                        │
                                 │ curl: (7) Failed to connect to downloads.nordcdn.com port 443: Connection refused

Joulinar, 2020-12-23:
Maybe NordVPN download server is not reachable. Did you try downloading the archive on you desktop computer?

MichaIng, 2020-12-23:
Also try to run the command manually from console to get some more output:
```
curl -LO https://downloads.nordcdn.com/configs/archives/servers/ovpn.zip
```
Works fine here 🤔.

aermak, 2020-12-23:
curl: (28) Resolving timed out after 2000 milliseconds
can't resolve anything since i installed nordvpn
removing it again

Joulinar, 2020-12-23:
Well NordVPN should not be online if you don't download the server archive. As well you could disconnect it if needed

MichaIng, 2020-12-23:
> curl: (28) Resolving timed out after 2000 milliseconds

Strange, default Linux DNS resolving timeout is 5 seconds with two attempts. Has this changed on your system? `cat /etc/resolv.conf`

aermak, 2020-12-23:
weird, have no outside connection on my Dietpi atm, can't ping 9.9.9.9 can't resolve anything only have lan access it seems
cat /etc/resolv.conf
nameserver 127.0.0.1
nameserver 192.168.0.1

aermak, 2020-12-23:
Checking network connectivity                                                                                        │
                                 │  - Command: ping -c 1 -W 10 9.9.9.9                                                                                  │
                                 │  - Exit code: 1                                                                                                      │
                                 │  - DietPi version: v6.34.3 (MichaIng/master) | HW_MODEL: 3 | HW_ARCH: 2 | DISTRO: 5                                  │
                                 │  - Image creator: DietPi Core Team                                                                                   │
                                 │  - Pre-image: Raspbian Lite                                                                                          │
                                 │  - Error log:                                                                                                        │
                                 │ PING 9.9.9.9 (9.9.9.9) 56(84) bytes of data.                                                                         │
                                 │                                                                                                                      │
                                 │ --- 9.9.9.9 ping statistics ---                                                                                      │
                                 │ 1 packets transmitted, 0 received, 100% packet loss, time 0ms

Joulinar, 2020-12-23:
Inside your network you can ping anything? Like your router?

MichaIng, 2020-12-23:
> nameserver 127.0.0.1

Huh, do you have a local DNS resolver installed, Pi-hole or Unbound? Ah or is that the way how OpenVPN forces the NordVPN DNS server, with OpenVPN as local resolver?

Is NordVPN connection active currently or not? If not, remove the first line in `/etc/resolv.conf`. I'll check how OpenVPN applies the local DNS resolver.

aermak, 2020-12-23:
So looking at my FW logs found a monitoring Pi like device doing some arp poisoning, stopped that and my DietPi can talk to the interwebs again. Also NordVPN is back in business, Closing this and will try to stop tinkering with my Pi till next year. Thanks everyone, xmas donation on the way!

closed 2020-12-23 by aermak

MichaIng, 2020-12-23:
> Closing this and will try to stop tinkering with my Pi till next year.

At least if there is a chance that it causes stress. I'll take the pandemic lock-down Christmas/holidays to tinker even more with my devices, but only since nothing really depends on it 😄.

I just checked and OpenVPN does not touch `/etc/resolv.conf`. Generally, if that Pi serves as DNS resolver, e.g. via Pi-hole or Unbound, I'd not make it using it's own services. If Pi-hole and/or Unbound need to be updated, it could easily get stuck when shutting down its own resolver. Better to keep it using an upstream DNS, in case of VPN connection one from NordVPN. Now I'm wondering if/how that isn't done when connection via OpenVPN to NordVPN 🤔. I'll play a bit more with it, but that's clearly OOT.

aermak, 2020-12-23:
this is my status with vpn running, using the router as dns and my other Pihole

root@DietPi:~# dietpi-nordvpn status
Connected - Sent = 0 MiB | Received = 0 MiB
root@DietPi:~# dietpi-nordvpn status
Connected - Sent = 8 MiB | Received = 519 MiB
root@DietPi:~# cat /etc/resolv.conf 
nameserver 192.168.0.1
nameserver 192.168.0.2

Joulinar, 2020-12-23:
Looks like it's working, correct?

hrvstr, 2021-01-20:
> You need to first configure it, before the alias is available. It is only loaded if the service file has been created. But it does not make much sense to have the condition in this case. When it has not been "installed" yet, the script will recognise it and even offer you to install it right away. Hence I changed it: [1205263](https://github.com/MichaIng/DietPi/commit/12052633fb99256ee695438f5c988cef8a62bba6)

This does not seem to work for me.

- I installed DietPi-NordVPN via DietPi-Software
- Run `/boot/dietpi/misc/dietpi-nordvpn` to configure the settings
- Connect to VPN
- Check if service is running: `Active: active (exited)`

But I still get this error when trying to run the client:

`-bash: dietpi-nordvpn: command not found`

MichaIng, 2021-01-21:
@hrvstr 
Can you try the following:
```
[[ -f '/etc/systemd/system/dietpi-nordvpn.service' ]] && alias dietpi-nordvpn='/boot/dietpi/misc/dietpi-nordvpn'
dietpi-nordvpn
```
Note that it becomes active only from the next bash session on (logout/login or reboot or new SSH session), after you connected the first time. If the service is active or not, is not relevant then anymore, only the service file needs to exist.

However, what you think about we just loading this alias always? If it's not actually installed, the script recognises it and offers to install right away.

Joulinar, 2021-01-21:
> However, what you think about we just loading this alias always? If it's not actually installed, the script recognises it and offers to install right away.

That's a good idea

MichaIng, 2021-01-21:
Ah lol, I made the change already as the solution for this issue 😄: https://github.com/MichaIng/DietPi/commit/12052633fb99256ee695438f5c988cef8a62bba6