MichaIng/DietPi #4000: Trying to reinstall dietpi + pihole with PNY PNY SDXC 64GB P-SDU64GV31100PRO-GE

asx441, 2020-12-23:
#### Details:
- Date           | Wed Dec 23 19:30:42 GMT 2020
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | Linux DietPi 5.4.79-v8+ #1373 SMP PREEMPT Mon Nov 23 13:32:41 GMT 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | mount /var/log
- Exit code      | 1
- Software title | DietPi-Software

mount: /etc/fstab: parse error at line 28 -- ignored
mount: /var/log: can't find in /etc/fstab.

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | 10.7
- Kernel version | Linux DietPi 5.4.79-v8+ #1373 SMP PREEMPT Mon Nov 23 13:32:41 GMT 2020 aarch64 GNU/Linux
- SBC model | RPi 3 Model B+ (aarch64)
- Power supply used | 5V 1A RAVpower)
- SDcard used | PNY SDXC 64GB P-SDU64GV31100PRO-GE

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? Yes. It is a fresh installation. 

- Bug report ID | 22834416-5edf-4f5c-b554-5347820815bf

[  OK  ] DietPi-Software | systemctl enable dietpi-ramlog
[  OK  ] DietPi-Software | Storing /var/log metadata to disk
[  OK  ] DietPi-Software | Cleaning /var/log mountpoint
[FAILED] DietPi-Software | Mounting tmpfs to /var/log
 - Command: mount /var/log
[FAILED] DietPi-Software | Mounting tmpfs to /var/log
 - Command: mount /var/log
[FAILED] DietPi-Software | Mounting tmpfs to /var/log
 - Command: mount /var/log
[FAILED] DietPi-Software | Mounting tmpfs to /var/log
 - Command: mount /var/log
[FAILED] DietPi-Software | Mounting tmpfs to /var/log
 - Command: mount /var/log

MichaIng, 2020-12-24:
Many thanks for your report, an merry Xmas/holidays. It looks like `/etc/fstab` is malformed. Could you please run and paste the following:
```
ls -l /etc/fstab
cat /etc/fstab
```

asx441, 2021-01-10:
> Many thanks for your report, an merry Xmas/holidays. It looks like `/etc/fstab` is malformed. Could you please run and paste the following:
> 
> ```
> ls -l /etc/fstab
> cat /etc/fstab
> ```

I found the problem - and it was like usual between the chair and the monitor = myself. 
I downloaded the 64-bit version of DietPie instead of the standard 32-Bit. After re-installing it with the new standard image it all worked out well. 

This problem was not related to a malformating problem, because i formated it fresh from scratch.

Joulinar, 2021-01-10:
hmm the 64bit image should be working as well. Even if it has a BETA flag.

MichaIng, 2021-01-10:
That is true, RPi 3 is 64-bit capable, even RPi 2 v1.2 is. However, it has still a few known issues (limited software support), so if the 32-bit image works fine now, there is no reason to change anything 🙂.

> This problem was not related to a malformating problem, because i formated it fresh from scratch.

Also a fresh image can have file system errors, if something went wrong while flashing or the card itself has bad blocks or on first boot something went wrong. But indeed the typical reason is a power loss or unclean removed file system of course.

closed 2021-02-15 by MichaIng