MichaIng/DietPi #4002: Did i set zram properly?

Dr0bac, 2020-12-24:
On fresh dietpi install and before first boot i opened dietpi.txt and edited few lines.

For swap file i entered

`AUTO_SETUP_SWAPFILE_SIZE=1`

and

`AUTO_SETUP_SWAPFILE_LOCATION=zram`

When i go to dietpi_config/Advanced Options

I get this result

[image]

Is this correct output for zram configuration?

Also, where exactly zram keep swapfile? If i understand correctly swap file is located on ram memory and not on sd card or external hdd?

MichaIng, 2020-12-24:
Many thanks for your report. I'll try to replicate.

The zram implementation does not create a swap "file" but uses the zram block device itself as swap space, similar to a swap partition. Indeed the `dietpi-config` output doesn't look like it worked correctly but default `/var/swap` file was used.

Dr0bac, 2020-12-24:
````
root@DietPi:~# swapon
NAME      TYPE SIZE USED PRIO
/var/swap file   1G   0B   -2
````

how can i change to zram method?

Joulinar, 2020-12-24:
you can try following `/boot/dietpi/func/dietpi-set_swapfile 1 zram`

Dr0bac, 2020-12-24:
Thank you. success!

````
root@DietPi:~# swapon
NAME       TYPE      SIZE USED PRIO
/dev/zram0 partition 487M   0B   -2
````

MichaIng, 2020-12-25:
There was indeed a bug in the logic of the script that dropped zram/zram0 dietpi.txt entries. Actually this was to prevent the attempt of removing the /dev/zram0 kernel device as if it was an ordinary file, but the dietpi.txt entry was not used at all then, hence was reset to default during first boot setup.
Fixed with: https://github.com/MichaIng/DietPi/commit/8eb0973a7dd8f3c6a7e58b8624a8509eea37fcf2
Changelog: https://github.com/MichaIng/DietPi/commit/c2a699612530c727aac3565e8c29b40b2152271b

closed 2021-02-14 by MichaIng