MichaIng/DietPi #4003: Fan is always running in idle mode

cankirca, 2020-12-24:
Hello,

I have very strange issue. Using latest Raspberry with latest DietPi, original adaptor, heatsinc and fan. I'm using headless mode.
Today I've install heatsync and fan. My system is 35°C, but unfortunately my fan is always spinning/working. How can I set my fan to work when the system is so hot? Details are below:
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH=’master’
Distro=Buster
Kernel=Linux hostname 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
SBC Model=RPi 4 Model B (armv7l)
Power supply=Original Type-C adaptor.
SDCard= Sandisk ultra
Software Title=LEMP, PHPMyAdmin, OpenSSH

Joulinar, 2020-12-24:
Hi,

marry Xmas and many thanks for your message. If this is a normal Fan just connected to the board without anything, it's quite normal to have it spinning all the time. Usually you need some additional hardware (transistor) to have ot controlled.

https://howchoo.com/g/ote2mjkzzta/control-raspberry-pi-fan-temperature-python

cankirca, 2020-12-25:
Hello,

Thank you for your answer. In Raspberry official webpage, I red
following explanation and steps about fan controling. Is it not
possible with DietPi?
"Once you have connected your fan, by default it will be on whenever
your Raspberry Pi is running. You can configure it to turn on only
when your Raspberry Pi reaches a temperature that you choose."
URL: https://www.raspberrypi.org/products/raspberry-pi-4-case-fan/

Thank you.

Joulinar, 2020-12-25:
do you have exactly this fan? Or something else?

cankirca, 2020-12-25:
Yes, i have it.

cankirca, 2020-12-25:
There is no Fan option in the config/performance as stated on the page.

ravenclaw900, 2020-12-25:
DietPi does not use the Raspberry Pi configuration tool, and as such does not have the same options.  However, it looks like the fan might work with the Python script in the link that @Joulinar  posted.  You could start on step 5, and use the blue wire as the transistor.  If you do so, make sure to install the `gpiozero` Python library. `apt install python3-gpiozero`.

cankirca, 2020-12-25:
Hello again,
 
 Thanks for your answers.
 I don't want to use an other device to control the fan, it seem that I have to buy transistor am I right?
 Also, if I install raspi original, would it be possible to use config tool via SSH? Because I'll continue to use in headless mode.
 If I to be honest, Dietpi is awesome for me and it's software packages are well-configured. If I find an other way to decrease fan working time, I don't want to reinstall an other OS.

@MichaIng do you have any suggestion please?

MichaIng, 2020-12-25:
Here is the relevant code from raspi-config: https://github.com/RPi-Distro/raspi-config/blob/master/raspi-config#L1205-L1287
So the following seems to be all what is needed:
```
echo 'dtoverlay=gpio-fan,gpiopin=14,temp=60' >> /boot/config.txt
```
With GPIO pin 14 (default, matches the instructions you linked) and starting the fan at 60°C, which is the lowest according to raspi-config. On RPi 4 with DietPi by default throttling starts at ~70°C (full throttling at 75°C), so 70°C would be enough to assure that throttling does never really kick in.

cankirca, 2020-12-25:
Hello,
 
 Thank you for the answer.
 @MichaIng , I've applied above command, but my fan is still spinning. There is no running app.
Dietpi shows: Temperature | 35’C : 95’F (Cool runnings)
Here the command out of 'cpu'
root@hostname:~# cpu 
───────────────────────────────────────────────────── 
DietPi CPU Info 
───────────────────────────────────────────────────── 
Architecture | armv7l 
Temperature | 35’C : 95’F (Cool runnings) 
Governor | ondemand 
Throttle up | 50% CPU usage 
Current Freq Min Freq Max Freq 
CPU0 | 1500 MHz 600 MHz 1500 MHz 
CPU1 | 1500 MHz 600 MHz 1500 MHz 
CPU2 | 1500 MHz 600 MHz 1500 MHz 
CPU3 | 1500 MHz 600 MHz 1500 MHz

Is it expected behaviar?
BTW: I ran 'shutdown' command to turn-off the unit, but fan is still working :-)

Thank you so much.

Joulinar, 2020-12-25:
@MichaIng 
Does the GIPO needs to be activated?

MichaIng, 2020-12-25:
The device tree overlay should enable it. @cankirca you did a reboot, didn't you?

cankirca, 2020-12-25:
I enter the command again and type 'reboot', fan stopped, after few seconds spin again, but after 2 seconds stopped again.
Now it is not spinning, system Temperature | 40’C : 104’F (Optimal temperature) I think this is expected result am I right?
When my system Temperature goes to 60°C , it start to spin again am I right?

Also @MichaIng, I have connected my fan by getting a help from a sided person as a visually impaired user, how can I to be sure that pins are correct?
Let me try to explain section that I've connected.
As you know, there are 3 cables comes with fan. 1 has double plug, other one single.
My double cables are connected from the second and 3rd ping from the right side of the unit, other cable is connected to the number 4. SO cables are plugged 2-3-4 pins from the right side, near the corner of the unit when I start counting from sd card durection.

Not sure told correctly 👍 
Where is pin 14?

Joulinar, 2020-12-25:
@MichaIng 
I just ordered one for testing purposes 😄

MichaIng, 2020-12-25:
Sounds like you did it as advised here: The upper two are for power, the third is for control and it's GPIO pin 14. To test, you could run `dietpi-config` > **Tools** > **Stress Test** to head it up.
[image]
Here the official docs about GPIO pins: https://www.raspberrypi.org/documentation/usage/gpio/README.md

cankirca, 2020-12-27:
Thank you for your awesome support, it seems that it works as expected now!
Hope that show this function in the dietpi-config soon.

Joulinar, 2020-12-30:
@MichaIng 
today I got my thin and noisy RPi Fan. I did some reengineering to have it fitting into my case 🤣 

[image]

The value for the fan to kick in can be lowered as you like. It was running with 60, 55, 50 as well as 45 °C. Once fan started, it always will try to lower temp by 10°C before it stop. 

- 60°C that's a good value for normal usage
- 55°C is quite ok as long as the system is not in use. Fan is not spinning that often
- 50°C there the fan is kicking in quite often as it's above idle temp
- 45°C as starting temp is useless as it is quite impossible to reach 35°C and it never stop in this case. 

To measure temp I used Influxdb, Grafana and Telegraf (which is still missing on the software repository 😜 ). That's the only thing running on my RPi4B 64bit

[image]

MichaIng, 2020-12-30:
Awesome test, great work. Also good to know that any temperature works, as raspi-config only allows to set 60 - 120 °C so I was not sure whether this is a hard kernel module limit or only thought a lowest reasonable value from RPi devs few. As we by default ship with a temperature limit of 75 °C (means the CPU is starting to be throttled at 70°C with max throttling at 75°C, 60°C indeed seems to be a reasonable value to start with as long as there are no special needs.

You're right, Telegraf needs to be added to complete the TIG stack (Telegraf-InfluxDB-Grafana) 👍.

Joulinar, 2020-12-30:
After 4 hour run time, 60 °C is still a good value on my demo system. The fan is activated 5-6 times a hour for around 30 secounds

[image]

Should be fine to follow RPi devs on their settings.

Joulinar, 2021-01-02:
@MichaIng 
I did a test and fan is working on my RPi3B+ as well.

Joulinar, 2021-01-06:
For the records, Fan is working as well on RPi1 but it require different temp settings. It's quite hard to get the RPi1 above 55°C. It require a CPU exhausting application like PaperMC. Idle is around 50°C.

MichaIng, 2021-01-06:
Jep, my RPi 2 idles around 32 °C and I never saw it above 50 °C. Not the temps need to be different, but it doesn't make sense to put a fan into anything below RPi 3, as long as you do not want to win an overclocking context 😄.

seancarney15, 2022-03-07:
@MichaIng  thank you very much, sorted my fan out (was driving the wife nuts)

MichaIng, 2022-03-07:
Applying a proper temperature, like 60 °C, so that the fan really only turns on to prevent thermal throttling, is not an option? Like here: https://github.com/MichaIng/DietPi/issues/4003#issuecomment-751254845

There is no hysteresis attribute, that would be actually nice to have it less often switching on and off. Though, in Joulinar's case it cooled down to 50 °C (10 °C below defined temperature?), but it heats up so quickly close to 60 °C that even starting at 40 °C I guess it wouldn't change much.

Another approach is to add some heat capacitor into the case, like a solid metal block, at best a passive cooling block for the SoC or other chips of course. That would absorb/store some of the heat so that heating up as well as cooling down takes more time and the fan hence has longer on and off phases, which is usually less annoying 😉.

zumbiepig, 2026-08-27:
Bumping since it's been over 4 years since the last comment here. Including the one-liner `dtoverlay=gpio-fan,gpiopin=14,temp=60000` in `/boot/firmware/config.txt` still works. Is it possible for you to ship DietPi with that line included by default in config.txt? Even further, add an actual dietpi-config menu for configuring the fan (looking at the code used in raspi-config, it seems pretty simple)?

MichaIng, 2026-08-27:
Enabling an overlay by default for a device which is not always attached does not make sense. It is pretty simple and well documented, hence not a priority for me.

Generally, I'd probably prefer a more generic new script to allow configuring PWM fans via sysfs on all hardware, not an RPi-only extra config entry. But we are open to PRs either way, of course.

Before someone starts with the `gpio-fan` overlay, doesn't this actually work with the `pwn-gpio-fan` overlay on the Raspberry Pi 4 (and earlier), to assign multiple trip points? Or does this work on RPi 5 only?

zumbiepig, 2026-08-27:
i dont know what pwn-gpio-fan is, nor do i know how to make a script to configure other fans. but i dont mind making a quick PR for configuring the gpio-fan overlay. i only have a raspberry pi 4 btw, so i cant test on rpi 5

> It is pretty simple and well documented, hence not a priority for me.

Could I ask where this is documented? Based on a quick search in the raspberry pi docs I couldnt find `gpio-fan` or `pwn-gpio-fan`

zumbiepig, 2026-08-29:
By the way to everyone subscribed on this thread, the 10°C below the target temp can be changed to something like 5°C (which is what the raspberry pi 5 fan uses)

```
Name:   gpio-fan
Info:   Configure a GPIO pin to control a cooling fan.
Load:   dtoverlay=gpio-fan,<param>=<val>
Params: gpiopin                 GPIO used to control the fan (default 12)
        temp                    Temperature at which the fan switches on, in
                                millicelsius (default 55000)
        hyst                    Temperature delta (in millicelsius) below
                                temp at which the fan will drop to minrpm
                                (default 10000)
```

closed 2026-11-14 by MichaIng

MichaIng, 2026-11-14:
Done in #8269. It can now be all configured with the new `dietpi-fan_control` script:
* `gpio-fan` overlay as mentioned above, including the hysteresis.
* `pwm-gpio-fan` for multiple trip points, which uses a software PWM.
* The new `pwm-fam` overlay added by @zumbiepig to the official RPi kernel package recently, which uses a hardware PWM for smoother fan drive. But it takes the PWM away from analogue onboard audio, hence not advised if you want to use the 3.5mm jack. It does not affect HDMI audio, any I2S HAT, USB DAC or similar, however.
* The dedicated PWM fan port of the RPi 5 can be configured likewise.