MichaIng/DietPi #4004: Radarr Down After Latest Update

shlatchz, 2020-12-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`

`9.13`

- Kernel version | `uname -a`

`Linux DietPi 4.9.0-14-amd64 #1 SMP Debian 4.9.246-2 (2020-12-17) x86_64 GNU/Linux`

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

`Native PC (x86_64)`

- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)

`Radarr`

- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

`3b0cd5c5-8000-40c6-ac83-14972cc6877b`

#### Steps to reproduce

ran `dietpi-update`ie

#### Expected behaviour

#### Actual behaviour

when running `dietpi-services status`

```
[FAILED] DietPi-Services | ● radarr.service - Radarr Daemon (DietPi)
   Loaded: loaded (/etc/systemd/system/radarr.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/radarr.service.d
           └─dietpi-process_tool.conf, dietpi-services_edit.conf
   Active: failed (Result: exit-code) since Fri 2020-12-25 13:52:28 IST; 1min 13s ago
 Main PID: 24973 (code=exited, status=137)

Dec 25 13:52:28 DietPi systemd[1]: radarr.service: Unit entered failed state.
Dec 25 13:52:28 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Dec 25 13:52:28 DietPi systemd[1]: radarr.service: Service hold-off time over, scheduling restart.
Dec 25 13:52:28 DietPi systemd[1]: Stopped Radarr Daemon (DietPi).
Dec 25 13:52:28 DietPi systemd[1]: radarr.service: Start request repeated too quickly.
Dec 25 13:52:28 DietPi systemd[1]: Failed to start Radarr Daemon (DietPi).
Dec 25 13:52:28 DietPi systemd[1]: radarr.service: Unit entered failed state.
Dec 25 13:52:28 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
```

#### Extra details

Joulinar, 2020-12-25:
Hi,

Many thanks for your message. Pls can you try to start Radarr manually as follow.

`sudo -u radarr /opt/radarr/Radarr -nobrowser -data=/mnt/dietpi_userdata/radarr`

shlatchz, 2020-12-25:
Hi @Joulinar

Ran the command and nothing changed..

Joulinar, 2020-12-25:
Pls can you share the output of this command

shlatchz, 2020-12-25:
```
root@DietPi:/var/log/radarr# dietpi-update sudo -u radarr /opt/radarr/Radarr -nobrowser -data=/mnt/dietpi_userdata/radarr

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | No update required, your DietPi installation is already up to date:
[ INFO ] DietPi-Update | Current version : v6.34.3
[ INFO ] DietPi-Update | Latest version  : v6.34.3
[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://downloads.plex.tv/repo/deb public InRelease
Hit:2 https://apt.sonarr.tv master InRelease
Ign:3 https://deb.debian.org/debian stretch InRelease
Hit:4 https://download.mono-project.com/repo/debian stretch InRelease
Hit:5 https://deb.debian.org/debian stretch-updates InRelease
Hit:6 https://deb.debian.org/debian-security stretch/updates InRelease
Hit:7 https://deb.debian.org/debian stretch-backports InRelease
Hit:8 https://deb.debian.org/debian stretch Release
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | No APT upgrades were found, not creating file: /run/dietpi/.apt_updates
```

Joulinar, 2020-12-25:
Pls don't put `dietpi-update` before the command. Use it as I listed above.

shlatchz, 2020-12-25:
Sorry! didn't notice I clicked back and wrote on top of a prev command. here's the output:

```
root@DietPi:/var/log/radarr# sudo -u radarr /opt/radarr/Radarr -nobrowser -data=/mnt/dietpi_userdata/radarr
[Info] Bootstrap: Starting Radarr - /opt/radarr/Radarr.dll - Version 3.0.1.4259
[Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/radarr]
[Info] Router: Application mode: Interactive
[Info] MigrationController: *** Migrating data source=/mnt/dietpi_userdata/radarr/radarr.db;cache size=-20000;datetimekind=Utc;journal mode=Wal;pooling=True;version=3 ***
[Info] MigrationController: *** Migrating data source=/mnt/dietpi_userdata/radarr/logs.db;cache size=-20000;datetimekind=Utc;journal mode=Wal;pooling=True;version=3 ***
[Info] MigrationLoggerProvider: *** VersionMigration migrating ***
[Info] MigrationLoggerProvider: *** VersionUniqueMigration migrating ***
[Info] MigrationLoggerProvider: *** VersionDescriptionMigration migrating ***
[Info] MigrationLoggerProvider: *** 1: InitialSetup migrating ***
[Info] InitialSetup: Starting migration to 1
[Info] MigrationLoggerProvider: *** 104: add_moviefiles_table migrating ***
[Info] add_moviefiles_table: Starting migration to 104
[Info] MigrationLoggerProvider: *** 105: fix_history_movieId migrating ***
[Info] fix_history_movieId: Starting migration to 105
[Info] MigrationLoggerProvider: *** 106: add_tmdb_stuff migrating ***
[Info] add_tmdb_stuff: Starting migration to 106
[Info] MigrationLoggerProvider: *** 107: fix_movie_files migrating ***
[Info] fix_movie_files: Starting migration to 107
[Info] MigrationLoggerProvider: *** 108: update_schedule_intervale migrating ***
[Info] update_schedule_intervale: Starting migration to 108
[Info] MigrationLoggerProvider: *** 109: add_movie_formats_to_naming_config migrating ***
[Info] add_movie_formats_to_naming_config: Starting migration to 109
[Info] MigrationLoggerProvider: *** 110: add_phyiscal_release migrating ***
[Info] add_phyiscal_release: Starting migration to 110
[Info] MigrationLoggerProvider: *** 111: remove_bitmetv migrating ***
[Info] remove_bitmetv: Starting migration to 111
[Info] MigrationLoggerProvider: *** 112: remove_torrentleech migrating ***
[Info] remove_torrentleech: Starting migration to 112
[Info] MigrationLoggerProvider: *** 113: remove_broadcasthenet migrating ***
[Info] remove_broadcasthenet: Starting migration to 113
[Info] MigrationLoggerProvider: *** 114: remove_fanzub migrating ***
[Info] remove_fanzub: Starting migration to 114
[Info] MigrationLoggerProvider: *** 115: update_movie_sorttitle migrating ***
[Info] update_movie_sorttitle: Starting migration to 115
[Info] MigrationLoggerProvider: *** 116: update_movie_sorttitle_again migrating ***
[Info] update_movie_sorttitle_again: Starting migration to 116
[Info] MigrationLoggerProvider: *** 117: update_movie_file migrating ***
[Info] update_movie_file: Starting migration to 117
[Info] MigrationLoggerProvider: *** 118: update_movie_slug migrating ***
[Info] update_movie_slug: Starting migration to 118
[Info] MigrationLoggerProvider: *** 119: add_youtube_trailer_id migrating ***
[Info] add_youtube_trailer_id: Starting migration to 119
[Info] MigrationLoggerProvider: *** 120: add_studio migrating ***
[Info] add_studio: Starting migration to 120
[Info] MigrationLoggerProvider: *** 121: update_filedate_config migrating ***
[Info] update_filedate_config: Starting migration to 121
[Info] MigrationLoggerProvider: *** 122: add_movieid_to_blacklist migrating ***
[Info] add_movieid_to_blacklist: Starting migration to 122
[Info] MigrationLoggerProvider: *** 123: create_netimport_table migrating ***
[Info] create_netimport_table: Starting migration to 123
[Info] MigrationLoggerProvider: *** 124: add_preferred_tags_to_profile migrating ***
[Info] add_preferred_tags_to_profile: Starting migration to 124
[Info] MigrationLoggerProvider: *** 125: fix_imdb_unique migrating ***
[Info] fix_imdb_unique: Starting migration to 125
[Info] MigrationLoggerProvider: *** 126: update_qualities_and_profiles migrating ***
[Info] update_qualities_and_profiles: Starting migration to 126
[Info] MigrationLoggerProvider: *** 127: remove_wombles migrating ***
[Info] remove_wombles: Starting migration to 127
[Info] MigrationLoggerProvider: *** 128: remove_kickass migrating ***
[Info] remove_kickass: Starting migration to 128
[Info] MigrationLoggerProvider: *** 129: add_parsed_movie_info_to_pending_release migrating ***
[Info] add_parsed_movie_info_to_pending_release: Starting migration to 129
[Info] MigrationLoggerProvider: *** 130: remove_wombles_kickass migrating ***
[Info] remove_wombles_kickass: Starting migration to 130
[Info] MigrationLoggerProvider: *** 131: make_parsed_episode_info_nullable migrating ***
[Info] make_parsed_episode_info_nullable: Starting migration to 131
[... 94 lines cut ...]
[Info] MigrationLoggerProvider: *** 179: movie_translation_indexes migrating ***
[Info] movie_translation_indexes: Starting migration to 179
[Info] MigrationLoggerProvider: *** 180: fix_invalid_profile_references migrating ***
[Info] fix_invalid_profile_references: Starting migration to 180
[Info] MigrationLoggerProvider: *** 181: list_movies_table migrating ***
[Info] list_movies_table: Starting migration to 181
[Info] MigrationLoggerProvider: *** 182: on_delete_notification migrating ***
[Info] on_delete_notification: Starting migration to 182
[Info] MigrationLoggerProvider: *** 183: download_propers_config migrating ***
[... cut ...]
[Info] fix_tmdb_duplicates: Starting migration to 186
[Info] MigrationLoggerProvider: *** 187: swap_filechmod_for_folderchmod migrating ***
[Info] swap_filechmod_for_folderchmod: Starting migration to 187
[Info] MigrationLoggerProvider: *** 188: mediainfo_channels migrating ***
[Info] mediainfo_channels: Starting migration to 188
[Info] WebHostController: Listening on the following URLs:
[Info] WebHostController:   http://*:7878
[Info] RadarrBootstrapper: Starting Web Server
[Info] CommandExecutor: Starting 2 threads for tasks.
```

shlatchz, 2020-12-25:
It works now :o

Joulinar, 2020-12-25:
Is it working after a reboot as well?

shlatchz, 2020-12-25:
nope

Joulinar, 2020-12-25:
But starting manually is working?

shlatchz, 2020-12-25:
yep

Joulinar, 2020-12-25:
can you do a reboot and post following

```
systemctl status radarr.service
journalctl -u radarr.service
cat /var/log/radarr/radarr.txt
systemctl cat radarr.service
```

shlatchz, 2020-12-25:
```
root@DietPi:~# systemctl status radarr.service
● radarr.service - Radarr Daemon (DietPi)
   Loaded: loaded (/etc/systemd/system/radarr.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/radarr.service.d
           └─dietpi-process_tool.conf, dietpi-services_edit.conf
   Active: failed (Result: exit-code) since Sat 2020-12-26 00:40:56 IST; 24s ago
  Process: 955 ExecStart=/opt/radarr/Radarr -nobrowser -data=/mnt/dietpi_userdata/radarr (code=exited, status=137)
 Main PID: 955 (code=exited, status=137)

Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=137/n/a
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Unit entered failed state.
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Service hold-off time over, scheduling restart.
Dec 26 00:40:56 DietPi systemd[1]: Stopped Radarr Daemon (DietPi).
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Start request repeated too quickly.
Dec 26 00:40:56 DietPi systemd[1]: Failed to start Radarr Daemon (DietPi).
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Unit entered failed state.
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
root@DietPi:~# journalctl -u radarr.service
-- Logs begin at Sat 2020-12-26 00:40:27 IST, end at Sat 2020-12-26 00:41:20 IST. --
Dec 26 00:40:54 DietPi systemd[1]: Started Radarr Daemon (DietPi).
Dec 26 00:40:54 DietPi Radarr[651]: Failed to create CoreCLR, HRESULT: 0x80004005
Dec 26 00:40:54 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=137/n/a
Dec 26 00:40:54 DietPi systemd[1]: radarr.service: Unit entered failed state.
Dec 26 00:40:54 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Dec 26 00:40:54 DietPi systemd[1]: radarr.service: Service hold-off time over, scheduling restart.
Dec 26 00:40:54 DietPi systemd[1]: Stopped Radarr Daemon (DietPi).
Dec 26 00:40:54 DietPi systemd[1]: Started Radarr Daemon (DietPi).
Dec 26 00:40:54 DietPi Radarr[749]: Failed to create CoreCLR, HRESULT: 0x80004005
Dec 26 00:40:54 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=137/n/a
Dec 26 00:40:54 DietPi systemd[1]: radarr.service: Unit entered failed state.
Dec 26 00:40:54 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Dec 26 00:40:54 DietPi systemd[1]: radarr.service: Service hold-off time over, scheduling restart.
Dec 26 00:40:54 DietPi systemd[1]: Stopped Radarr Daemon (DietPi).
Dec 26 00:40:54 DietPi systemd[1]: Started Radarr Daemon (DietPi).
Dec 26 00:40:55 DietPi Radarr[792]: Failed to create CoreCLR, HRESULT: 0x80004005
Dec 26 00:40:55 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=137/n/a
Dec 26 00:40:55 DietPi systemd[1]: radarr.service: Unit entered failed state.
Dec 26 00:40:55 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Dec 26 00:40:55 DietPi systemd[1]: radarr.service: Service hold-off time over, scheduling restart.
Dec 26 00:40:55 DietPi systemd[1]: Stopped Radarr Daemon (DietPi).
Dec 26 00:40:55 DietPi systemd[1]: Started Radarr Daemon (DietPi).
Dec 26 00:40:55 DietPi Radarr[822]: Failed to create CoreCLR, HRESULT: 0x80004005
Dec 26 00:40:55 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=137/n/a
Dec 26 00:40:55 DietPi systemd[1]: radarr.service: Unit entered failed state.
Dec 26 00:40:55 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Dec 26 00:40:55 DietPi systemd[1]: radarr.service: Service hold-off time over, scheduling restart.
Dec 26 00:40:55 DietPi systemd[1]: Stopped Radarr Daemon (DietPi).
Dec 26 00:40:55 DietPi systemd[1]: Started Radarr Daemon (DietPi).
Dec 26 00:40:56 DietPi Radarr[955]: Failed to create CoreCLR, HRESULT: 0x80004005
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Main process exited, code=exited, status=137/n/a
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Unit entered failed state.
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Service hold-off time over, scheduling restart.
Dec 26 00:40:56 DietPi systemd[1]: Stopped Radarr Daemon (DietPi).
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Start request repeated too quickly.
Dec 26 00:40:56 DietPi systemd[1]: Failed to start Radarr Daemon (DietPi).
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Unit entered failed state.
Dec 26 00:40:56 DietPi systemd[1]: radarr.service: Failed with result 'exit-code'.
root@DietPi:~# cat /var/log/radarr/radarr.txt
root@DietPi:~# systemctl cat radarr.service
# /etc/systemd/system/radarr.service
[Unit]
Description=Radarr Daemon (DietPi)
Wants=network-online.target
After=network-online.target dietpi-boot.service

[Service]
SyslogIdentifier=Radarr
[... 26 lines cut ...]
#Wants=network-online.target
#After=network-online.target dietpi-boot.service

[Service]
#SyslogIdentifier=Radarr
User=radarr
#UMask=002
Environment="TMPDIR=/tmp/radarr"
#ExecStart=/usr/bin/mono -O=-aot /opt/Radarr/Radarr.exe -nobrowser -data=/mnt/dietpi_userdata/radarr
#TimeoutStopSec=20
#KillMode=process
#Restart=on-failure

# Hardening
#ProtectSystem=strict
#ProtectHome=true
#PrivateDevices=true
#ProtectKernelTunables=true
#ProtectControlGroups=true
#ReadWritePaths=/opt/Radarr /mnt /media /var/log/radarr /tmp

[Install]
#WantedBy=multi-user.target

```

Joulinar, 2020-12-25:
hmm maybe an issue with one of the drop-in config files `/etc/systemd/system/radarr.service.d/dietpi-services_edit.conf`

There a temp directory is specified `Environment="TMPDIR=/tmp/radarr"`. And I guess this is not existing after reboot. Pls can you check following `ls -la /tmp`

shlatchz, 2020-12-25:
```
root@DietPi:~# ls -la /tmp
total 4
drwxrwxrwt 10 root root    200 Dec 26 01:00 .
drwxrwxr-x 22 root root   4096 Dec 25 01:31 ..
drwxrwxrwt  2 root root     40 Dec 26 00:40 .font-unix
drwxrwxrwt  2 root root     40 Dec 26 00:40 .ICE-unix
drwxr-xr-x  2 plex dietpi   40 Dec 26 00:40 pms-3ae806bf-52e7-4433-b343-a9a981d6a266
drwx------  3 root root     60 Dec 26 00:40 systemd-private-6d20fd19f7c4456dad82993dd5701078-haveged.service-CzMsLf
drwx------  3 root root     60 Dec 26 00:40 systemd-private-6d20fd19f7c4456dad82993dd5701078-jackett.service-2ted0S
drwxrwxrwt  2 root root     40 Dec 26 00:40 .Test-unix
drwxrwxrwt  2 root root     40 Dec 26 00:40 .X11-unix
drwxrwxrwt  2 root root     40 Dec 26 00:40 .XIM-unix
```

Joulinar, 2020-12-25:
yep it's missing. Was there a reason for this? Because this is not a default setup. The drop-in is usually manually created using `dietpi-services`.

shlatchz, 2020-12-26:
@Joulinar how would I do that?

Joulinar, 2020-12-26:
open `dietpi-services` > `radarr` > `Edit` > change line `Environment="TMPDIR=/tmp/radarr"` into `Environment="TMPDIR=/tmp"`

Once leaving the editor and `dietpi-services`, you will be ask to restart the service. Hopefully Radarr will be working now.

shlatchz, 2020-12-26:
It works!!
Thanks @Joulinar! You're a life saver :)

Joulinar, 2020-12-26:
Did you test a reboot as well?

shlatchz, 2020-12-26:
still works!

Joulinar, 2020-12-26:
Perfect, I'm going to close the issue for now?

closed 2020-12-26 by Joulinar

MichaIng, 2020-12-26:
@shlatchz 
Is there any reason you or someone edited the service file at all? The `User=` override just matches the the one of the original service, hence has no effect, and the `TMPDIR` override breaks things naturally as long as nothing is creating this file. I'm not sure why we didn't apply a private `/tmp` mount for it, probably either the downloader or Jackett can require access to Radarr's temporary files, not sure.

So my suggestion is to simply remove that file to prevent any future issues with it. It should only be changed that way if you exactly know what for 😉:
```
rm /etc/systemd/system/radarr.service.d/dietpi-services_edit.conf
systemctl daemon-reload
systemctl restart radarr
```

shlatchz, 2020-12-26:
@MichaIng the only changes I made are the ones in the workaround here - https://github.com/MichaIng/DietPi/issues/3179

MichaIng, 2020-12-26:
Ah, hmm that workaround doesn't involve TMPDIR and one idea at least was to run Radarr as root user if it requires access to a root-mounted CIFS or FAT family mount. The other to mount them as user radarr.

`dietpi-services_edit.conf` is definitely not required, so you can safely remove it 🙂.

Joulinar, 2020-12-26:
@MichaIng 
ast least having a temporary files pointing to `/tmp` isn't a bad idea. At least it could reduce r/w operation?

MichaIng, 2020-12-26:
`TMPDIR` defaults to `/tmp` anyway, which is totally fine and right. But having it pointing to a sub dir is an issue as long as it is not created by the systemd unit (like `ExecStartPre=mkdir -p /tmp/radarr`) or by systemd-tmpfiles.

The only reason I can think of to touch defaults regarding temporary files directories would be to further secure things by adding `PrivateTmp=true` to the systemd unit so that the service has its own private `/tmp` mount that lives for and as long as the service runs, only. I wonder why we didn't do this with Sonarr/Radarr/Lidarr etc. Either I simply forgot to add it or there was a case that either the downloader or Jackett required access to something created in `/tmp` 🤔.

TDuffinNTU, 2023-04-12:
I seem to be getting the same issue here, but I don't have an "Environment" setting in my radarr sysctl config. Only caused a problem after updating to the latest radarr version.

MichaIng, 2023-04-12:
Could you please open a new bug report and fill the template? Whatever it is, it is unrelated to this 2 years old closed issue. Also add the service log, please:
```sh
journalctl -u radarr
```

MichaIng, 2023-04-12:
Ah that is good. Indeed a `reinstall` alone does not remove `/opt/radarr` but refers to better using the internal updater. I'm thinking whether we should just do the `/opt/radarr` on a reinstall automatically. AFAIK, the internal updater does nothing else and all database migration steps are done on first service start after the update (file replacement).