MichaIng/DietPi #4005: Heimdall Dashboard support/request

N7-BADA, 2020-12-25:
I installed this on a Raspberry Pi Zero W and has been working really good, and would love to see this in dietpi supported list.

#### Formal software information
- Software title | Heimdall Dashboard
- Short description | A dashboard for most of your installer services/applications
- Official URL (if available) | https://heimdall.site/
- GitHub URL (if available) | https://github.com/linuxserver/Heimdall
- (Official) install documentation | https://github.com/linuxserver/Heimdall#installing

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
1. I followed this guide for the Pi Zero W running DietPi | https://varhowto.com/install-heimdall-dashboard-ubuntu-20-04/
2. I needed to add --host 0.0.0.0 before --port 7889 in heimdall.service in order to access it remotely. 
3. phpMyAdmin from DietPi was also needed for this to work.
4. and needed this fix for an error | https://jonathanmh.com/solved-laravel-supported-ciphers-aes-128-cbc-aes-256-cbc-correct-key-lengths/

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
Sorry, I am totally unexperienced in bash coding.

N7-BADA, 2020-12-25:
[image]
My setup running on a Raspberry PI Zero W with latest DietPi

Ap0them, 2021-10-05:
Plus 1 for this, it'd make the install a little cleaner & Heimdall is a pretty well used manager

Bwangster12, 2021-12-15:
Is there anything already available through DietPi like this?  This would be slick.

Ap0them, 2021-12-16:
> Is there anything already available through DietPi like this?  This would be slick.

Idk I kind of stopped using DietPi past some messing around, I personally felt as someone with a better understanding of Linux that Rasbian worked better. So I'm not sure but good luck finding something

MichaIng, 2021-12-17:
> Is there anything already available through DietPi like this? This would be slick.

Not that I'm aware of. I mean it is no interface integration(?), like HTPC Manager, aside of some status via API requests shown on the icon. But otherwise you can achieve the same via browser bookmarks/quick launch buttons.

But it should install just fine following the instructions: https://github.com/linuxserver/Heimdall#installing
Installing PHP and SQLite via `dietpi-software` should setup all dependencies:
```sh
dietpi-software install 87 89
```
MariaDB would be total overkill and phpMyAdmin is generally not required in any case, regarding what has been noted in OP.