MichaIng/DietPi #4007: On desktop-env switch, autologin doesn't work

C0D3-M4513R, 2020-12-26:
### Creating a bug report/issue
When you switch from a desktop environment, such as lxde, mate, xfce or lxqt, to another by uninstalling the first and then installing the other, `startx` doesn't work as expected anymore
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
`buster` or `10.7`
- Kernel version | `uname -a`
```
Linux DietPi 5.4.79-v8+ #1373 SMP PREEMPT Mon Nov 23 13:32:41 GMT 2020 aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
RPi 4 Model B (aarch64)
```
- Power supply used | (EG: 5V 1A RAVpower)
Official 5V 3A USB-C PSU
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
LXDE/XFCE
- Was the software title installed freshly or updated/migrated?
- Fresh install
- Can this issue be replicated on a fresh installation of DietPi?
- Not tested, but should. Will test further if nessesary

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Have a running dietpi install
2. Install LXDE fully
3. Reboot if asked
4. Ideally make sure, that the LXDE works, when running `startx`
3. Uninstall LXDE
4. Install XFCE
5. Reboot if asked
6. Ideally make sure, that the XFCE works, when running `startx`
7. Observe `startx `failing

#### Expected behaviour

- `startx` should just start XFCE with no issues

#### Actual behaviour

- `startx` is timing out with something

#### Extra details

Notice: 
I had a physical display , but also "TigerVNC Server" intalled, and both were starting a desptop environment, that would complain, if it was active more than one time? I eventually uninstalled, and found, that the initial problem still persisted. I do not think, that the VNC Server interfered here, because I uninstalled, then rebooted, and found the same auto behavior (I wasn't doing anything to the Pi remotely, such as accessing ssh or vnc, until I saw, that it tried to start the desktop-environment) as the boot before.

The Fix:
I got everything working, after diagnosing the issue using these steps:
1. Find, that the package of the `startx` command is `xinit`
2. `sudo apt purge xinit xauth`
3. `sudo apt install xinit xauth`

I do NOT know, if I actually had to uninstall `xauth` here, but it was unsinstalled with a autopurge, so I let apt do it's thing, and just reinstalled it later.

C0D3-M4513R, 2020-12-26:
On further investigation it seems, as if it is Not intended to autologin to a desktop enviorment as a non-root user? The command `startxfce4` is giving me one permission error after the other. If I solve something, that is easy to do (such as not having perms to access /dev/ttyX) others just pop up.
If I do `startx`, the error is `xf86OpenConsole: Switching VT failed`, or `xf86OpenConsole: Cannot open virtual console 2 (Permission denied)`, depending, if I do not use ssh or if I do.

MichaIng, 2020-12-26:
Many thanks for your report.

I would have expected that the `startx` command default client gets adjusted when uninstalling one desktop and installing a new one. We could check that on a desktop install and in case adjust it manually, but it would require some more checks, e.g. other installed X applications which may reasonably be default display managers, like other desktops of course but also `LightDM` login mask. I had the impression that this was handled satisfying in most situations, but I'll try to replicate the case and see if/what we can do to enhance it.

> On further investigation it seems, as if it is Not intended to autologin to a desktop enviorment as a non-root user?

It was planned but not yet implemented to automatically assign the chosen autologin user the required permissions for the desktop environment. E.g. `usermod -aG video <username>` to grant it access to the X server. `usermod -aG tty <username>` should only be required if you want to start a desktop session from a different console, e.g. run `startx` command on SSH to start a desktop on `/dev/tty1`, i.e. attached monitor. But it makes sense to have this restricted by default as long as the same user is not clearly logged in at `/dev/tty1` already.

C0D3-M4513R, 2020-12-26:
`usermod -aG tty <username>` helped, when sending a command from ssh to the tty on my pi's monitor.  `usermod -aG video <username>` didn't do anything. I checked again, after removing video from <username>. Btw, as <username> I'm just using dietpi.

C0D3-M4513R, 2020-12-26:
>I would have expected that the startx command default client gets adjusted when uninstalling one desktop and installing a new one. 
I dunno if that's the case, but I do belive so. Please try it. I am unsure, if I actually started the desktop-environment as non-root or as root.

MichaIng, 2020-12-26:
With user `dietpi` in `video` group there were still permissions issues when starting Xfce? There is as well the `render` group for DRM access, but that should not be required for an X server session (desktop).

C0D3-M4513R, 2020-12-26:
Just referencing a couple things here: Apperantly this bug is well known...
[https://gist.github.com/alepez/6273dc5220c1c5ec5f3f126e739d58bf](https://gist.github.com/alepez/6273dc5220c1c5ec5f3f126e739d58bf)
[https://github.com/mviereck/x11docker/issues/5](https://github.com/mviereck/x11docker/issues/5)
[https://bbs.archlinux.org/viewtopic.php?id=192329](https://bbs.archlinux.org/viewtopic.php?id=192329)
[https://bugs.launchpad.net/ubuntu/+source/xinit/+bug/1562219](https://bugs.launchpad.net/ubuntu/+source/xinit/+bug/1562219)
[https://superuser.com/questions/845069/permissions-messed-up-on-x-server](https://superuser.com/questions/845069/permissions-messed-up-on-x-server)
[https://unix.stackexchange.com/questions/346383/run-second-x-server-from-within-x-as-a-systemd-service](https://unix.stackexchange.com/questions/346383/run-second-x-server-from-within-x-as-a-systemd-service)
[https://ubuntu-bugs.narkive.com/ZsgjjZIg/bug-1562219-new-xinit-will-not-work-as-non-root#post33]()

C0D3-M4513R, 2020-12-26:
a ` chmod 660 /dev/tty*` helped things further... The error went from `xf86OpenConsole: Cannot open virtual console 3 (Permission denied)` to `xf86OpenConsole: Switching VT failed`
(ignore the different tty numbers everywhere)

C0D3-M4513R, 2020-12-26:
I think the important thing is to get tty1 be spawned as the user dietpi, so the tty will be owned by dietpi?

C0D3-M4513R, 2020-12-26:
After executing 
```
systemd-run \
--property PAMName=login \
--property StandardInput=tty \
--property TTYPath=/dev/tty1 \
 sh -c 'chvt 1 && startx'
 ``` 
 (from [https://unix.stackexchange.com/questions/346383/run-second-x-server-from-within-x-as-a-systemd-service](https://unix.stackexchange.com/questions/346383/run-second-x-server-from-within-x-as-a-systemd-service)) and killing the login-shell, the server starts as expected. Maybe add that as a service?
This should only be temporary, but I do not know, how to reverse it. For now I just hope, that keeping the pi plugged out of everything will reset that change. 
While that change is active, the desktop autostart works as expected. I assume, that this would also be unnessesary: [https://github.com/MichaIng/DietPi/blob/b354689141bf3b238b87d9ff83f33df5cf52173f/dietpi/dietpi-login#L46](https://github.com/MichaIng/DietPi/blob/b354689141bf3b238b87d9ff83f33df5cf52173f/dietpi/dietpi-login#L46)

MichaIng, 2020-12-26:
First of all, the TTY permissions are exactly as they should be, do not change those, that should never be required by any process that is intended to be executed by non-root users. The `tty` group is only intended to **write** to other TTYs, but not to directly read from them.

/dev/tty* devices are owned by the respective login user, once a login has been done, hence the `dietpi` owner once you logged in on that TTY via local console login or autologin. `dietpi-autologin.conf` leads to autologin with the user you chose via `dietpi-autostart`, the result is the same as if you would login manually with that user. `/dev/tty1` owner is then changed, but not due to the config file itself, but due to the fact that user `dietpi` logged in on that TTY. It is handed back to root user, once you log out.

`noclear.conf` has no influence in this regards, it only leads to boot messages being not removed when the login prompt appears. But after login, everything is cleared anyway. That is a non-default file placed on Raspbian images (pi-gen) only: https://github.com/RPi-Distro/pi-gen/blob/master/stage1/01-sys-tweaks/files/noclear.conf

> This should only be temporary, but I do not know, how to reverse it.

Logging out the desktop session stops that temporary systemd service. You only started `startx` as temporary service with `PAMName=login` (aside of everything else that is standard), so no permanent changes. You mean starting the desktop now still works after reboot? It cannot really be due to executing this command, actually. Still, good to know about `PAMName=login`, not exactly sure what role it plays here, will have to play around to find the essence and how to implement best for non-root desktop sessions.

Joulinar, 2020-12-26:
linking an older issue with possible workaround that was still on the list to test https://github.com/MichaIng/DietPi/issues/3586

C0D3-M4513R, 2020-12-27:
> First of all, the TTY permissions are exactly as they should be, do not change those, that should never be required by any process that is intended to be executed by non-root users.

That was for temporary testing. Those changes are not required, just like you said.

> /dev/tty1 owner is then changed, but not due to the config file itself, but due to the fact that user dietpi logged in on that TTY.

That somehow didn't happen.

> You mean starting the desktop now still works after reboot?

Yes. That is correct. After executing the command in my previous post, it works, even after a reboot.

C0D3-M4513R, 2020-12-27:
>Still, good to know about PAMName=login, not exactly sure what role it plays here, will have to play around to find the essence and how to implement best for non-root desktop sessions.

I can help with that: from [https://unix.stackexchange.com/a/473041/448036](https://unix.stackexchange.com/a/473041/448036)

The magic is in defining PAMName=, to open a PAM session, and associating that session with the specific TTY. This gets [pam_systemd](https://www.freedesktop.org/software/systemd/man/pam_systemd.html) to do what you want. I spoofed login - though technically you're supposed to define a new PAM "service name" in case it needs some special treatment.

MichaIng, 2020-12-27:
Yeah, a desktop session however automatically invoke PAM, AFAIK.

C0D3-M4513R, 2020-12-28:
>Yeah, a desktop session however automatically invoke PAM, AFAIK.

Maybe so, but apparently running `startx` from dietpi has the consequense, that on start it cannot configure everything correctly, because it is not root. Therefore the desktop doesn't boot?
If you run 
```
systemd-run \
--property PAMName=login \
--property StandardInput=tty \
--property User=dietpi \
--property TTYPath=/dev/tty1 \
 sh -c 'chvt 1 && startx'
 ```
 as root, it sets all permissions correctly for startx (?), but the startx, from dietpi-login in run (observed).

ravenclaw900, 2021-01-16:
I tried running that, and it didn't work for me.  However, `chmod ug+s /usr/lib/xorg/Xorg` did.

MichaIng, 2021-01-16:
Note that `chmod ug+s /usr/lib/xorg/Xorg` means that every user who calls Xorg will gain full root permissions through it, breaking basically the whole security concept :wink:.

If there is no reliable way to allow non-root users start a fully functional desktop session via system group memberships, the only viable solution I see is to use LightDM with autologin, which is pretty easy: https://www.maketecheasier.com/enable-autologin-lightdm/
The X server session then runs as `lightdm` user (we could review the way that user is able to do it) and the client desktop session as login user of course.

C0D3-M4513R, 2021-01-17:
> I tried running that, and it didn't work for me.

How about we remove (non-root) Desktop Autologin for now, add it back later, if there is a better way found in this issue and flag this issue as an improvement/enhancement?

MichaIng, 2021-01-17:
Probably a warning would do. The method to choose the user for auto-logins is implemented in a generic way currently and in other cases as well permissions might be required. So to keep some freedom for admins who manage to deal with the permissions, I think it would be sufficient to warn about it and recommend to switch back to root or LightDM login in case issues are faced. LightDM autologin actually looks so easy and reasonable that I'll find time to implement it with next release.

MichaIng, 2021-02-15:
Marking this as closed in favour of #3586.

closed 2021-02-15 by MichaIng