MichaIng/DietPi #4008: Update Problem

Volker13, 2020-12-26:
#### Details:
- Date           | Sat Dec 26 14:32:27 GMT 2020
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 4.19.73-v7l+ #1266 SMP Tue Sep 17 20:54:23 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -qq --allow-change-held-packages install rpi-eeprom libraspberrypi-bin raspberrypi-bootloader
- Exit code      | 100
- Software title | DietPi-Set_hardware
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Unable to locate package rpi-eeprom
```

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2020-12-26:
Hi, 

Many thanks for your message. Pls can you try apt update && apt upgrade

Volker13, 2020-12-26:
root@DietPi:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Hit:3 https://packages.sury.org/php stretch InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPi:~#

Thanks. You are very fast, even on christmas

Joulinar, 2020-12-26:
Hmm I can't really verify it as I'm not at home but probably package `rpi-eeprom` is not available on Raspberry OS Stretch. As you are using RPi4, DietPi is trying to install that package. 🤔

MichaIng, 2020-12-26:
How is it even possible to run a Stretch image on RPi 4? This has never been compatible, AFAIK RPi 4 support has never been added to this 4.19 kernel shipped with Stretch, has it? That was the reason why Raspbian Buster was released together with Raspberry Pi 4 even before Debian Buster was.

@Volker13 
Did you move the SD card from an older RPi model or how did you set this system?

Volker13, 2020-12-26:
I bought the RPi4 together with an allo digione signature hat an the SD card with the preinstalled dietpi.

Joulinar, 2020-12-26:
Probably better to use the new Allo Buster image than continuing running old Stretch

MichaIng, 2020-12-26:
Ah, so Allo ships the SD card with Stretch pre-installed? That would be an issue indeed, I'm confused that it boots at all. In case, I see there is a selector on the product page for RPi 3 and RPi 4 choice. Do you remember if this existed when you ordered it? Since the hardware most likely it the same (HATs dimensions should be cross-compatible), it probably is to choose the image.

As @Joulinar suggested, I'd also recommend to re-flash our current Buster image, the one with Allo web UI pre-installed if required: https://dietpi.com/downloads/images/DietPi_RPi-ARMv7-Buster_AlloGUI.7z
Else the regular one from our download page: https://dietpi.com/#download
Another reason for this, aside of much never software packages, is that we'll drop update support for Debian/Raspbian Stretch in summer, when Debian Bullseye is officially released.

Volker13, 2020-12-26:
Where can I download this image from?

Volker13, 2020-12-26:
I bought this all together 2 days ago. I select the rpi 4 and not the rpi 3. it was a plug&play set
https://www.allo.com/sparky/digione-signature-player.html

Volker13, 2020-12-26:
I will  re-flash the current Buster image tomorrow.
Thanks so much and merry Christmas from Berlin. Stay healthy

Joulinar, 2020-12-26:
Merry Christmas to you as well from Berlin to Berlin 😉

MichaIng, 2020-12-26:
Many thanks and merry Xmas back.

Thanks for the info, I need to ask Allo then to use our new image or update the ones they use, as that Stretch one with RPi 4 and in general will be no joy for anyone 😄.

Volker13, 2020-12-26:
Update ok, problem solved

closed 2020-12-26 by Volker13

Joulinar, 2020-12-26:
indeed would be good if Allo can switch to Buster as we had some messages for Allo images running Stretch 😉