MichaIng/DietPi #4009: Cant install Roon extension manager

Volker13, 2020-12-26:
### Creating a bug report/issue

[ INFO ] DietPi-Software | ./setup.sh, please wait...
roon-extension-manager setup script - version 0.5.2

Checking prerequisites...
    OK
npm ERR! invalid options argument

Joulinar, 2020-12-26:
Hi,

Many thanks for your message. Pls can you share some more details like SBC used as well as DietPi version. Best would be to fill the questionnaire during issue creation.

MichaIng, 2020-12-26:
Just linking the other issue #4008 which shows a strange setup with RPi 4 + Raspbian Stretch, which should not work at all. However, I'll do a Roon Extension Manager test install on Stretch to see if there is a general issue.

Volker13, 2020-12-26:
Solved

closed 2020-12-26 by Volker13

Joulinar, 2020-12-26:
probably related to the old Stretch image as well