MichaIng/DietPi #4012: nand-sata-install doesn't check for prerequisites before running

f00b4r0, 2020-12-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
buster

- Kernel version | `uname -a`
Linux NanoPiNEOAir 5.9.14-sunxi #20.11.3 SMP Fri Dec 11 20:31:12 CET 2020 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
NanoPi NEO Air (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 1A

- SDcard used | (EG: SanDisk ultra)
SanDisk Ultra 16GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
nand-sata-install

- Was the software title installed freshly or updated/migrated?
fresh install (from just after initial boot)

- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Steps to reproduce
1. flash dietpi image to SD card
2. first boot, finish setup in minimal (barebone) state
3. try to run `nand-sata-install`
4. it produces no output

#### Expected behaviour
The tool should at the very least confirm that all the necessary utilities are present on the system (in particular `dialog`, `lsof`, `rsync`, and/or filesystem-related tools when e.g. f2fs or brtfs format is selected, and provide an error message when any is missing

#### Actual behaviour
No output, which the average user will not be able to figure out.

#### Extra details
I had to apt-get install the missing packages (listed above) after reading through the script to figure out which were needed to get this to work.
As an aside, the tool doesn't cleanup after itself when it fails midway through (when e.g. specific filesystem utils are missing), leaving mounted filesystems in temporary locations.

HTH

Joulinar, 2020-12-27:
Hi,

Many thanks for your message. I don't think this script is provided by DietPi. Probably it's provided by the Armbian base image?

f00b4r0, 2020-12-27:
> Hi,
> 
> Many thanks for your message. I don't think this script is provided by DietPi. Probably it's provided by the Armbian base image?

I don't know. I installed dietpi, not armbian :)
I'm not familiar with the relationship between the two.

If that comes from armbian, maybe armbian ensures the necessary prerequisites are available in the base install, in which case the bug lies with dietpi.

MichaIng, 2020-12-27:
I suggest to flash the image to the target drive directly and skip that script, if possible. We only ship it as part of the Armbian root overlay package, actually only since it contains the required kernel update script to convert kernel image to U-Boot format. I aimed to skip that package completely by shipping an own kernel update script, but probably switching from Armbian-based builds to pure Debian-based builds with Bullseye anyway. So we'll not gonna start investigating and supporting that script but the opposite.

If the DEB package that ships this script does not pull all dependencies required to run it, it is something that Armbian might want to fix, e.g. not sure if their minimal images come with those tools.
__________
What I can think of is that we extend the (currently RPi + Odroid only) feature of `dietpi-drive_manager` for moving the rootfs to external drive to allow moving the boot partition as well. That requires flashing the bootloader on non-RPi but scripts for that are present. I'm anyway playing with building images from scratch currently, which requires basically the same steps.

f00b4r0, 2020-12-27:
> I suggest to flash the image to the target drive directly and skip that script, if possible. 

That sounds far more error-prone and less user-friendly than what this script does 😛 
Once the dependencies are met, the script works flawlessly. It also makes it very easy to switch the root filesystem to something a little more sensible than ext4 😉 
It also correctly handles flashing the bootloader to mmcblk1boot0 as well. And I suspect it wouldn't be too difficult to make it cope with partitioned targets, something that would be impossible with a raw flash of the image, since it always autoexpands. (and obviously you can't extract a builtin eMMC to mount it on a separate machine).

> We only ship it as part of the Armbian root overlay package, actually only since it contains the required kernel update script to convert kernel image to U-Boot format. I aimed to skip that package completely by shipping an own kernel update script, but probably switching from Armbian-based builds to pure Debian-based builds with Bullseye anyway. So we'll not gonna start investigating and supporting that Armbian script, but kind of the opposite 😉.
> 
> If the DEB package that ships this script does not pull all dependencies required to run it, it is something that Armbian might want to fix, e.g. not sure if their minimal images come with those tools.

```
$ dpkg -S /usr/sbin/nand-sata-install 
linux-buster-root-current-nanopineoair: /usr/sbin/nand-sata-install
```

I guess I could forward this to armbian, but I feel uneasy reporting a "bug" observed on a derivative distro. That's usually frowned upon.

Cheers

Joulinar, 2020-12-27:
> I guess I could forward this to armbian, but I feel uneasy reporting a "bug" observed on a derivative distro. That's usually frowned upon.

DietPi is not an own distro. It's a set of scripts on top of a base image. In your case, base image is Armibian. Means, you are running an on extrem slim/diet version of Armbian with a couple of magic scrips.

MichaIng, 2020-12-27:
As said, I see the use case for this feature and we likely will implement it into our own scripts, but since we won't directly use or promote Armbian's scripts, we won't manually maintain a dependency list for those. Either their dependencies are declared correctly within the package, or they need to be installed manually, as it is with every other 3rd party package 😉.

> I guess I could forward this to armbian, but I feel uneasy reporting a "bug" observed on a derivative distro. That's usually frowned upon.

Yes they likely won't care when it doesn't affect their own images. The chance is higher when contributing the change directly. Here is the related code: https://github.com/armbian/build/blob/master/lib/makeboarddeb.sh#L56
But the dependencies for this script might be missing purposefully when it is not seen as a base functionality of the package (basic Armbian boot scripts and services + kernel update support).

closed 2021-02-15 by MichaIng