MichaIng/DietPi #4014: LXDE Desktop modifications for all pre-installed users

C0D3-M4513R, 2020-12-28:
### Creating a feature request
I would like for the default LXDE Desktop modifications, to also apply to all other pre-installed users, other than root.

#### Is your feature request related to a problem? Please describe:

- Something only works as root. root should only be used as minimally, as possibile. It is much too easy to destroy your whole system all at once. Permissions are there for a reason. 
- #4007 

#### Describe the solution you'd like:

- I would like for all desktops to have all modifications applied to all users equally. Equality for non-roots!

#### Describe alternatives you've considered:

- Just applying the modifications to all user homes, or applying all modifications to /usr/share

MichaIng, 2020-12-28:
Many thanks for your request.

Indeed it generally makes sense. The `dietpi` user is the only pre-installed one, but I aim to remove it and instead offer custom login user creation on first login of a new image. It is not good to have a second login pre-installed user with a default password assigned.

What I don't like as well is to loop through all available login users (UIDs > 1000 + root) and copy the files in place for all of them. That would be quite some overhead and include users which can be logged in with but are aimed as service users only, actually, like "qbittorrent".

/usr/share is used by the APT packages, and I don't want to override their default files. So, probably /usr/local/share would work as a default for all users, overriding the package defaults but allowing user-level overrides as well.

Another approach would be to only create/copy configs/files (when not existent already) from `/root` to the one user that is selected for desktop auto-login via `dietpi-autostart`.

_______________

Btw, for non-root X sessions in general, another approach for non-root desktop session would be to move away from the idea of allowing non-root X sessions but instead using LightDM in every case, which AFAIK does support auto-login as well. The `lightdm.service` is started as root, including the surrounding X session, but the desktop itself is started with limited permissions as of the user that you login with via LightDM. That would match the default desktop of Debian/Ubuntu etc desktop images, AFAIK and causes least permissions friction like #4007. But it limits the possibilities, e.g. having a really fast boot into desktop while other services and even network are still setting up in background, like we do with Amiberry fast boot.

C0D3-M4513R, 2021-02-05:
Ok, so this should be already fixed by #4015 . Closing this.

closed 2021-02-05 by C0D3-M4513R

MichaIng, 2021-02-05:
It is btw automatically closed once the commits have been merged into the default/master branch, i.e. once the next version is released. I generally like to keep issues open for beta phase, but to be true it did rarely serve the intended purpose that beta testers keep posting to those issues when facing related problems or questions.

PR #4015 (2021-02-03): Apply desktop Customisations to all users
**Status**: Testing
- [x] LXDE
- [x] MATE
- [x] LXQt
- [x] Xfce
- [x] GNUstep

Also please link the issues below.
**Reference**: fixes #4014 fixes #3994