MichaIng/DietPi #4016: Lost HDM output  after last update 6.34

drteevee, 2020-12-28:
Good morning, 
after  last update 6.34 performed without any  problems  is  HDMI output suddenly lost . 
tvsetvice -s    show  only composite outupt . 
I  tried   alll recommended  settings in config. txt, dietpi.txt  without success.
my Dietpi  runs  on SSD via USB boot  on pi 3B+.
If I replace  SSD with SD card  with same  veresion of  update it works  normally - HDMI  is  working  normally
I  tried  fsck na SSD, repair filesystem  actually  without errors.    All another installed software   works normally except HDMI . 
What can I do  more ? 
Tomas

MichaIng, 2020-12-28:
Many thanks for your report.

> If I replace SSD with SD card with same veresion of update it works normally - HDMI is working normally

So it's not a hardware or kernel issue. Let's see assure config is correct and see what the kernel does during boot:
```
grep '^[[:blank:]]*[^#]' /boot/config.txt
dmesg
```

MichaIng, 2021-02-14:
I mark this as closed. Feel free to reopen if issue persists.

closed 2021-02-14 by MichaIng