MichaIng/DietPi #4017: FriendlyARM NanoPi R1

robyscar, 2020-12-28:
### Creating an image request

#### Formal device information
- Device name | 
- Official product URL | 
- GitHub resource URL (if available) | 
- Image download URL | 
- Wiki URL | 

#### Is the SBC officially supported by the Debian installer?
- only UNSTABLE on sunny H5 | 

#### If not, is a reliable 3rd party Debian image available for this SBC?
- ARMBIAN | 

#### If not, are there install instructions for Debian available?

- ...

#### Vote for this image on FeatHub: https://feathub.com/MichaIng/DietPi/

#### SBC Spec

midzelis, 2021-03-18:
+1, I have this board, and will be willing to contribute testing and whatever I as far as development goes. I noticed this dietpi is basically a TON of bash? scripts. shell scripts are not my strong suite, but I can get by. It seems like Nano Pi R1 uses the same chip as ZeroPi, NanoPi NEO/Neo Air/M1/T1/M1 Plus -- so it seems like adapting one of those could work. In fact, i'm going to download the NanoPi M1/T1 image and see if it loads. 

What goes into creating support for a new SBC? 

Thanks!

MichaIng, 2021-03-19:
Easiest world be to implement it with Armbian base image / kernel repository. I'll do that next week together with two other outstanding NanoPi SBCs. I'm tinkering with the Debian kernel as well, but that will take a bit longer.

robyscar, 2021-03-25:
thanks @MichaIng really appreciated!
we (at least me and @midzelis ) we are sure that 1st version will be good, and the Debian one will be great!
thanks
Roby

ScottSturdivant, 2021-04-01:
The ZeroPi image boots and runs on the Nano Pi R1. Is there any thing I should be on the lookout for that might not work? Or should I be hopeful that by virtue of it booting and seeing to operate OK that it'll fine?

Thank you!

MichaIng, 2021-04-02:
Comparing the hardware specs:
- https://wiki.friendlyarm.com/wiki/index.php/ZeroPi#Hardware_Spec
- https://wiki.friendlyarm.com/wiki/index.php/NanoPi_R1#Hardware_Spec

The ZeroPi has only one Ethernet, no WiFi/BT module and no eMMC, so those three might not work. On the other hand, the kernel + device tree package should be the same, so likely the correct device tree is present and picked already by the bootloader. Since Armbian uses minimally modified upstream kernels, sharing then across a large span of SBCs, the images are often cross-compatible.

MichaIng, 2021-06-01:
Image ready: https://dietpi.com/downloads/images/DietPi_NanoPiR1-ARMv7-Buster.7z

closed 2021-06-02 by MichaIng