MichaIng/DietPi #4018: DietPi-Software | Stop dependencies from being installed when they already are

ravenclaw900, 2020-12-28:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- When installing something with dependencies, it always tries to install those dependences.  Even APT packages can take a while on Raspberry Pi, and when things have to be built from source it takes even longer.

#### Describe the solution you'd like:

- Check if software is already installed before trying to install it as a dependency

#### Describe alternatives you've considered:

- Not installing dependencies that build from source.

MichaIng, 2020-12-28:
I think reinstalling everything is the lesser of the two evils. The dependencies might need to be setup a way to work correctly with the dependants. Or users apply custom settings or remove/manipulate the instance manually so that the dependant cannot work with it, or simply something has gone wrong so that the install flag has not been adjusted correctly. A reinstall assures that a minimum required install and config based on which issues can be debugged, without that I'm afraid we'd get a lot more bug reports/support requests which are as well harder to debug.

As well sometimes we change the way certain software titles are installed or configured but do not always force a reinstall on DietPi update, if it is not required to keep it running or custom steps are required to allow a clean reinstall (which are then done via incremental patches before the forced reinstall). If newly installed dependants rely on the "new" setup or their dependencies, we'd need to force reinstalls more often on DietPi updates, while I'd like to keep those at a minimum, as well to keep issues more separated (=easier to debug) that are either related to the DietPi update or to the software reinstall itself.

For APT packages I could imaging it, since their install state is more reliable (let's say it needs to be more reliable to not cause worse issues elsewhere). An install implies an upgrade as well, but `dietpi-software` does an upgrade already when installs start. I even aim to remove this general "apt upgrade" as it is not required. `dietpi-update` forces all packages upgrade and the version state they have at that time is the latest we might require/expect within our scripts. So that general package upgrade is more like a "services are stopped anyway and stuff installed, so why not bringing all packages on a current state as well". But again, we often get bug reports or support requests related to software installs, while they were caused by an APT package upgrade that has nothing to do with the software install. That is as well an argument to leave APT upgrades to either being made manually by user (allowing interactive config file or debconf choices and allowing to track down issues to that exact package upgrade call) or during `dietpi-update`, where it is simply required so that we know which minimum package versions we can expect.
_______________
So what I am thinking of:
- Merge `G_AGI` with `G_AG_CHECK_INSTALL_PREREQ` to only install a package when it is not installed yet.
    - Add a `-f` (force) option to override it and install all named packages regardless of install state and probably an `-r` (reinstall) flag, if we already start with such?
    - `G_AG_CHECK_INSTALL_PREREQ` currently does `G_AGUP` as well when a package install is required, since it is usually used within our config scripts where the last list update could have been a while. I was already thinking of another flag file (or merging it with the file that holds the available packages now) that holds the information that APT lists are up-to-date, respectively has run during the last our while within the current script execution. Easiest would be likely this:
        - When our scripts call G_AGUP, they create a new flag file. If it exists, G_AGUP won't do something (probably prints an info that it does not do something because that flag file exists). When a script with `HIERARCHY == 0` (the parent script in case cascaded scripts were called) exists, it removes that flag file. This assures that whenever one of our scripts is called and within that "session" G_AGUP is called once or multiple times, is executed exactly once, regardless if it was the parent script or a child script. If `G_AGUP` needs to be forced, e.g. because a new repo has been installed, the flag file simply needs to be removed, or we add a "`--force`" flag... dammit "`-f`" is already used to create the file with the amount of available APT updates 😄.
- Additionally we can review often used dependencies install/config steps and see it some could or should be skipped on a reinstall, to speed up things. But especially time consuming source builds as well assure that the current (hence hence probably expected/required) version is installed, hence not sure if that could or should be skipped. Since the dependency system in `dietpi-software` currently is not really standardised, is could as well be adjusted for single cases, e.g. `(( ${aSOFTWARE_INSTALL_STATE[132] == 2 )) || aSOFTWARE_INSTALL_STATE[132]=1`.

MichaIng, 2021-02-08:
While it's not yet what was thought above: I just opened a PR to remove the mandatory package upgrade on every `dietpi-software` install, which can speed up things quite a bit. This however actually makes the forced G_AGI for individual packages more reasonable, since those are now assured to be up-to-date now.

I'm not yet sure how to handle best, e.g. create a list of upgradeable packages once to check for individual installs if those apt calls would actually install or upgrade anything, or ignore it so that dietpi-software installs may finish with outdated packages, even of the software just installed/reinstalled? Definitely an option!