MichaIng/DietPi #4021: Failed Upgrade of DietPi from v6.33.3 to v6.34.3.

cdwebb0, 2020-12-29:
#### Details:
- Date           | Tue Dec 29 16:28:43 EST 2020
- DietPi version | v6.33.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux DietPi 5.4.72+ #1356 Thu Oct 22 13:56:00 BST 2020 armv6                                                                                                                                                             l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Segmentation fault
Setting up libssl1.1:armhf (1.1.1d-0+deb10u4+rpt1) ...
dpkg: error processing package libssl1.1:armhf (--configure):
 installed libssl1.1:armhf package post-installation script subprocess was kille                                                                                                                                                             d by signal (Segmentation fault)
Setting up raspberrypi-kernel (1.20201201-1) ...
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.4.79+ /boot/kerne                                                                                                                                                             l.img
/etc/kernel/postinst.d/apt-auto-removal:
Segmentation fault
run-parts: /etc/kernel/postinst.d/apt-auto-removal exited with return code 139
dpkg: error processing package raspberrypi-kernel (--configure):
 installed raspberrypi-kernel package post-installation script subprocess return                                                                                                                                                             ed error exit status 1
Setting up tzdata (2020e-0+deb10u1) ...
dpkg: error processing package tzdata (--configure):
 installed tzdata package post-installation script subprocess was killed by sign                                                                                                                                                             al (Segmentation fault)
Setting up udev (241-7~deb10u5+rpi1) ...
Segmentation fault
dpkg: error processing package udev (--configure):
 installed udev package post-installation script subprocess returned error exit                                                                                                                                                              status 139
dpkg: dependency problems prevent configuration of openssl:
 openssl depends on libssl1.1 (>= 1.1.1); however:
  Package libssl1.1:armhf is not configured yet.

dpkg: error processing package openssl (--configure):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.28-10+rpi1) ...
Errors were encountered while processing:
 libssl1.1:armhf
 raspberrypi-kernel
 tzdata
 udev
 openssl
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2020-12-29:
Hi,

many thanks for your message. There quite some errors on the `apt upgrade` process. Pls can you run following and post the outcome.

`apt update && apt upgrade`

cdwebb0, 2020-12-29:
> Hi,
> 
> many thanks for your message. There quite some errors on the `apt upgrade` process. Pls can you run following and post the outcome.
> 
> `apt update && apt upgrade`

Hello,

root@DietPi:~# apt update && apt upgrade
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
4 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Setting up libssl1.1:armhf (1.1.1d-0+deb10u4+rpt1) ...
Setting up tzdata (2020e-0+deb10u1) ...

Current default time zone: 'America/New_York'
Local time is now:      Tue Dec 29 17:00:53 EST 2020.
Universal Time is now:  Tue Dec 29 22:00:53 UTC 2020.
Run 'dpkg-reconfigure tzdata' if you wish to change it.

Setting up udev (241-7~deb10u5+rpi1) ...
Setting up openssl (1.1.1d-0+deb10u4+rpt1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...

Joulinar, 2020-12-29:
Does this complete without issue? If yes, you can try running `dietpi-update` again

cdwebb0, 2020-12-29:
> Does this complete without issue? If yes, you can try running `dietpi-update` again

Yes, 'apt update && apt upgrade' completed without issue.

dietpi-update has now completed successfully.  Any idea what was causing the issue before 'apt update && apt upgrade'?

Joulinar, 2020-12-29:
before DietPi will update it's own scripts, an `apt update && apt upgrade` is called. This first initial step failed on your side. And we did nothing else than repeating this step. Probably you had to many packages and you where running into some memory issues. Or there was an issue switching the kernel to a newer version, leading to these side effects. As you can see, you had as well errors related to updating the raspberry kernel

```
Setting up raspberrypi-kernel (1.20201201-1) ...
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.4.79+ /boot/kernel.img
/etc/kernel/postinst.d/apt-auto-removal:
Segmentation fault
```

5 packages failed on your first attempt
```
Errors were encountered while processing:
 libssl1.1:armhf
 raspberrypi-kernel
 tzdata
 udev
 openssl
```

however on the 2nd run, only 4 ot setup. The kernel package was missing
```
Setting up libssl1.1:armhf (1.1.1d-0+deb10u4+rpt1) ...
Setting up tzdata (2020e-0+deb10u1) ...
Setting up udev (241-7~deb10u5+rpi1) ...
Setting up openssl (1.1.1d-0+deb10u4+rpt1) ...
```

at the end hard to say.

Joulinar, 2020-12-29:
will close the issue, feel free to reopen if needed

closed 2020-12-29 by Joulinar