MichaIng/DietPi #4023: erro upgrade dietpi

carlosspre, 2020-12-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

ravenclaw900, 2020-12-30:
Hello,
Could you please provide some information about your system and the error?  Filling out the bug report form would be easiest.

carlosspre, 2020-12-30:
[image]

ravenclaw900, 2020-12-30:
After the `sudo dpkg --configure -a` completes, could you try again?

carlosspre, 2020-12-30:
[image]

no more errors, how can I solve this

Joulinar, 2020-12-30:
ho long is you system in this status?

carlosspre, 2020-12-30:
2 months ,, I'm using pi4 4g

Joulinar, 2020-12-30:
no I mean the command, how long is it hanging?

carlosspre, 2020-12-30:
I didn't understand friend

Joulinar, 2020-12-30:
you have entered the command `dpkg --configure -a`. Does you system do anything? Dos it nothing? How long is it doing nothing? Did you cancelled the command in meantime?

carlosspre, 2020-12-30:
I did run the command dpkg --configure -a, but it doesn't do it anymore, and it doesn't solve the problem

Joulinar, 2020-12-30:
> but it doesn't do it anymore

what do you mean by this? Is your system stuck or did the command finished

carlosspre, 2020-12-30:
command ended

Joulinar, 2020-12-30:
can you run `apt update && apt upgrade`

carlosspre, 2020-12-30:
[image]

apt update ok

carlosspre, 2020-12-30:
[image]

apt upgrade. not going correctly

Joulinar, 2020-12-30:
pls reboot your system. And there is no need to copy screen prints. You can just copy/past the text out of the SSH gui. this is much more readable.

MichaIng, 2020-12-30:
General hint here: Whenever `rpi-eeprom` is stuck/terminated, it should be assured that the EEPROM has been successfully flashed: `rpi-eeprom-update -a`
Else the RPi 4 might become unbootable without the ability to fix it by flashing any OS image. I guess there is a possibility for offline EEPROM flash... ...ah okay an official recovery image exists that obviously is able to boot with broken EEPROM: https://jamesachambers.com/raspberry-pi-4-bootloader-firmware-updating-recovery-guide/

However, it seems to have ended. 220 package upgrades are indeed nearly impossible on a Raspbian Buster based image, probably something with that package states file is wrong.

carlosspre, 2021-02-08:
- DietPi has encounted an error, and, is unable to continue         -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket              -
---------------------------------------------------------------------

#### Details:
- Date           | Mon  8 Feb 22:50:36 GMT 2021
- Bug report     | d7a220ef-87b4-40aa-85fb-dba9d86db73f
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux NetPlexPi4 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
```

Joulinar, 2021-02-08:
you posted into 2 different issues now, pls try to stick to one.

Joulinar, 2021-02-09:
btw we should close this one in favour of https://github.com/MichaIng/DietPi/issues/4097

closed 2021-02-15 by MichaIng