MichaIng/DietPi #4025: Move radarr database folder to ssd.

Dr0bac, 2020-12-30:
I tried to move database folder to the new drive without success.
First i stoped radarr service `service radarr stop`
then i plugged ssd to raspberry pi, pount point is `/mnt/database`
then i edited `/etc/systemd/system/radarr.service` to change `-data` flag

````
[Unit]
Description=Radarr Daemon (DietPi)
Wants=network-online.target
After=network-online.target dietpi-boot.service

[Service]
SyslogIdentifier=Radarr
User=radarr
UMask=002
ExecStart=/opt/radarr/Radarr -nobrowser -data=/mnt/database/radarr
TimeoutStopSec=20
KillMode=process
Restart=on-failure

# Hardening
ProtectSystem=strict
ProtectHome=true
PrivateDevices=true
ProtectKernelTunables=true
ProtectControlGroups=true
ReadWritePaths=/opt/radarr /mnt /media /var/log/radarr /tmp

[Install]
WantedBy=multi-user.target
````

and finally i moved database folder from `/mnt/dietpi_userdata` to `/mnt/database`
and when i started radarr with `service radarr start` i got warning:

`Warning: The unit file, source configuration file or drop-ins of radarr.service changed on disk. Run 'systemctl daemon-reload' to reload units.`

And radarr created new database folder on `/mnt/dietpi_userdata` and it using this folder, and not the one i moved to `/mnt/database`

when i entered `systemctl daemon-reload` i got this error:
````
root@DietPi:~# systemctl status radarr.service
● radarr.service - Radarr Daemon (DietPi)
   Loaded: loaded (/etc/systemd/system/radarr.service; disabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-12-30 13:04:37 CET; 4min 43s ago
 Main PID: 3609 (Radarr)
    Tasks: 11 (limit: 2182)
   CGroup: /system.slice/radarr.service
           └─3609 /opt/radarr/Radarr -nobrowser -data=/mnt/database/radarr

Dec 30 13:04:37 DietPi systemd[1]: Started Radarr Daemon (DietPi).
Dec 30 13:04:40 DietPi Radarr[3609]: [Info] Bootstrap: Starting Radarr - /opt/radarr/Radarr.dll - Version 3.0.1.4259
Dec 30 13:04:42 DietPi Radarr[3609]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/database/radarr]
Dec 30 13:04:42 DietPi Radarr[3609]: [Fatal] ConsoleApp: EPIC FAIL!
Dec 30 13:04:42 DietPi Radarr[3609]: [v3.0.1.4259] NzbDrone.Common.Exceptions.RadarrStartupException: Radarr failed to start: Cannot create AppFolder, Access to the path /mnt/database/radar
r is denied
Dec 30 13:04:42 DietPi Radarr[3609]:    at NzbDrone.Common.EnvironmentInfo.AppFolderFactory.Register() in D:\a\1\s\src\NzbDrone.Common\EnvironmentInfo\AppFolderFactory.cs:line 48
Dec 30 13:04:42 DietPi Radarr[3609]:    at Radarr.Host.Bootstrap.Start(StartupContext startupContext, IUserAlert userAlert, Action`1 startCallback) in D:\a\1\s\src\NzbDrone.Host\Bootstrap.c
s:line 33
Dec 30 13:04:42 DietPi Radarr[3609]:    at NzbDrone.Console.ConsoleApp.Main(String[] args) in D:\a\1\s\src\NzbDrone.Console\ConsoleApp.cs:line 41
Dec 30 13:04:42 DietPi Radarr[3609]: Press enter to exit...
Dec 30 13:04:43 DietPi Radarr[3609]: Non-recoverable failure, waiting for user intervention...
````
What should i do?

Joulinar, 2020-12-30:
Hi,

Did you check for correct permission? Looks like Radarr is not allowed to write to the new location. What file system format you are using on the SSD?

`Cannot create AppFolder, Access to the path /mnt/database/radarr is denied`

Dr0bac, 2020-12-30:
I know that @MichaIng hardened permissions of radarr/sonarr programs..
I see in .service file `ReadWritePaths=/opt/radarr /mnt`

So `/mnt` us included than?

Joulinar, 2020-12-30:
I guess that's the path for storing download

Dr0bac, 2020-12-30:
What should i do? What permissions to set to this ssd?
I want to move database folders of radarr, sonarr, bazarr to this ssd.

Joulinar, 2020-12-30:
best is to copy files with current permission `cp -p`. Still the question. What file system format you are using on the SSD. Another option is to move whole `/mnt/dietpi_userdata` to your SSD, or you move whole RootFS, or (if supported) the entire system

Dr0bac, 2020-12-30:
Last time i tried to start radarr i didn't move database folder, and radarr tried to make new radarr database on /mnt/database and it failed, and i guess radarr make database folder with correct permissions. So if i move radarr folder with -p sign i guess it will be the same.. ssd is formated with ext4. I can't move whole mnt/dietpi_userdata, i have 300 gb of movies there..

Joulinar, 2020-12-30:
I guess you need to give r/w access for `radarr` user to `/mnt/database` all related files. If you copy all files with `cp -p`, permission will be keeped. That's what `-p` does. Worst case scenario is to use 777 permission.

MichaIng, 2020-12-30:
Please try:
```
chown -R radarr:dietpi /mnt/database
```

General hint: Do not edit `/etc/systemd/system/radarr.service` directly, as this gets overwritten on reinstall. Instead create an override config in e.g. `/etc/systemd/system/radarr.service.d/custom_database.conf` with only content:
```
[Service]
ExecStart=/opt/radarr/Radarr -nobrowser -data=/mnt/database/radarr
```
to override the start command. This can be done as well via `dietpi-services` > `radarr` > `Edit`.

closed 2020-12-31 by Dr0bac