MichaIng/DietPi #4026: install fails on dietpi v6.34.3

jeroen1973, 2020-12-30:
When i try to install Unbound on dietpi via the diepti-software command it fails with this error

DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring Unbound: validating, recursive, caching DNS resolver

[  OK  ] DietPi-Software | Checking URL: https://www.internic.net/domain/named.root
[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | curl -sSfL https://www.internic.net/domain/named.root -o named.root
[  OK  ] DietPi-Software | Verifying download target: /var/lib/unbound/root.hints
[ INFO ] DietPi-Software | Updating file: /var/lib/unbound/root.hints
[  OK  ] DietPi-Software | mv named.root /var/lib/unbound/root.hints
[  OK  ] DietPi-Software | Setting in /etc/unbound/unbound.conf.d/dietpi.conf adjusted:     do-ip6: no
[ INFO ] DietPi-Software | Configuring Unbound to work with Pi-hole
[  OK  ] DietPi-Software | Desired setting in /etc/pihole/setupVars.conf was already set: PIHOLE_DNS_1=127.0.0.1#5353
[  OK  ] DietPi-Software | Desired setting in /etc/pihole/setupVars.conf was already set: PIHOLE_DNS_2=
[FAILED] DietPi-Software | systemctl restart unbound
[FAILED] DietPi-Software | systemctl restart unbound

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Wed Dec 30 16:52:08 CET 2020
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | systemctl restart unbound
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Job for unbound.service failed because the control process exited with error code.
See "systemctl status unbound.service" and "journalctl -xe" for details.
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

root@DietPi:~#

Joulinar, 2020-12-30:
Hi,

Many thanks for your message. Currently there is an issue on Unbound install. Pls can you retry.

MichaIng, 2020-12-30:
When running into the same issue, please open a subshell from the error prompt and do:
```
rm /etc/unbound/unbounc.conf.d/dietpi-pihole.conf
G_CONFIG_INJECT 'port:[[:blank:]]' '	port: 5335' /etc/unbound/unbound.conf.d/dietpi.conf
G_CONFIG_INJECT 'interface:[[:blank:]]' '	interface: 127.0.0.1' /etc/unbound/unbound.conf.d/dietpi.conf
```
Note that we changed the default port for usage with Pi-hole from 5353 to 5335, since the first is already the default port for mDNS (Multicast DNS), so this is what needs to be set in Pi-hole then as well as upstream DNS: `127.0.0.1#5335`

jeroen1973, 2020-12-30:
Just tried it and again it fails

Jeroen van der Wal

ravenclaw900, 2020-12-30:
What is the output of the command `journalctl -u unbound`?

jeroen1973, 2020-12-31:
journalctl -u unbound Gave this:

Jeroen van der Wal

Op 30 dec. 2020 om 23:00 heeft ravenclaw900 <notifications@github.com> het volgende geschreven:

journalctl -u unbound

Joulinar, 2020-12-31:
Pls try to upload your information to GitHub directly

jeroen1973, 2021-01-02:
root@DietPi:~# journalctl -u unbound
-- Logs begin at Thu 2019-02-14 11:11:58 CET, end at Sat 2021-01-02 12:03:20 CET. --
dec 30 22:10:40 DietPi systemd[1]: Starting Unbound DNS server...
dec 30 22:10:47 DietPi package-helper[407]: /var/lib/unbound/root.key has content
dec 30 22:10:47 DietPi package-helper[407]: success: the anchor is ok
dec 30 22:10:48 DietPi unbound[457]: [1609362648] unbound[457:0] error: cannot parse ip address: '.0/24'
dec 30 22:10:48 DietPi unbound[457]: [1609362648] unbound[457:0] error: cannot parse access control: .0/24 allow
dec 30 22:10:48 DietPi unbound[457]: [1609362648] unbound[457:0] fatal error: Could not setup access control list
dec 30 22:10:48 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
dec 30 22:10:48 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
dec 30 22:10:48 DietPi systemd[1]: Failed to start Unbound DNS server.
dec 30 22:10:48 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
dec 30 22:10:48 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 1.
dec 30 22:10:48 DietPi systemd[1]: Stopped Unbound DNS server.
dec 30 22:10:48 DietPi systemd[1]: Starting Unbound DNS server...
dec 30 22:10:48 DietPi package-helper[484]: /var/lib/unbound/root.key has content
dec 30 22:10:48 DietPi package-helper[484]: success: the anchor is ok
dec 30 22:10:48 DietPi unbound[506]: [1609362648] unbound[506:0] error: cannot parse ip address: '.0/24'
dec 30 22:10:48 DietPi unbound[506]: [1609362648] unbound[506:0] error: cannot parse access control: .0/24 allow
dec 30 22:10:48 DietPi unbound[506]: [1609362648] unbound[506:0] fatal error: Could not setup access control list
dec 30 22:10:48 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
dec 30 22:10:48 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
dec 30 22:10:48 DietPi systemd[1]: Failed to start Unbound DNS server.
dec 30 22:10:48 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
dec 30 22:10:48 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 2.
dec 30 22:10:48 DietPi systemd[1]: Stopped Unbound DNS server.
dec 30 22:10:48 DietPi systemd[1]: Starting Unbound DNS server...
dec 30 22:10:48 DietPi package-helper[510]: /var/lib/unbound/root.key has content
dec 30 22:10:48 DietPi package-helper[510]: success: the anchor is ok
dec 30 22:10:48 DietPi unbound[514]: [1609362648] unbound[514:0] error: cannot parse ip address: '.0/24'
dec 30 22:10:48 DietPi unbound[514]: [1609362648] unbound[514:0] error: cannot parse access control: .0/24 allow
dec 30 22:10:48 DietPi unbound[514]: [1609362648] unbound[514:0] fatal error: Could not setup access control list
dec 30 22:10:48 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
dec 30 22:10:48 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
dec 30 22:10:48 DietPi systemd[1]: Failed to start Unbound DNS server.
dec 30 22:10:49 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
dec 30 22:10:49 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 3.
dec 30 22:10:49 DietPi systemd[1]: Stopped Unbound DNS server.
dec 30 22:10:49 DietPi systemd[1]: Starting Unbound DNS server...
dec 30 22:10:50 DietPi package-helper[520]: /var/lib/unbound/root.key has content
dec 30 22:10:50 DietPi package-helper[520]: success: the anchor is ok
dec 30 22:10:50 DietPi unbound[525]: [1609362650] unbound[525:0] error: cannot parse ip address: '.0/24'
dec 30 22:10:50 DietPi unbound[525]: [1609362650] unbound[525:0] error: cannot parse access control: .0/24 allow
dec 30 22:10:50 DietPi unbound[525]: [1609362650] unbound[525:0] fatal error: Could not setup access control list
dec 30 22:10:50 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
dec 30 22:10:50 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
dec 30 22:10:50 DietPi systemd[1]: Failed to start Unbound DNS server.
dec 30 22:10:50 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
dec 30 22:10:50 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 4.
dec 30 22:10:50 DietPi systemd[1]: Stopped Unbound DNS server.
dec 30 22:10:50 DietPi systemd[1]: Starting Unbound DNS server...
dec 30 22:11:02 DietPi package-helper[532]: /var/lib/unbound/root.key has content
dec 30 22:11:02 DietPi package-helper[532]: success: the anchor is ok
dec 30 22:11:02 DietPi unbound[559]: [1609362662] unbound[559:0] error: cannot parse ip address: '.0/24'
dec 30 22:11:02 DietPi unbound[559]: [1609362662] unbound[559:0] error: cannot parse access control: .0/24 allow
dec 30 22:11:02 DietPi unbound[559]: [1609362662] unbound[559:0] fatal error: Could not setup access control list
dec 30 22:11:02 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
dec 30 22:11:02 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
dec 30 22:11:02 DietPi systemd[1]: Failed to start Unbound DNS server.
dec 30 22:11:02 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
dec 30 22:11:02 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 5.
[... cut ...]
jan 02 12:03:20 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
jan 02 12:03:20 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 5.
jan 02 12:03:20 DietPi systemd[1]: Stopped Unbound DNS server.
jan 02 12:03:20 DietPi systemd[1]: unbound.service: Start request repeated too quickly.
jan 02 12:03:20 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
jan 02 12:03:20 DietPi systemd[1]: Failed to start Unbound DNS server.

Joulinar, 2021-01-02:
Ahh, what IP address you are using? Currently there is an issue not allowing all privat IP adresses.

jeroen1973, 2021-01-02:
the raspberrypi has a static ipadress 10.19.19.100

MichaIng, 2021-01-02:
Jep that it is it. Take your new configuration file which as well contains some informational comments, is more sorted and allows all private IP ranges:
```
curl -# https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_182/unbound.conf > /etc/unbound/unbound.conf.d/dietpi.conf
G_CONFIG_INJECT 'port:[[:blank:]]' '	port: 5335' /etc/unbound/unbound.conf.d/dietpi.conf
G_CONFIG_INJECT 'interface:[[:blank:]]' '	interface: 127.0.0.1' /etc/unbound/unbound.conf.d/dietpi.conf
```

Joulinar, 2021-01-02:
is that correct? `curl -#`

MichaIng, 2021-01-02:
`-#` option reduces the fully fledged progress meter into a one-liner bar, sufficient for such a small text file.

MichaIng, 2021-01-02:
Whoospie, sorry wrong branch, just updated it above 😉.

jeroen1973, 2021-01-02:
thank you for helping out, but with the (updated) commands it still fails to install....

^[[Aroot@DietPi:~# journalctl -u unbound
-- Logs begin at Thu 2019-02-14 11:11:58 CET, end at Sat 2021-01-02 13:12:52 CET. --
jan 02 13:06:24 DietPi systemd[1]: Starting Unbound DNS server...
jan 02 13:06:32 DietPi package-helper[406]: /var/lib/unbound/root.key has content
jan 02 13:06:32 DietPi package-helper[406]: success: the anchor is ok
jan 02 13:06:32 DietPi unbound[457]: [1609589192] unbound[457:0] error: cannot parse ip address: '.0/24'
jan 02 13:06:32 DietPi unbound[457]: [1609589192] unbound[457:0] error: cannot parse access control: .0/24 allow
jan 02 13:06:32 DietPi unbound[457]: [1609589192] unbound[457:0] fatal error: Could not setup access control list
jan 02 13:06:32 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
jan 02 13:06:32 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
jan 02 13:06:32 DietPi systemd[1]: Failed to start Unbound DNS server.
jan 02 13:06:32 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
jan 02 13:06:32 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 1.
jan 02 13:06:32 DietPi systemd[1]: Stopped Unbound DNS server.
jan 02 13:06:32 DietPi systemd[1]: Starting Unbound DNS server...
jan 02 13:06:32 DietPi package-helper[491]: /var/lib/unbound/root.key has content
jan 02 13:06:32 DietPi package-helper[491]: success: the anchor is ok
jan 02 13:06:32 DietPi unbound[506]: [1609589192] unbound[506:0] error: cannot parse ip address: '.0/24'
jan 02 13:06:32 DietPi unbound[506]: [1609589192] unbound[506:0] error: cannot parse access control: .0/24 allow
jan 02 13:06:32 DietPi unbound[506]: [1609589192] unbound[506:0] fatal error: Could not setup access control list
jan 02 13:06:32 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
jan 02 13:06:32 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
jan 02 13:06:32 DietPi systemd[1]: Failed to start Unbound DNS server.
jan 02 13:06:33 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
jan 02 13:06:33 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 2.
jan 02 13:06:33 DietPi systemd[1]: Stopped Unbound DNS server.
jan 02 13:06:33 DietPi systemd[1]: Starting Unbound DNS server...
jan 02 13:06:33 DietPi package-helper[510]: /var/lib/unbound/root.key has content
jan 02 13:06:33 DietPi package-helper[510]: success: the anchor is ok
jan 02 13:06:33 DietPi unbound[514]: [1609589193] unbound[514:0] error: cannot parse ip address: '.0/24'
jan 02 13:06:33 DietPi unbound[514]: [1609589193] unbound[514:0] error: cannot parse access control: .0/24 allow
jan 02 13:06:33 DietPi unbound[514]: [1609589193] unbound[514:0] fatal error: Could not setup access control list
jan 02 13:06:33 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
jan 02 13:06:33 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
jan 02 13:06:33 DietPi systemd[1]: Failed to start Unbound DNS server.
jan 02 13:06:33 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
jan 02 13:06:33 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 3.
jan 02 13:06:33 DietPi systemd[1]: Stopped Unbound DNS server.
jan 02 13:06:33 DietPi systemd[1]: Starting Unbound DNS server...
jan 02 13:06:33 DietPi package-helper[521]: /var/lib/unbound/root.key has content
jan 02 13:06:33 DietPi package-helper[521]: success: the anchor is ok
jan 02 13:06:33 DietPi unbound[525]: [1609589193] unbound[525:0] error: cannot parse ip address: '.0/24'
jan 02 13:06:33 DietPi unbound[525]: [1609589193] unbound[525:0] error: cannot parse access control: .0/24 allow
jan 02 13:06:33 DietPi unbound[525]: [1609589193] unbound[525:0] fatal error: Could not setup access control list
jan 02 13:06:33 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
jan 02 13:06:33 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
jan 02 13:06:33 DietPi systemd[1]: Failed to start Unbound DNS server.
jan 02 13:06:34 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
jan 02 13:06:34 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 4.
jan 02 13:06:34 DietPi systemd[1]: Stopped Unbound DNS server.
jan 02 13:06:34 DietPi systemd[1]: Starting Unbound DNS server...
jan 02 13:06:34 DietPi package-helper[529]: /var/lib/unbound/root.key has content
jan 02 13:06:34 DietPi package-helper[529]: success: the anchor is ok
jan 02 13:06:34 DietPi unbound[533]: [1609589194] unbound[533:0] error: cannot parse ip address: '.0/24'
jan 02 13:06:34 DietPi unbound[533]: [1609589194] unbound[533:0] error: cannot parse access control: .0/24 allow
jan 02 13:06:34 DietPi unbound[533]: [1609589194] unbound[533:0] fatal error: Could not setup access control list
jan 02 13:06:34 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
jan 02 13:06:34 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
jan 02 13:06:34 DietPi systemd[1]: Failed to start Unbound DNS server.
jan 02 13:06:34 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
[... cut ...]
jan 02 13:12:52 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
jan 02 13:12:52 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 5.
jan 02 13:12:52 DietPi systemd[1]: Stopped Unbound DNS server.
jan 02 13:12:52 DietPi systemd[1]: unbound.service: Start request repeated too quickly.
jan 02 13:12:52 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
jan 02 13:12:52 DietPi systemd[1]: Failed to start Unbound DNS server.

MichaIng, 2021-01-02:
Please run again the commands and assure no errors happen, as obviously it did not work, as you still use the old configuration file. Check back with `cat /etc/unbound/unbound.conf.d/dietpi.conf`, which should look similar to that one, with just the settings from the other three commands adjusted: https://raw.githubusercontent.com/MichaIng/DietPi/unbound/.conf/dps_182/unbound.conf

jeroen1973, 2021-01-03:
Looks like the installation now worked and unbound is running. I put 127.0.0.1#5335 in the Pi-hole upstream dnsserver.

When in then test with "dig pi-hole.net @127.0.0.1 -p 5335"  the outcome is:   connection timed out; no servers could be reached

jeroen1973, 2021-01-03:
i give up, ditching dietpi and starting over with clean install Raspbian

jeroen1973, 2021-01-03:
thanks for the help

Joulinar, 2021-01-03:
You could check LISTEN ports. Probably it's running on a different one `ss -alnp | grep LISTEN`

MichaIng, 2021-01-03:
I guess there is a conflicting configuration file in place somewhere, maybe you tried another solutions from Pi-hole side?

There should be four configuration files:
- `/etc/unbound/unbound.conf` must contain nothing else than including the files from `/etc/unbound/unbound.conf.d`.
- `/etc/unbound/unbound.conf.d/qname-minimisation.conf` privacy-related snipped, shipped with the Debian package itself.
- `/etc/unbound/unbound.conf.d/root-auto-trust-anchor-file.conf` snipped for DNSSEC validation, shipped with the Debian package itself.
- `/etc/unbound/unbound.conf.d/dietpi.conf` our main configuration, configured for Pi-hole as above: https://github.com/MichaIng/DietPi/issues/4026#issuecomment-753464752

No other file should exist.

Following the Pi-hole guide the only difference would be that our `dietpi-conf` is/must be replaced with `pi-hole.conf` taken from here: https://docs.pi-hole.net/guides/unbound/#configure-unbound
Taking into account the defaults, it mostly matches ours, but misses the access limitation to local networks. But that is only relevant when using it directly, not when using it in combination with Pi-hole, as then it is bond to the loopback IP where only services from the same machine can access. I didn't think about hat when thinking that `10.19.19.0/24` network might be the issue 😉.

xczh, 2021-01-16:
Same issue. It seems a bug. Is there any solution? @Joulinar

Joulinar, 2021-01-16:
There a couple of solutions. But first we would need to know what your issue is exactly

xczh, 2021-01-16:
---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Sat Jan 16 23:25:09 CST 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | systemctl restart unbound
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Job for unbound.service failed because the control process exited with error code.
See "systemctl status unbound.service" and "journalctl -xe" for details.
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

Joulinar, 2021-01-16:
Pls can you share `journalctl -u unbound`

xczh, 2021-01-16:
Jan 17 00:09:01 DietPi systemd[1]: Starting Unbound DNS server...
Jan 17 00:09:01 DietPi package-helper[1593]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 00:09:01 DietPi package-helper[1593]: /etc/unbound/unbound.conf.d/dietpi.conf:75: error: unknown keyword '192.168.4.0/24'
Jan 17 00:09:01 DietPi package-helper[1593]: /etc/unbound/unbound.conf.d/dietpi.conf:75: error: unknown keyword 'allow'
Jan 17 00:09:01 DietPi package-helper[1593]: read /etc/unbound/unbound.conf failed: 3 errors in configuration file
Jan 17 00:09:01 DietPi package-helper[1596]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 00:09:01 DietPi package-helper[1596]: /etc/unbound/unbound.conf.d/dietpi.conf:75: error: unknown keyword '192.168.4.0/24'
Jan 17 00:09:01 DietPi package-helper[1596]: /etc/unbound/unbound.conf.d/dietpi.conf:75: error: unknown keyword 'allow'
Jan 17 00:09:01 DietPi package-helper[1596]: read /etc/unbound/unbound.conf failed: 3 errors in configuration file
Jan 17 00:09:01 DietPi unbound[1599]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 00:09:01 DietPi unbound[1599]: /etc/unbound/unbound.conf.d/dietpi.conf:75: error: unknown keyword '192.168.4.0/24'
Jan 17 00:09:01 DietPi unbound[1599]: /etc/unbound/unbound.conf.d/dietpi.conf:75: error: unknown keyword 'allow'
Jan 17 00:09:01 DietPi unbound[1599]: read /etc/unbound/unbound.conf failed: 3 errors in configuration file
Jan 17 00:09:01 DietPi unbound[1599]: [1610813341] unbound[1599:0] fatal error: Could not read config file: /etc/unbound/unbound.conf. Maybe try unbound -dd, it stays on the commandline to see more errors, or unbound-checkconf
Jan 17 00:09:01 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Jan 17 00:09:01 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
Jan 17 00:09:01 DietPi systemd[1]: Failed to start Unbound DNS server.
Jan 17 00:09:02 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Jan 17 00:09:02 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 5.
Jan 17 00:09:02 DietPi systemd[1]: Stopped Unbound DNS server.
Jan 17 00:09:02 DietPi systemd[1]: unbound.service: Start request repeated too quickly.
Jan 17 00:09:02 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
Jan 17 00:09:02 DietPi systemd[1]: Failed to start Unbound DNS server.

ravenclaw900, 2021-01-16:
What is the contents of your unbound configuration file? `cat /etc/unbound/unbound.conf.d/dietpi.conf`

xczh, 2021-01-16:
server:
    verbosity: 0

    interface: 0.0.0.0
    port: 53
    do-ip4: yes
    do-udp: yes
    do-tcp: yes

    do-ip6: yes

    prefer-ip6: no

    root-hints: "/var/lib/unbound/root.hints"

    harden-glue: yes
    harden-large-queries: yes

    harden-dnssec-stripped: yes

    use-caps-for-id: yes

    edns-buffer-size: 1472

    rrset-roundrobin: yes

    cache-min-ttl: 300
    cache-max-ttl: 86400

    serve-expired: yes

    harden-algo-downgrade: yes

    harden-short-bufsize: yes

    hide-identity: yes

    identity: "Server"

    hide-version: yes

    do-daemonize: no

    neg-cache-size: 4M

    qname-minimisation: yes

    minimal-responses: yes

    prefetch: yes
    prefetch-key: yes

    num-threads: 1

    msg-cache-size: 50m
    rrset-cache-size: 100m

    so-reuseport: yes

    so-rcvbuf: 4m
    so-sndbuf: 4m

    unwanted-reply-threshold: 10000

    ratelimit: 1000

    log-queries: no
    log-replies: no
    logfile: ''

    access-control: 0.0.0.0/0 refuse
    access-control: 127.0.0.1 allow
    access-control: 192.168.1
192.168.1
192.168.4.0/24 allow

    private-address: 192.168.0.0/16
    private-address: 169.254.0.0/16
    private-address: 172.16.0.0/12
    private-address: 10.0.0.0/8
    private-address: fd00::/8
    private-address: fe80::/10

xczh, 2021-01-16:
This is generated automatically and I haven't edited it.

xczh, 2021-01-16:
https://github.com/MichaIng/DietPi/blob/b354689141bf3b238b87d9ff83f33df5cf52173f/dietpi/dietpi-software#L8774

Maybe, this line is incorrect. On my machine, its execution result is:

```
192.168.1
192.168.1
192.168.4
```

@Joulinar @ravenclaw900

ravenclaw900, 2021-01-16:
That is the problem.  Does your Pi have more than 1 IPv4 address?  If so, do you want to use the `192.168.1` subnet or the `192.168.4` subnet?

xczh, 2021-01-16:
I do have more than one IP address. Even so, it gets the wrong subnet. Actually, my subnet is `192.168.137.0/24`.
I manually modified the correct subnet and rerun the `diet-software`. The problem has been solved, thanks!
Do you need me to make a PR?

Joulinar, 2021-01-16:
it's up to @MichaIng if he like to support this

ravenclaw900, 2021-01-16:
Ah, it seems that it only grabbed the first digit of `137`.  No need to make a PR, there are many enhancements for Unbound planned in 6.35, and this is already fixed there.

MichaIng, 2021-01-16:
With next version we have that hostname IP specific entry removed and allow all IP ranges that are reserved for private networks instead. That check had also the issue that it allows 192.168.X IPs only while there are a few other ranges common and reserved for LANs. Please use this one:
```
curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_182/unbound.conf > /etc/unbound/unbound.conf.d/dietpi.conf
systemctl restart unbound
```

burstina, 2021-01-17:
hello,
same here, 
I copy-paste `systemctl status unbound.service`` output too.

have fun

```
`--------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Sun Jan 17 19:20:19 GMT 2021
- Bug report     | 9acbdc7e-42d5-4261-bb34-f10c6125197d
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | StephanStS
- Pre-image      | Armbian
- Hardware       | NanoPi NEO3 (aarch64) (ID=56)
- Kernel version | Linux DietPi 5.9.14-rockchip64 #20.11.4 SMP PREEMPT Tue Dec 15 08:52:20 CET 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | systemctl restart unbound
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ...
2. ...
#### Expected behaviour:
<!-- What SHOULD happen? -->
- ...
#### Actual behaviour:
<!-- What IS happening? -->
- ...
#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...
#### Additional logs:
```
Job for unbound.service failed because the control process exited with error code.
See "systemctl status unbound.service" and "journalctl -xe" for details.
```
`
```

```
root@DietPi:/tmp/DietPi-Software# systemctl status unbound.service
● unbound.service - Unbound DNS server
   Loaded: loaded (/lib/systemd/system/unbound.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2021-01-17 19:22:47 GMT; 10s ago
     Docs: man:unbound(8)
  Process: 4957 ExecStartPre=/usr/lib/unbound/package-helper chroot_setup (code=exited, status=1/FAILURE)
  Process: 4960 ExecStartPre=/usr/lib/unbound/package-helper root_trust_anchor_update (code=exited, status=1/FAILURE)
  Process: 4963 ExecStart=/usr/sbin/unbound -d $DAEMON_OPTS (code=exited, status=1/FAILURE)
 Main PID: 4963 (code=exited, status=1/FAILURE)

Jan 17 19:22:47 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Jan 17 19:22:47 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 5.
Jan 17 19:22:47 DietPi systemd[1]: Stopped Unbound DNS server.
**Jan 17 19:22:47 DietPi systemd[1]: unbound.service: Start request repeated too quickly.
Jan 17 19:22:47 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.**
Jan 17 19:22:47 DietPi systemd[1]: Failed to start Unbound DNS server.
```

burstina, 2021-01-17:
sure,
but I think it is incomplete:

Jan 17 19:15:54 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
Jan 17 19:15:54 DietPi systemd[1]: Failed to start Unbound DNS server.
Jan 17 19:15:54 DietPi systemd[1]: unbound.service: Start request repeated too quickly.
Jan 17 19:15:54 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
Jan 17 19:15:54 DietPi systemd[1]: Failed to start Unbound DNS server.
Jan 17 19:16:25 DietPi systemd[1]: Starting Unbound DNS server...
Jan 17 19:16:25 DietPi package-helper[3154]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:25 DietPi package-helper[3154]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:25 DietPi package-helper[3154]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:25 DietPi package-helper[3157]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:25 DietPi package-helper[3157]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:25 DietPi package-helper[3157]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:25 DietPi unbound[3160]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:25 DietPi unbound[3160]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:25 DietPi unbound[3160]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:25 DietPi unbound[3160]: [1610910985] unbound[3160:0] fatal error: Could not read config file: /etc/unbound/unbound.conf. Maybe try unbound -dd, it stays on the commandline to see more errors, or unbound-checkconf
Jan 17 19:16:25 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Jan 17 19:16:25 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
Jan 17 19:16:25 DietPi systemd[1]: Failed to start Unbound DNS server.
Jan 17 19:16:26 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Jan 17 19:16:26 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 1.
Jan 17 19:16:26 DietPi systemd[1]: Stopped Unbound DNS server.
Jan 17 19:16:26 DietPi systemd[1]: Starting Unbound DNS server...
Jan 17 19:16:26 DietPi package-helper[3174]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:26 DietPi package-helper[3174]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:26 DietPi package-helper[3174]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:26 DietPi package-helper[3177]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:26 DietPi package-helper[3177]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:26 DietPi package-helper[3177]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:26 DietPi unbound[3180]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:26 DietPi unbound[3180]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:26 DietPi unbound[3180]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:26 DietPi unbound[3180]: [1610910986] unbound[3180:0] fatal error: Could not read config file: /etc/unbound/unbound.conf. Maybe try unbound -dd, it stays on the commandline to see more errors, or unbound-checkconf
Jan 17 19:16:26 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Jan 17 19:16:26 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
Jan 17 19:16:26 DietPi systemd[1]: Failed to start Unbound DNS server.
Jan 17 19:16:26 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Jan 17 19:16:26 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 2.
Jan 17 19:16:26 DietPi systemd[1]: Stopped Unbound DNS server.
Jan 17 19:16:26 DietPi systemd[1]: Starting Unbound DNS server...
Jan 17 19:16:26 DietPi package-helper[3181]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:26 DietPi package-helper[3181]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:26 DietPi package-helper[3181]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:26 DietPi package-helper[3184]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
[... cut ...]
Jan 17 19:20:59 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Jan 17 19:20:59 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 4.
Jan 17 19:20:59 DietPi systemd[1]: Stopped Unbound DNS server.
Jan 17 19:20:59 DietPi systemd[1]: Starting Unbound DNS server...
Jan 17 19:20:59 DietPi package-helper[4864]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:20:59 DietPi package-helper[4864]: /et

Joulinar, 2021-01-17:
`journalctl -u unbound` for @burstina 

<details>
  <summary>Logfile attached. Click to expand!</summary>

```
Jan 17 19:15:54 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
Jan 17 19:15:54 DietPi systemd[1]: Failed to start Unbound DNS server.
Jan 17 19:15:54 DietPi systemd[1]: unbound.service: Start request repeated too quickly.
Jan 17 19:15:54 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
Jan 17 19:15:54 DietPi systemd[1]: Failed to start Unbound DNS server.
Jan 17 19:16:25 DietPi systemd[1]: Starting Unbound DNS server...
Jan 17 19:16:25 DietPi package-helper[3154]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:25 DietPi package-helper[3154]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:25 DietPi package-helper[3154]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:25 DietPi package-helper[3157]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:25 DietPi package-helper[3157]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:25 DietPi package-helper[3157]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:25 DietPi unbound[3160]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:25 DietPi unbound[3160]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:25 DietPi unbound[3160]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:25 DietPi unbound[3160]: [1610910985] unbound[3160:0] fatal error: Could not read config file: /etc/unbound/unbound.conf. Maybe try unbound -dd, it stays on the commandline to see more errors, or unbound-checkconf
Jan 17 19:16:25 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Jan 17 19:16:25 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
Jan 17 19:16:25 DietPi systemd[1]: Failed to start Unbound DNS server.
Jan 17 19:16:26 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Jan 17 19:16:26 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 1.
Jan 17 19:16:26 DietPi systemd[1]: Stopped Unbound DNS server.
Jan 17 19:16:26 DietPi systemd[1]: Starting Unbound DNS server...
Jan 17 19:16:26 DietPi package-helper[3174]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:26 DietPi package-helper[3174]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:26 DietPi package-helper[3174]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:26 DietPi package-helper[3177]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:26 DietPi package-helper[3177]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:26 DietPi package-helper[3177]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:26 DietPi unbound[3180]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:26 DietPi unbound[3180]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:26 DietPi unbound[3180]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
Jan 17 19:16:26 DietPi unbound[3180]: [1610910986] unbound[3180:0] fatal error: Could not read config file: /etc/unbound/unbound.conf. Maybe try unbound -dd, it stays on the commandline to see more errors, or unbound-checkconf
Jan 17 19:16:26 DietPi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Jan 17 19:16:26 DietPi systemd[1]: unbound.service: Failed with result 'exit-code'.
Jan 17 19:16:26 DietPi systemd[1]: Failed to start Unbound DNS server.
Jan 17 19:16:26 DietPi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Jan 17 19:16:26 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 2.
Jan 17 19:16:26 DietPi systemd[1]: Stopped Unbound DNS server.
Jan 17 19:16:26 DietPi systemd[1]: Starting Unbound DNS server...
Jan 17 19:16:26 DietPi package-helper[3181]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:16:26 DietPi package-helper[3181]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: unknown keyword 'allow'
Jan 17 19:16:26 DietPi package-helper[3181]: read /etc/unbound/unbound.conf failed: 2 errors in configuration file
[... cut ...]
Jan 17 19:20:59 DietPi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 4.
Jan 17 19:20:59 DietPi systemd[1]: Stopped Unbound DNS server.
Jan 17 19:20:59 DietPi systemd[1]: Starting Unbound DNS server...
Jan 17 19:20:59 DietPi package-helper[4864]: /etc/unbound/unbound.conf.d/dietpi.conf:74: error: expected deny, refuse, deny_non_local, refuse_non_local, allow, allow_setrd or allow_snoop in access control action
Jan 17 19:20:59 DietPi package-helper[4864]: /et
```
</details>

Joulinar, 2021-01-17:
@burstina 
do you like to use `PiHole` in addition or just `Unbound`?  If it is `Unbound` only, you could download our new config file 

```
curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_182/unbound.conf -o /etc/unbound/unbound.conf.d/dietpi.conf
systemctl restart unbound
```

burstina, 2021-01-19:
thanks for the advice, the true is I don't know both 😄
I begun learning linux just some months ago and I'm triying to building up a 4G gateway from scratch.
Atm I turned to ** dnsmasq ** and it was helpful to achieve the result, the best solution should be a web interface: Does PiHole allow web interface?

thanks

Joulinar, 2021-01-19:
ahm PiHole has an own web interface

burstina, 2021-01-19:
then I'm interested!
Thanks a lot

Joulinar, 2021-01-19:
I would recommand following

1. remove `unbound`
2. install PiHole (without `unbound`)
3. wait until 6.35 before re-install `unbound` as we will deliver quite some fixes on this release

MichaIng, 2021-01-19:
It's not too difficult to apply these changes directly, with and without Pi-hole: https://github.com/MichaIng/DietPi/issues/4026#issuecomment-753464752
- The first command for standalone Unbound.
- Additionally the two command below in combination with Pi-hole.
- The last fourth command only if you have IPv6 disabled.

burstina, 2021-01-19:
Ok, I tried PiHole, but PiHole itself tried to install Unbound and got the same error, even after 
curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_182/unbound.conf -o /etc/unbound/unbound.conf.d/dietpi.conf
systemctl restart unbound

burstina, 2021-01-19:
> if you install PiHole, you get a dialog asking if you like to install Unbound as well. There you need to decline. Otherwise , yes Unbound will be installed as well.
> 
> @MichaIng maybe we would need to switch option to make unbound op-in instead of opt-out

ah, ok, sorry I didnt noticed it was Unbound to be asked for. Sorry I'm a bit dyslexic and console font is a bit of aproblem.
thanks again

Joulinar, 2021-01-19:
no problem, we are here to help

closed 2021-02-15 by MichaIng