MichaIng/DietPi #4027: Feat request: Add support in drive manager to specify owner

danrneal, 2021-01-01:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

When using drive manager, an external drive is mounted as root:root.  I'm order to have nextcloud be able to write to it, I have to edit /etc/fstab and add a uid and gid so that it is mounted by www-data user.

#### Describe the solution you'd like:

Allow drive manager to have an option to select a different owner than root

#### Describe alternatives you've considered:

Having supper support in drive manager would be ideal rather than having to mess with fstab or chown or chmod manually.

#### Additional context

- ...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2021-01-01:
Many thanks for your request. Probably we can generally add some file system specific mount options. Auto-compression on Btrfs has been requested, e.g.. The problem with those permissions is that every file system has its own options for those, although most share "uid"/"gid".

danrneal, 2021-01-01:
Happy new year 🥳

I totally get that going too deep into config would get pretty messy since there is so much divergence.  However, like you said uid and gid are pretty standard and would solve most if not all permission issues.