MichaIng/DietPi #4028: Software request: ufw

danrneal, 2021-01-01:
### Creating a software request

#### Formal software information
- Software title | ufw
- Short description | program for managing a netfilter firewall
- Official URL (if available) | http://manpages.ubuntu.com/manpages/focal/en/man8/ufw.8.html

#### Are there similar/alternative software titles available with DietPi-Software?

Failtoban is relevant in that they both are for security but ufw addresses it in a much different way.

#### What makes your requested software better than the above solutions, if available?

Super easy to implement firewall for increased security.  Can easily just allow ssh and or 80/443

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

Could just put it under other software and allow user management from the command line or could create a gui for optimized software.  When it says "uncomplicated" it really is so cli might be sufficient.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

Install ufw
Optional: Open ports automatically when relevant software is installed

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

Am available to test

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2021-01-01:
Many thanks for your request.

ufw is basically a frontend for iptables with a simplified command set. We currently use iptables for network packet forwarding with VPNs and hotspots, so we'd need to make sure that ufw does not touch any netfilter/iptables rules applied manually, via iptables-restore (from network bring-up configs), iptables-persistent and such. My personal aim when adding something to `dietpi-software` is that it is either OOTB or configured to be not conflicting with any other `dietpi-software` implementation. Everyone can simply run `apt install ufw` but then we assume that necessary configuration is done as well to not break e.g. shared network forwarding. Keeping ports for server software open is another thing, but that is probably easy and obvious enough.

I was actually thinking about an own firewall script, that as well invokes the modern bpfilter when supported by the kernel. I'm not sure about the plans for ufw, if they migrate to modern firewall technologies or fade out with iptables (which practically already utilises nftables, not sure if an iptables->bpfilter incorporation will be done as well).
____________________
Btw a reason why I personally never started with a firewall implementation/integration is that, apart for lets say being deployed on a router to restrict access from/to the whole local network, there is no point in blocking all ports but 22/80/443 when there is simply nothing listening (or sending) on other ports. Check `ss -tulp`: if all that is listed is what you want to allow with the firewall anyway, the firewall has, simplified speaking, no point 😉. Very most software as well has their own access restriction settings, to listen on certain IPs/Interfaces only, to allow access from certain hosts/IPs only etc. Additionally most users have a router so that NAT with required port forwarding serves as natural firewall layer. If DietPi is attached to www directly but some application shall be accessible from a second local network only, IP binding again is a simple solution. A firewall is mostly required if you run applications yourself that you don't trust (e.g. also regarding outgoing traffic) or cannot configure to limit access to your needs. Both is very rare in case of a Debian server system, I'd say. But of course, its another layer of security that usually does not hurt much, CPU usage/performance-wise.

danrneal, 2021-01-01:
My default mode on any linux setup is to always setup ufw..... Which is not a bad habit but you are of course right that the router should handle this with port forwarding.  However defense in depth is never bad either. :) So this should be a low or non priority.