MichaIng/DietPi #4030: resolvconf prepends cloudflare, breaks local resolution

f00b4r0, 2021-01-01:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.7

- Kernel version | `uname -a`
Linux NanoPiNEOm 5.9.14-sunxi #20.11.3 SMP Fri Dec 11 20:31:12 CET 2020 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
NanoPi NEO (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 1A

- SDcard used | (EG: SanDisk ultra)
SanDisk ultra

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
resolvconf

- Was the software title installed freshly or updated/migrated?
fresh install

- Can this issue be replicated on a fresh installation of DietPi?
yes

#### Steps to reproduce
Install DietPi and use DHCP on the LAN interface.

#### Expected behaviour
resolvconf shouldn't be preset to override DHCP data / outsmart its users.

#### Actual behaviour
By default resovlconf ships `/etc/resolvconf/resolv.conf.d/head` which contains:
```
nameserver 1.0.0.1
```

This gets prepended to the generated resolv.conf when using DHCP, overriding local (DHCP-provided) DNS, which in turn breaks local resolution when the provided DNS resolves LAN hosts: cloudflare is tried first by the resolver, and it obviously returns NX DOMAIN for the LAN hosts:
```
ping some_lan_host
ping: some_lan_host: Name or service not known
```

Joulinar, 2021-01-01:
Hi,

indeed it's part of the NanoPi NEO image. 

@MichaIng can you have a look

MichaIng, 2021-01-01:
Ah that is true, I remember this being an issue with some Armbian-based images, not sure why they actively created this fixed resolvconf entry, which will always be preferred as it's at the top of the list. I'll update the image now, it's quite old anyway.

MichaIng, 2021-01-01:
Image updated: https://dietpi.com/downloads/images/DietPi_NanoPiNEO-ARMv7-Buster.7z

f00b4r0, 2021-01-03:
HI, HNY.
FWIW it's not just the Neo, the NeoAir is also affected. I don't know how you generate your images but I guess they're close enough :)
FWIW I have an odroidC2 with armbian 20.11.3 installed and the resolvconf-shipped `head` file is empty.
Of course purging `resolvconf` entirely fixes the issue (tested locally on the NeoAir).
Cheers

MichaIng, 2021-01-03:
Makes sense, I guess it was a certain image charge around February-March this year. I'll update the NEO Air image as well.

Since I cannot think of a situation where `resolvconf` is required without the admin using it manually to e.g. add static DNS-related entries, or a DEB package pulling it in as dependency, it is pretty safe to purge it.

MichaIng, 2021-01-08:
NanoPi NEO Air image has been updated as well.

closed 2021-02-14 by MichaIng